bluez/bluez #453: OOB support for pairing Agent API

beku-epitome, 2023-01-12:
I'm looking for a way to implement custom OOB channel (so not using `neard`), but it seems there is no support for such use case in BlueZ. Would it be possible to extent the Agent1 API to support OOB pairing?

Vudentz, 2023-01-12:
@beku-epitome Not sure if extending the Agent API would be a good idea, usually OOB means you have to provide some data upfront, anyway perhaps we know better what is the usecase we can better assist with the introduction of a dedicated API.

beku-epitome, 2023-01-13:
The use case is that two devices have a low speed wired connection to each other when they are being charged, this would serve as the secure OOB channel. When in factory reset state, they perform the OOB pairing to get a secure BR/EDR link. I guess it would also be possible to pretend that both sides have KeyboardDisplay capability, and verify the passkey via the wired connection, but my understanding is that OOB provides the highest level of security.

Vudentz, 2023-01-20:
@beku-epitome not sure I follow how OOB provides the highest level of security? If you are using another transport to transfer the link-key and then install it as a new device we have no idea what level of security you had in the original transport where the keys were distributed. What we could do is add a new method to create a Device providing a link-key and then use is the Agent to authorize the new link-key so the agent gets to approve it at least.

beku-epitome, 2023-01-20:
> not sure I follow how OOB provides the highest level of security?

Ignoring how the OOB is implemented in practice, is using the actual OOB data capability during pairing more secure, than (ab)using Keyboard/Display capabilities and exchanging the passkey via the same OOB method?