bluez/bluez #454: CSIS support

beku-epitome, 2023-01-12:
I see that some initial progress has been made for adding Coordinated Set Identification Service support. When is it planned to be released? I'm specifically interested in BlueZ support for Set Member role, which requires knowledge of the exchanged pairing keys to encrypt the SIRK. The specification is not exactly a light read, so I'm not sure I understand it correctly, is it true that this service can already be implemented solely on the application side, if the SIRK is transferred in plaintext format?

Vudentz, 2023-01-12:
@beku-epitome Im working on it, the problem with encrypt SIRK is that you would require the LTK in order to encrypt it.

beku-epitome, 2023-01-13:
> the problem with encrypt SIRK is that you would require the LTK in order to encrypt it.

Yes, this is what led me to the conclusion that CSIS needs support on BlueZ library level, and thus this issue. However, the CSIS spec says that
> A server shall expose the SIRK in encrypted form or in plain text or by using an OOB procedure.

So if I choose to implement the CSIS server with SIRK provided in plaintext format, can this then be implemented with the current BlueZ version, only on the application side? Or are there other dependencies of a CSIS server that need context from BlueZ internally?

closed 2023-05-17 by beku-epitome