bluez/bluez #455: Devices connect them immediately disconnect

pierrelegall, 2023-01-12:
My bluetooth devices connect them immediately disconnect (sometimes looping a bit) after trying to connect them. I was supposing it was a Pipewire problem, however it does the same with all kind of devices, not only audio devices.

I tried removing `/var/lib/bluetooth` and restarting bluetooth service as [suggest in this topic](https://bbs.archlinux.org/viewtopic.php?id=267233) but without success.

It doesn't work since my last OS installation of Archlinux in December 2021. It worked like a charm on this same computer on Ubuntu 20.04.

Configuration:
- Computer: Lenovo X1 Carbon 5th gen, 2017
- OS: Linux 6.1.4-arch1-1 x86_64
- Desktop: GNOME Shell 43.2
- Bluez version: 5.66-1
- Pipewire version: 0.3.63

Vudentz, 2023-01-12:
@pierrelegall please add the HCI trace using btmon when you attempt to connect.

pierrelegall, 2023-01-15:
When I'm trying to connect an Orochi V2 mouse, `btmon` outputs [this](https://gist.github.com/pierrelegall/4689ad740021d788fccbfe67ef3b2876).

pierrelegall, 2023-03-06:
My bad, applying a `rm -rf /var/lib/bluetooth` force me to pair my devices as a new devices. Which I didn't do.

However, I just test my bluetooth on the same laptop, on a live Ubuntu 22.04 (fresh, not updated, Bluez 5.64 installed by default). There is no connect-disconnect issue, it just refused to connect, which is a more expected behavior.

Solved for me :)

closed 2023-03-06 by pierrelegall