bluez/bluez #456: input-hog profile accept failed for XX:XX:XX

WhyNotHugo, 2023-01-17:
I'm trying to connect to a bluetooth mouse. Bluetooth is otherwise working on this system; I can use headphones just fine.

I'm connecting using `bluetoothctl` to keep things simple (also, blueman fails to connect entirely).

My basic process is:

- `bluetoothctl`
- `scan on`
- `connect XX:XX:XX...`
- I get a prompt to confirm pairing. I type `yes` and hit enter.
- `bluetoothctl` now indicates: `Bonded: yes` and `Paired: yes`.

However, the mouse don't work (e.g.: moving it doesn't move the cursor). `dmesg` provides nothing of use, but `/var/log/messages` shows:

    daemon.err bluetoothd[19440]: src/service.c:service_accept() input-hog profile accept failed for XX:XX:XX:XX:XX:XX

I've no idea what this means tho. Have I hit a bug, or is this an issue with my setup?

WhyNotHugo, 2023-01-17:
Huh, pairing the mouse with the logitech receiver (using [solaar](https://github.com/pwr-Solaar/Solaar)) prompts me to click left/right in a specified sequence in order to pair.

I wonder if support for something similar is required via bluetooth, or if that's just the receiver.

The device is  a "Logitech MX Master 3B".

icedream, 2023-03-06:
I have run into this issue as well, the mouse will pair and connect just fine, but this exact error message will pop up twice during a second after connection and then no inputs will work.

In my case this is the Logitech G604.

terrancewong, 2024-01-27:
fixed / workaround found. check #531

WhyNotHugo, 2024-04-01:
The fix in https://github.com/bluez/bluez/issues/531#issuecomment-1913058753 works for me.

I'm not entirely sure if this is the same issue or not; seems to be different hardware, but the problem is the same module not being loaded in time.

honkey48, 2024-04-21:
I had this problem with bluetooth mice. laptor with 3 distros and all was well. installed Nobira and it was ok. The other distros, various flavors of Debian, lost bluetooth mouse. They connected but no action. Someting from the Nobira install "bled in the other distros (separate partitions) cause bluetooth mouse fail. Can anyone venture a guess as to what happened? The above worked.