bluez/bluez #457: ASAN use-after-free in src/shared/bap.c:bap_stream_clear_cfm (909)

pv, 2023-01-17:
Seeing an ASAN use-after-free when running BlueZ from master (a1736d8990ff56bba453ff81a25156316bdd118f) and using BAP. 

From log looks like media.c can free the local PAC data while it's still referred to by an existing stream.

Log:
<details>

```
bluetoothd[100177]: profiles/audio/media.c:proxy_removed_cb() Proxy removed - removing endpoint: /MediaEndpoint/BAPSink/lc3
bluetoothd[100177]: Endpoint unregistered: sender=:1.464 path=/MediaEndpoint/BAPSink/lc3
bluetoothd[100177]: profiles/audio/media.c:media_endpoint_destroy() sender=:1.464 path=/MediaEndpoint/BAPSink/lc3
bluetoothd[100177]: profiles/audio/transport.c:media_transport_remove_owner() Transport /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_source0/fd6 Owner :1.464
bluetoothd[100177]: profiles/audio/transport.c:linked_transport_remove_owner() Transport /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_sink0/fd7 Owner :1.464
bluetoothd[100177]: profiles/audio/transport.c:media_owner_free() Owner :1.464
bluetoothd[100177]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_source0/fd6: TRANSPORT_STATE_ACTIVE -> TRANSPORT_STATE_IDLE
bluetoothd[100177]: profiles/audio/transport.c:transport_set_state() State changed /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_sink0/fd7: TRANSPORT_STATE_ACTIVE -> TRANSPORT_STATE_IDLE
bluetoothd[100177]: profiles/audio/transport.c:link_set_state() Unable to find transport
bluetoothd[100177]: src/shared/bap.c:bap_queue_req() req 0x606000074b40 (op 0x05) queue 0x60300001eca0
bluetoothd[100177]: src/shared/bap.c:bap_queue_req() req 0x606000074ba0 (op 0x05) queue 0x603000096070
bluetoothd[100177]: src/shared/bap.c:bap_queue_req() req 0x606000074de0 (op 0x08) queue 0x60300001eca0
bluetoothd[100177]: profiles/audio/bap.c:pac_removed() pac 0x608000015620
bluetoothd[100177]: profiles/audio/media.c:proxy_removed_cb() Proxy removed - removing endpoint: /MediaEndpoint/BAPSource/lc3
bluetoothd[100177]: Endpoint unregistered: sender=:1.464 path=/MediaEndpoint/BAPSource/lc3
bluetoothd[100177]: profiles/audio/media.c:media_endpoint_destroy() sender=:1.464 path=/MediaEndpoint/BAPSource/lc3
bluetoothd[100177]: src/shared/bap.c:bap_queue_req() req 0x6060000750e0 (op 0x08) queue 0x603000096970
bluetoothd[100177]: profiles/audio/bap.c:pac_removed() pac 0x608000015720
bluetoothd[100177]: profiles/audio/media.c:client_disconnect_cb() Client disconnected
bluetoothd[100177]: profiles/audio/media.c:proxy_removed_cb() Proxy removed - removing endpoint: /MediaEndpoint/BAPSink/lc3
bluetoothd[100177]: Endpoint unregistered: sender=:1.464 path=/MediaEndpoint/BAPSink/lc3
bluetoothd[100177]: profiles/audio/media.c:media_endpoint_destroy() sender=:1.464 path=/MediaEndpoint/BAPSink/lc3
bluetoothd[100177]: profiles/audio/bap.c:pac_removed() pac 0x6080000159a0
bluetoothd[100177]: profiles/audio/media.c:proxy_removed_cb() Proxy removed - removing endpoint: /MediaEndpoint/BAPSource/lc3
bluetoothd[100177]: Endpoint unregistered: sender=:1.464 path=/MediaEndpoint/BAPSource/lc3
bluetoothd[100177]: profiles/audio/media.c:media_endpoint_destroy() sender=:1.464 path=/MediaEndpoint/BAPSource/lc3
bluetoothd[100177]: profiles/audio/bap.c:pac_removed() pac 0x608000015aa0
bluetoothd[100177]: profiles/audio/media.c:client_disconnect_cb() Client disconnected
bluetoothd[100177]: src/shared/bap.c:bap_process_queue()
bluetoothd[100177]: src/shared/bap.c:bap_send() req 0x606000074b40
bluetoothd[100177]: src/shared/att.c:can_read_data() (chan 0x607000032d00) ATT PDU received: 0x1b
bluetoothd[100177]: src/shared/bap.c:bap_process_queue()
bluetoothd[100177]: src/shared/bap.c:bap_send() req 0x606000074de0
bluetoothd[100177]: src/shared/att.c:can_read_data() (chan 0x607000032bb0) ATT PDU received: 0x1b
bluetoothd[100177]: src/shared/bap.c:bap_ep_set_status() ASE status: ep 0x60300005a490 id 0x03 handle 0x004b state disabling len 7
bluetoothd[100177]: src/shared/bap.c:ep_status_metadata() CIS 0x00 CIG 0x00 metadata len 4
bluetoothd[100177]: src/shared/bap.c:bap_stream_state_changed() stream 0x60b000122030 dir 0x02: streaming -> disabling
bluetoothd[100177]: src/shared/bap.c:bap_stream_io_detach() stream 0x60b000122030
bluetoothd[100177]: profiles/audio/bap.c:bap_state() stream 0x60b000122030: streaming(4) -> disabling(5)
bluetoothd[100177]: src/shared/att.c:can_read_data() (chan 0x607000032d00) ATT PDU received: 0x1b
bluetoothd[100177]: src/shared/bap.c:bap_ep_set_status() ASE status: ep 0x60300005a250 id 0x01 handle 0x0045 state qos len 15
bluetoothd[100177]: src/shared/bap.c:ep_status_qos() CIG 0x00 CIS 0x00 interval 10000 framing 0x00 phy 0x02 rtn 5 latency 100 pd 40000
bluetoothd[100177]: src/shared/bap.c:bap_stream_state_changed() stream 0x60b000122a80 dir 0x01: streaming -> qos
bluetoothd[100177]: src/shared/bap.c:bap_stream_io_detach() stream 0x60b000122a80
bluetoothd[100177]: src/shared/bap.c:stream_io_free() fd 57
bluetoothd[100177]: profiles/audio/bap.c:bap_state() stream 0x60b000122a80: streaming(4) -> qos(2)
bluetoothd[100177]: profiles/audio/bap.c:bap_create_io() ep 0x60b000121530 stream 0x60b000122a80 defer true
bluetoothd[100177]: src/shared/bap.c:bt_bap_stream_io_get_qos() in 0x60b000122068 out 0x60b000122ab8
bluetoothd[100177]: profiles/audio/bap.c:bap_io_disconnected() ep 0x60b000120f00 recreate false
bluetoothd[100177]: profiles/audio/bap.c:bap_io_close() ep 0x60b000120f00
bluetoothd[100177]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x001c
bluetoothd[100177]: src/adapter.c:new_conn_param() hci0 28:3D:C2:4A:7D:2A (1) min 0x0006 max 0x0018 latency 0x0000 timeout 0x01f4
bluetoothd[100177]: src/adapter.c:store_conn_param()
bluetoothd[100177]: src/shared/att.c:can_read_data() (chan 0x607000032bb0) ATT PDU received: 0x1b
bluetoothd[100177]: src/shared/bap.c:bap_process_queue()
bluetoothd[100177]: src/shared/att.c:can_read_data() (chan 0x607000032bb0) ATT PDU received: 0x1b
bluetoothd[100177]: src/shared/bap.c:bap_ep_set_status() ASE status: ep 0x60300005a250 id 0x01 handle 0x0045 state releasing len 0
bluetoothd[100177]: src/shared/bap.c:bap_stream_state_changed() stream 0x60b000122a80 dir 0x01: qos -> releasing
bluetoothd[100177]: profiles/audio/bap.c:bap_state() stream 0x60b000122a80: qos(2) -> releasing(6)
bluetoothd[100177]: src/shared/att.c:can_read_data() (chan 0x607000032d00) ATT PDU received: 0x1b
bluetoothd[100177]: src/shared/bap.c:bap_ep_set_status() ASE status: ep 0x60300005a490 id 0x03 handle 0x004b state releasing len 0
bluetoothd[100177]: src/shared/bap.c:bap_stream_state_changed() stream 0x60b000122030 dir 0x02: disabling -> releasing
[... 50 lines cut ...]
bluetoothd[100177]:   0x0c107fffab20: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fd
bluetoothd[100177]:   0x0c107fffab30: fa fa fa fa fd fd fd fd fd fd fd fd fd fd fd fa
bluetoothd[100177]: Shadow byte legend (one shadow byte represents 8 application bytes):
bluetoothd[100177]:   Addressable:           00
bluetoothd[100177]:   Partially addressable: 01 02 03 04 05 06 07
bluetoothd[100177]:   Heap left redzone:       fa
bluetoothd[100177]:   Freed heap region:       fd
bluetoothd[100177]:   Stack left redzone:      f1
bluetoothd[100177]:   Stack mid redzone:       f2
bluetoothd[100177]:   Stack right redzone:     f3
bluetoothd[100177]:   Stack after return:      f5
bluetoothd[100177]:   Stack use after scope:   f8
bluetoothd[100177]:   Global redzone:          f9
bluetoothd[100177]:   Global init order:       f6
bluetoothd[100177]:   Poisoned by user:        f7
bluetoothd[100177]:   Container overflow:      fc
bluetoothd[100177]:   Array cookie:            ac
bluetoothd[100177]:   Intra object redzone:    bb
bluetoothd[100177]:   ASan internal:           fe
bluetoothd[100177]:   Left alloca redzone:     ca
bluetoothd[100177]:   Right alloca redzone:    cb
bluetoothd[100177]: ==100177==ABORTING
```

</details>

Vudentz, 2023-01-19:
@pv will take a look, are there any steps to reproduce locally?

pv, 2023-01-20:
What I was doing:

1. Connect BlueZ (master) + Pipewire (master) via BAP to a device
2. Play audio
3. Kill Pipewire processes (eg SIGINT) while there is audio stream to the device
4. bluetoothd hits the ASAN crash

Step 1 is a bit flaky right now (I'll need to figure it out a bit later), but what I need to do on top of bluetoothd BAP experimental flags:  (i) in `/etc/bluetooth/main.conf` set `ControllerMode = le` (media.c seems to do only SelectProperties but no SetProperties otherwise), (ii) create file `~/.config/wireplumber/bluetooth.lua.d/51-bap-only.lua` containing the line `bluez_monitor.properties["bluez5.codecs"] = "[ lc3 ]"` so it doesn't try to register A2DP and fail with LE only, and restart bluetoothd+all pipewire processes. (iii) bluetoothctl shows no transports on first device connect, but it works on second try.

Vudentz, 2023-01-20:
@pv try with the following change:

```
diff --git a/src/shared/bap.c b/src/shared/bap.c
index db7def7999b7..610e039facd4 100644
--- a/src/shared/bap.c
+++ b/src/shared/bap.c
@@ -2478,8 +2478,10 @@ static void remove_streams(void *data, void *user_data)
        struct bt_bap_stream *stream;
 
        stream = queue_remove_if(bap->streams, match_stream_lpac, pac);
-       if (stream)
+       if (stream) {
                bt_bap_stream_release(stream, NULL, NULL);
+               bap_stream_detach(stream);
+       }
 }
 
 bool bt_bap_remove_pac(struct bt_bap_pac *pac)
```

pv, 2023-01-21:
That fixes the crash on Pipewire killed.

There's however another ASAN crash when Pipewire is restarted again after being killed:

<details>
```
bluetoothd[32623]: src/profile.c:register_profile() sender :1.329
bluetoothd[32623]: src/profile.c:create_ext() Created "Hands-Free Voice gateway"
bluetoothd[32623]: src/profile.c:ext_adapter_probe() "Hands-Free Voice gateway" probed
bluetoothd[32623]: src/profile.c:ext_start_servers() Hands-Free Voice gateway listening on chan 13
bluetoothd[32623]: src/adapter.c:adapter_service_add() /org/bluez/hci1
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Adding record with handle 0x1001b
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000003-0000-1000-8000-00805f9
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000100-0000-1000-8000-00805f9
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00001002-0000-1000-8000-00805f9
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 0000111e-0000-1000-8000-00805f9
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 0000111f-0000-1000-8000-00805f9
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00001203-0000-1000-8000-00805f9
bluetoothd[32623]: src/adapter.c:adapter_service_insert() /org/bluez/hci1
bluetoothd[32623]: src/adapter.c:add_uuid() sending add uuid command for index 1
bluetoothd[32623]: src/profile.c:ext_adapter_probe() "Hands-Free Voice gateway" probed
bluetoothd[32623]: src/profile.c:ext_start_servers() Hands-Free Voice gateway listening on chan 13
bluetoothd[32623]: src/adapter.c:adapter_service_add() /org/bluez/hci0
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Adding record with handle 0x1001c
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000003-0000-1000-8000-00805f9
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000100-0000-1000-8000-00805f9
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00001002-0000-1000-8000-00805f9
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 0000111e-0000-1000-8000-00805f9
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 0000111f-0000-1000-8000-00805f9
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00001203-0000-1000-8000-00805f9
bluetoothd[32623]: src/adapter.c:adapter_service_insert() /org/bluez/hci0
bluetoothd[32623]: src/adapter.c:add_uuid() sending add uuid command for index 0
bluetoothd[32623]: src/profile.c:ext_device_probe() Hands-Free Voice gateway probed with UUID 0000111e-0000-1000-8000-00805f9b34fb
bluetoothd[32623]: src/service.c:change_state() 0x604000065510: device 5C:51:4F:BA:FF:F3 profile Hands-Free Voice gateway state changed: unavailable -> disconnected (0)
bluetoothd[32623]: src/profile.c:ext_device_probe() Hands-Free Voice gateway probed with UUID 0000111e-0000-1000-8000-00805f9b34fb
bluetoothd[32623]: src/service.c:change_state() 0x604000065550: device 70:1A:B8:98:FF:A2 profile Hands-Free Voice gateway state changed: unavailable -> disconnected (0)
bluetoothd[32623]: src/profile.c:ext_device_probe() Hands-Free Voice gateway probed with UUID 0000111e-0000-1000-8000-00805f9b34fb
bluetoothd[32623]: src/service.c:change_state() 0x604000065590: device 00:1B:66:C9:77:E2 profile Hands-Free Voice gateway state changed: unavailable -> disconnected (0)
bluetoothd[32623]: src/profile.c:ext_device_probe() Hands-Free Voice gateway probed with UUID 0000111e-0000-1000-8000-00805f9b34fb
bluetoothd[32623]: src/service.c:change_state() 0x6040000655d0: device E0:67:81:3B:34:C3 profile Hands-Free Voice gateway state changed: unavailable -> disconnected (0)
bluetoothd[32623]: src/profile.c:register_profile() sender :1.329
bluetoothd[32623]: src/profile.c:create_ext() Created "Hands-Free unit"
bluetoothd[32623]: src/profile.c:ext_adapter_probe() "Hands-Free unit" probed
bluetoothd[32623]: src/profile.c:ext_start_servers() Hands-Free unit listening on chan 7
bluetoothd[32623]: src/adapter.c:adapter_service_add() /org/bluez/hci1
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Adding record with handle 0x1001d
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000003-0000-1000-8000-00805f9
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000100-0000-1000-8000-00805f9
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00001002-0000-1000-8000-00805f9
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 0000111e-0000-1000-8000-00805f9
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00001203-0000-1000-8000-00805f9
bluetoothd[32623]: src/adapter.c:adapter_service_insert() /org/bluez/hci1
bluetoothd[32623]: src/adapter.c:add_uuid() sending add uuid command for index 1
bluetoothd[32623]: src/profile.c:ext_adapter_probe() "Hands-Free unit" probed
bluetoothd[32623]: src/profile.c:ext_start_servers() Hands-Free unit listening on chan 7
bluetoothd[32623]: src/adapter.c:adapter_service_add() /org/bluez/hci0
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Adding record with handle 0x1001e
bluetoothd[32623]: src/sdpd-service.c:add_record_to_server() Record pattern UUID 00000003-0000-1000-8000-00805f9
[... cut ...]
bluetoothd[32623]:   Global init order:       f6
bluetoothd[32623]:   Poisoned by user:        f7
bluetoothd[32623]:   Container overflow:      fc
bluetoothd[32623]:   Array cookie:            ac
bluetoothd[32623]:   Intra object redzone:    bb
bluetoothd[32623]:   ASan internal:           fe
bluetoothd[32623]:   Left alloca redzone:     ca
bluetoothd[32623]:   Right alloca redzone:    cb
bluetoothd[32623]: ==32623==ABORTING
systemd[1]: bluetooth.service: Main process exited, code=exited, status=1/FAILURE
```
</details>

Vudentz, 2023-01-23:
> Thanks, that fixes the crash on Pipewire killed.
> 
> There's however another ASAN crash when Pipewire is restarted again after being killed:
> 
> The device (and PW) have only 1 source and 1 sink BAP endpoint, so looks like it fails to remove old ones or creates new ones when it shouldn't.

Check with the following set:

https://patchwork.kernel.org/project/bluetooth/list/?series=714905

Looks like we were not cleaning up the ep->stream properly when we destroy the stream.

pv, 2023-01-24:
Thanks, applied on top of a1736d8990ff56b. There's now a different ASAN crash after pipewire is restarted after being killed.

<details>

```
bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpointLE/BAPSink/lc3, iface: org.bluez.MediaEndpoint1
bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpointLE/BAPSource/lc3, iface: org.bluez.MediaEndpoint1
bluetoothd[31951]: profiles/audio/media.c:bap_debug() Length 19
bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #0: len 3 type 1
bluetoothd[31951]: profiles/audio/media.c:bap_debug()   95 00                                            ..
bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #1: len 2 type 2
bluetoothd[31951]: profiles/audio/media.c:bap_debug()   03                                               .
bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #2: len 2 type 3
bluetoothd[31951]: profiles/audio/media.c:bap_debug()   03                                               .
bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #3: len 5 type 4
bluetoothd[31951]: profiles/audio/media.c:bap_debug()   14 00 90 01                                      ....
bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #4: len 2 type 5
bluetoothd[31951]: profiles/audio/media.c:bap_debug()   02                                               .
bluetoothd[31951]: profiles/audio/bap.c:pac_added() pac 0x60800001ee20
bluetoothd[31951]: profiles/audio/bap.c:pac_found() lpac 0x60800001ee20 rpac 0x6080000167a0
bluetoothd[31951]: profiles/audio/bap.c:ep_register() ep 0x60b0000fa230 lpac 0x60800001ee20 rpac 0x6080000167a0 path /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_source1
bluetoothd[31951]: profiles/audio/media.c:endpoint_init_pac() PAC :1.343:/MediaEndpointLE/BAPSink/lc3 registered
bluetoothd[31951]: Endpoint registered: sender=:1.343 path=/MediaEndpointLE/BAPSink/lc3
bluetoothd[31951]: profiles/audio/media.c:bap_debug() Length 19
bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #0: len 3 type 1
bluetoothd[31951]: profiles/audio/media.c:bap_debug()   95 00                                            ..
bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #1: len 2 type 2
bluetoothd[31951]: profiles/audio/media.c:bap_debug()   03                                               .
bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #2: len 2 type 3
bluetoothd[31951]: profiles/audio/media.c:bap_debug()   03                                               .
bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #3: len 5 type 4
bluetoothd[31951]: profiles/audio/media.c:bap_debug()   14 00 90 01                                      ....
bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #4: len 2 type 5
bluetoothd[31951]: profiles/audio/media.c:bap_debug()   02                                               .
bluetoothd[31951]: profiles/audio/bap.c:pac_added() pac 0x60800001ef20
bluetoothd[31951]: profiles/audio/bap.c:pac_found() lpac 0x60800001ee20 rpac 0x6080000167a0
bluetoothd[31951]: profiles/audio/bap.c:ep_register() ep 0x60b0000fb4c0 lpac 0x60800001ee20 rpac 0x6080000167a0 path /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_source2
bluetoothd[31951]: profiles/audio/media.c:media_endpoint_async_call() Calling SelectProperties: name = :1.343 path = /MediaEndpointLE/BAPSink/lc3
bluetoothd[31951]: profiles/audio/bap.c:pac_found() lpac 0x60800001ef20 rpac 0x6080000166a0
bluetoothd[31951]: profiles/audio/bap.c:ep_register() ep 0x60b0000fbaf0 lpac 0x60800001ef20 rpac 0x6080000166a0 path /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_sink1
bluetoothd[31951]: profiles/audio/media.c:endpoint_init_pac() PAC :1.343:/MediaEndpointLE/BAPSource/lc3 registered
bluetoothd[31951]: Endpoint registered: sender=:1.343 path=/MediaEndpointLE/BAPSource/lc3
bluetoothd[31951]: profiles/audio/media.c:client_ready_cb() Media application registered: :1.343:/MediaEndpointLE
bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSource/ldac, iface: org.bluez.MediaEndpoint1
bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSink/aptx_hd, iface: org.bluez.MediaEndpoint1
bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSource/aptx_hd, iface: org.bluez.MediaEndpoint1
bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSink/aptx, iface: org.bluez.MediaEndpoint1
bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSource/aptx, iface: org.bluez.MediaEndpoint1
bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSink/aac, iface: org.bluez.MediaEndpoint1
bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSource/aac, iface: org.bluez.MediaEndpoint1
bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSink/lc3plus_hr, iface: org.bluez.MediaEndpoint1
bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSource/lc3plus_hr, iface: org.bluez.MediaEndpoint1
bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSink/sbc, iface: org.bluez.MediaEndpoint1
bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSource/sbc, iface: org.bluez.MediaEndpoint1
[... cut ...]
[NEW] Endpoint /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_source3 
[NEW] Endpoint /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_sink2 
[NEW] Endpoint /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_source4 
[NEW] Endpoint /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_sink3 
```
when pipewire is started again. When the device is connected before killing pipewire, there are only
```
[NEW] Endpoint /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_source0 
[NEW] Endpoint /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_sink0 
```

Vudentz, 2023-01-25:
> Thanks, applied on top of [a1736d8](https://github.com/bluez/bluez/commit/a1736d8990ff56bba453ff81a25156316bdd118f). There's now a different ASAN crash after pipewire is restarted after being killed.
> 
> ```
> bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpointLE/BAPSink/lc3, iface: org.bluez.MediaEndpoint1
> bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpointLE/BAPSource/lc3, iface: org.bluez.MediaEndpoint1
> bluetoothd[31951]: profiles/audio/media.c:bap_debug() Length 19
> bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #0: len 3 type 1
> bluetoothd[31951]: profiles/audio/media.c:bap_debug()   95 00                                            ..
> bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #1: len 2 type 2
> bluetoothd[31951]: profiles/audio/media.c:bap_debug()   03                                               .
> bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #2: len 2 type 3
> bluetoothd[31951]: profiles/audio/media.c:bap_debug()   03                                               .
> bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #3: len 5 type 4
> bluetoothd[31951]: profiles/audio/media.c:bap_debug()   14 00 90 01                                      ....
> bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #4: len 2 type 5
> bluetoothd[31951]: profiles/audio/media.c:bap_debug()   02                                               .
> bluetoothd[31951]: profiles/audio/bap.c:pac_added() pac 0x60800001ee20
> bluetoothd[31951]: profiles/audio/bap.c:pac_found() lpac 0x60800001ee20 rpac 0x6080000167a0
> bluetoothd[31951]: profiles/audio/bap.c:ep_register() ep 0x60b0000fa230 lpac 0x60800001ee20 rpac 0x6080000167a0 path /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_source1
> bluetoothd[31951]: profiles/audio/media.c:endpoint_init_pac() PAC :1.343:/MediaEndpointLE/BAPSink/lc3 registered
> bluetoothd[31951]: Endpoint registered: sender=:1.343 path=/MediaEndpointLE/BAPSink/lc3
> bluetoothd[31951]: profiles/audio/media.c:bap_debug() Length 19
> bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #0: len 3 type 1
> bluetoothd[31951]: profiles/audio/media.c:bap_debug()   95 00                                            ..
> bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #1: len 2 type 2
> bluetoothd[31951]: profiles/audio/media.c:bap_debug()   03                                               .
> bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #2: len 2 type 3
> bluetoothd[31951]: profiles/audio/media.c:bap_debug()   03                                               .
> bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #3: len 5 type 4
> bluetoothd[31951]: profiles/audio/media.c:bap_debug()   14 00 90 01                                      ....
> bluetoothd[31951]: profiles/audio/media.c:bap_debug() CC #4: len 2 type 5
> bluetoothd[31951]: profiles/audio/media.c:bap_debug()   02                                               .
> bluetoothd[31951]: profiles/audio/bap.c:pac_added() pac 0x60800001ef20
> bluetoothd[31951]: profiles/audio/bap.c:pac_found() lpac 0x60800001ee20 rpac 0x6080000167a0
> bluetoothd[31951]: profiles/audio/bap.c:ep_register() ep 0x60b0000fb4c0 lpac 0x60800001ee20 rpac 0x6080000167a0 path /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_source2
> bluetoothd[31951]: profiles/audio/media.c:media_endpoint_async_call() Calling SelectProperties: name = :1.343 path = /MediaEndpointLE/BAPSink/lc3
> bluetoothd[31951]: profiles/audio/bap.c:pac_found() lpac 0x60800001ef20 rpac 0x6080000166a0
> bluetoothd[31951]: profiles/audio/bap.c:ep_register() ep 0x60b0000fbaf0 lpac 0x60800001ef20 rpac 0x6080000166a0 path /org/bluez/hci0/dev_28_3D_C2_4A_7D_2A/pac_sink1
> bluetoothd[31951]: profiles/audio/media.c:endpoint_init_pac() PAC :1.343:/MediaEndpointLE/BAPSource/lc3 registered
> bluetoothd[31951]: Endpoint registered: sender=:1.343 path=/MediaEndpointLE/BAPSource/lc3
> bluetoothd[31951]: profiles/audio/media.c:client_ready_cb() Media application registered: :1.343:/MediaEndpointLE
> bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSource/ldac, iface: org.bluez.MediaEndpoint1
> bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSink/aptx_hd, iface: org.bluez.MediaEndpoint1
> bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSource/aptx_hd, iface: org.bluez.MediaEndpoint1
> bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSink/aptx, iface: org.bluez.MediaEndpoint1
> bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSource/aptx, iface: org.bluez.MediaEndpoint1
> bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSink/aac, iface: org.bluez.MediaEndpoint1
> bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSource/aac, iface: org.bluez.MediaEndpoint1
> bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSink/lc3plus_hr, iface: org.bluez.MediaEndpoint1
> bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSource/lc3plus_hr, iface: org.bluez.MediaEndpoint1
> bluetoothd[31951]: profiles/audio/media.c:proxy_added_cb() Proxy added: /MediaEndpoint/A2DPSink/sbc, iface: org.bluez.MediaEndpoint1
[... cut ...]
This should take care of the crashes but there is something something not quite right when I restart bluetoothctl -e it doesn't recreate the transports just the endpoints:

[BlueZ 5.66]# endpoint.list 
Endpoint /org/bluez/hci0/dev_00_AA_01_01_00_02/pac_source1 
Endpoint /org/bluez/hci0/dev_00_AA_01_01_00_02/pac_sink0 
[BlueZ 5.66]# transport.list

It probably because the code doesn't automatically does the stream configuration if the service is not connecting, but that perhaps can be changed if we detect there was a change in the endpoints, etc.

pv, 2023-01-25:
Thanks, applied v2 on top a1736d8990ff56bb. Fixes the use-after-free. A different ASAN crash now occurs when restarting Pipewire.

<details>

```
bluetoothd[19793]: src/device.c:device_connect_le() Connection attempt to: 28:3D:C2:4A:7D:2A
bluetoothd[19793]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000b
bluetoothd[19793]: src/adapter.c:connected_callback() hci0 device 28:3D:C2:4A:7D:2A connected eir_len 0
bluetoothd[19793]: src/device.c:device_attach_att() Elevating security level since LTK is available
bluetoothd[19793]: attrib/gattrib.c:g_attrib_ref() 0x6070000ab0b0: g_attrib_ref=1
bluetoothd[19793]: src/device.c:device_accept_gatt_profiles() initiator true
bluetoothd[19793]: profiles/audio/bap.c:bap_accept() 28:3D:C2:4A:7D:2A
bluetoothd[19793]: profiles/audio/bap.c:bap_attached() 0x60e000001620
bluetoothd[19793]: src/shared/bap.c:bap_endpoint_attach() ASE handle 0x0046
bluetoothd[19793]: src/shared/bap.c:bap_endpoint_attach() ASE handle 0x0049
bluetoothd[19793]: src/shared/bap.c:bap_endpoint_attach() ASE handle 0x004c
bluetoothd[19793]: src/shared/bap.c:bap_cp_attach() ASE CP handle 0x0043
bluetoothd[19793]: src/service.c:change_state() 0x604000013a10: device 28:3D:C2:4A:7D:2A profile bap state changed: disconnected -> connected (0)
bluetoothd[19793]: profiles/gap/gas.c:gap_accept() GAP profile accept (28:3D:C2:4A:7D:2A)
bluetoothd[19793]: profiles/gap/gas.c:handle_characteristic() Unsupported characteristic: 00002aa6-0000-1000-8000-00805f9b34fb
bluetoothd[19793]: src/service.c:change_state() 0x604000034b10: device 28:3D:C2:4A:7D:2A profile gap-profile state changed: disconnected -> connected (0)
bluetoothd[19793]: profiles/deviceinfo/deviceinfo.c:deviceinfo_accept() deviceinfo profile accept (28:3D:C2:4A:7D:2A)
bluetoothd[19793]: profiles/deviceinfo/deviceinfo.c:handle_characteristic() Unsupported characteristic: 00002a29-0000-1000-8000-00805f9b34fb
bluetoothd[19793]: profiles/deviceinfo/deviceinfo.c:handle_characteristic() Unsupported characteristic: 00002a24-0000-1000-8000-00805f9b34fb
bluetoothd[19793]: profiles/deviceinfo/deviceinfo.c:handle_characteristic() Unsupported characteristic: 00002a25-0000-1000-8000-00805f9b34fb
bluetoothd[19793]: profiles/deviceinfo/deviceinfo.c:handle_characteristic() Unsupported characteristic: 00002a27-0000-1000-8000-00805f9b34fb
bluetoothd[19793]: profiles/deviceinfo/deviceinfo.c:handle_characteristic() Unsupported characteristic: 00002a26-0000-1000-8000-00805f9b34fb
bluetoothd[19793]: profiles/deviceinfo/deviceinfo.c:handle_characteristic() Unsupported characteristic: 00002a28-0000-1000-8000-00805f9b34fb
bluetoothd[19793]: profiles/deviceinfo/deviceinfo.c:handle_characteristic() Unsupported characteristic: 00002a23-0000-1000-8000-00805f9b34fb
bluetoothd[19793]: profiles/deviceinfo/deviceinfo.c:handle_characteristic() Unsupported characteristic: 00002a2a-0000-1000-8000-00805f9b34fb
bluetoothd[19793]: src/service.c:change_state() 0x604000034d90: device 28:3D:C2:4A:7D:2A profile deviceinfo state changed: disconnected -> connected (0)
bluetoothd[19793]: profiles/battery/battery.c:batt_accept() BATT profile accept (28:3D:C2:4A:7D:2A)
bluetoothd[19793]: profiles/battery/battery.c:foreach_batt_service() More than one BATT service exists for this device
bluetoothd[19793]: src/service.c:change_state() 0x604000034f10: device 28:3D:C2:4A:7D:2A profile batt-profile state changed: disconnected -> connected (0)
bluetoothd[19793]: src/gatt-client.c:btd_gatt_client_connected() Device connected.
bluetoothd[19793]: src/gatt-client.c:btd_gatt_client_eatt_connect() Connection attempt to: 28:3D:C2:4A:7D:2A defer true
bluetoothd[19793]: src/gatt-client.c:btd_gatt_client_eatt_connect() Connection attempt to: 28:3D:C2:4A:7D:2A defer false
bluetoothd[19793]: src/shared/att.c:can_read_data() (chan 0x6070000ab120) ATT PDU received: 0x02
bluetoothd[19793]: src/shared/gatt-server.c:exchange_mtu_cb() MTU exchange complete, with MTU: 512
bluetoothd[19793]: src/shared/att.c:bt_att_attach_chan() Channel 0x6070000ab2e0 attached
bluetoothd[19793]: src/device.c:device_attach_att() EATT channel connected
bluetoothd[19793]: src/shared/att.c:can_read_data() (chan 0x6070000ab2e0) ATT PDU received: 0x08
bluetoothd[19793]: src/shared/gatt-server.c:read_by_type_cb() Read By Type - start: 0x0001 end: 0xffff
bluetoothd[19793]: src/gatt-database.c:gap_car_read_cb() GAP Central Address Resolution read request
bluetoothd[19793]: src/shared/att.c:bt_att_attach_chan() Channel 0x6070000ab430 attached
bluetoothd[19793]: src/device.c:device_attach_att() EATT channel connected
bluetoothd[19793]: src/shared/att.c:can_read_data() (chan 0x6070000ab430) ATT PDU received: 0x06
bluetoothd[19793]: src/shared/gatt-server.c:find_by_type_val_cb() Find By Type Value - start: 0x0001 end: 0xffff uuid: 0x2800
bluetoothd[19793]: src/shared/att.c:can_read_data() (chan 0x6070000ab120) ATT PDU received: 0x06
bluetoothd[19793]: src/shared/gatt-server.c:find_by_type_val_cb() Find By Type Value - start: 0x0001 end: 0xffff uuid: 0x2800
bluetoothd[19793]: [0x61100000acc0] src/shared/gatt-client.c:exchange_mtu_cb() MTU exchange complete, with MTU: 512
bluetoothd[19793]: src/shared/bap.c:bap_ep_set_status() ASE status: ep 0x60300007a7a0 id 0x01 handle 0x0045 state idle len 0
bluetoothd[19793]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x001c
bluetoothd[19793]: src/adapter.c:new_conn_param() hci0 28:3D:C2:4A:7D:2A (1) min 0x0006 max 0x0018 latency 0x0000 timeout 0x01f4
[... cut ...]
bluetoothd[19793]:   Stack use after scope:   f8
bluetoothd[19793]:   Global redzone:          f9
bluetoothd[19793]:   Global init order:       f6
bluetoothd[19793]:   Poisoned by user:        f7
bluetoothd[19793]:   Container overflow:      fc
bluetoothd[19793]:   Array cookie:            ac
bluetoothd[19793]:   Intra object redzone:    bb
bluetoothd[19793]:   ASan internal:           fe
bluetoothd[19793]:   Left alloca redzone:     ca
bluetoothd[19793]:   Right alloca redzone:    cb
bluetoothd[19793]: ==19793==ABORTING
```

</details>

Vudentz, 2023-01-26:
> Thanks, applied v2 on top [a1736d8](https://github.com/bluez/bluez/commit/a1736d8990ff56bba453ff81a25156316bdd118f). Fixes the use-after-free. A different ASAN crash now occurs when restarting Pipewire.

I managed to reproduce this problem, for some reason we end up enumerating more endpoints than when we disconnect:

```
[BlueZ 5.66]# endpoint.list 
Endpoint /org/bluez/hci0/dev_00_AA_01_01_00_02/pac_source0 
Endpoint /org/bluez/hci0/dev_00_AA_01_01_00_02/pac_source1 
Endpoint /org/bluez/hci0/dev_00_AA_01_01_00_02/pac_sink0
```

Most likely the cleanup is leaving something behind, anyway I managed to fix the crash since we were not checking if the requests grow bigger than the actual MTU, anyway there is still some work to do to resolve all the problems when the sound server restarts while a device is connected.

pv, 2023-01-27:
Thanks, applied v3. No further ASAN crashes in the same test case.

closed 2023-01-28 by pv