bluez/bluez #459: [Question] Which Kernel options are required?

kuraga, 2023-01-21:
Beyond Linux® From Scratch 11.2 [says](https://www.linuxfromscratch.org/blfs/view/stable/general/bluez.html) that BlueZ 5.65 requires such Kernel options:

```
 [*] Configure standard kernel features (expert users)               [CONFIG_EXPERT]
    [*] Enable timerfd() system call               [CONFIG_TIMERFD]
    [*] Enable eventfd() system call               [CONFIG_EVENTFD]

[*] Networking support --->                [CONFIG_NET]
  <*/M> Bluetooth subsystem support --->    [CONFIG_BT]
    <*/M> RFCOMM protocol support          [CONFIG_BT_RFCOMM]
    [*]   RFCOMM TTY support               [CONFIG_BT_RFCOMM_TTY]
    <*/M> BNEP protocol support            [CONFIG_BT_BNEP]
    [*]   Multicast filter support         [CONFIG_BT_BNEP_MC_FILTER]
    [*]   Protocol filter support          [CONFIG_BT_BNEP_PROTO_FILTER]
    <*/M> HIDP protocol support            [CONFIG_BT_HIDP]
        Bluetooth device drivers --->
          (Select the appropriate drivers for your Bluetooth hardware)

   <*/M> RF switch subsystem support ----   [CONFIG_RFKILL]

-*- Cryptographic API --->
   <*/M*> User-space cryptographic algorithm configuration         [CONFIG_CRYPTO_USER]
   <*/M*> User-space interface for hash algorithms                 [CONFIG_CRYPTO_USER_API_HASH]
   <*/M*> User-space interface for symmetric key cipher algorithms [CONFIG_CRYPTO_USER_API_SKCIPHER]
   <*/M*> MD5 digest algorithm                                     [CONFIG_CRYPTO_MD5]
   <*/M*> SHA1 digest algorithm                                    [CONFIG_CRYPTO_SHA1]

Security Options --->
  <*/M*> Diffie-Hellman operations on retained keys [CONFIG_KEY_DH_OPERATIONS]
```

Some distributions [refer to this](https://gitweb.gentoo.org/repo/gentoo.git/tree/net-wireless/bluez/bluez-5.65-r1.ebuild#n81).

But I don't see these Kernel options in any `master`'s documentation pages:

| Kernel option description                  | Kernel option              |
|--------------------------------------------|----------------------------|
| Enable timerfd() system call               | `CONFIG_TIMERFD`           |
| Enable eventfd() system call               | `CONFIG_EVENTFD`           |
| HIDP protocol support                      | `CONFIG_BT_HIDP`           |
| MD5 digest algorithm                       | `CONFIG_CRYPTO_MD5`        |
| SHA1 digest algorithm                      | `CONFIG_CRYPTO_SHA1`       |
| Diffie-Hellman operations on retained keys | `CONFIG_KEY_DH_OPERATIONS` |

Are they required?

Thanks!

bgix, 2023-01-23:
This won't address your Headphone issue, but the following kernel crypto functions should also be enabled:
	CONFIG_CRYPTO_USER_API
	CONFIG_CRYPTO_USER_API_AEAD
	CONFIG_CRYPTO_USER_API_HASH
	CONFIG_CRYPTO_USER_API_SKCIPHER

	CONFIG_CRYPTO_AES
	CONFIG_CRYPTO_CCM
	CONFIG_CRYPTO_AEAD
	CONFIG_CRYPTO_CMAC

kuraga, 2023-01-23:
@bgix, thanks!

And what about (second) table's options?

P.S. Which "my Headphone issue"?

bgix, 2023-01-24:
I'm sorry... I thought this might be related to an Audio issue that was recently discussed.  If it isn't, then my bad.

kuraga, 2023-01-24:
@bgix , it's ok :) But what about options in the table?

bgix, 2023-01-24:
Your second table lays outside my area of expertise.  The kernel options I added to your list have been required since Bluetooth Mesh support was added to BlueZ, but I do not know what or if the other options you list are required.

kuraga, 2023-01-26:
@bgix , thanks! Let's ping somebody with expertise? :)

kuraga, 2023-01-29:
@bluez, @mabhay125, @Vudentz, please?

Vudentz, 2023-01-29:
@kuraga the config options are explained in:

https://github.com/bluez/bluez/blob/master/doc/test-runner.txt

kuraga, 2023-01-29:
@Vudentz , thanks!

But are all parts of BlueZ tested? Because not [all options for Mesh](https://github.com/bluez/bluez/blob/5.66/README#L77) are even listed.

Vudentz, 2023-01-29:
> @Vudentz , thanks!
> 
> But are all parts of BlueZ tested? Because not [all options for Mesh](https://github.com/bluez/bluez/blob/5.66/README#L77) are even listed.

Right, it is inconsistent since we are not listing the crypto requirements although we do seem to cover that on the tester.config itself which is what is used by our CI:

https://github.com/bluez/bluez/blob/master/doc/tester.config 

@bgix @tedd-an we should probably try to make it more consistent what are the kernel crypto requirements.

closed 2025-08-13 by github-actions[bot]