bluez/bluez #460: AcquireWrite / AcquireNotify sockets aren't closed in Server mode

beku-epitome, 2023-01-24:
I am trying to use the GattCharacteristic1 AcquireWrite / AcquireNotify API to create a GATT control point characteristic on the server side, as it was [intended](https://github.com/bluez/bluez/issues/199#issuecomment-916616785). I create a socket pair, use one side in the server, pass the other to BlueZ, and the data transfer itself works fine. The problem is, BlueZ doesn't close this socket when the client device disconnects, or another client starts communicating with it, thereby defeating the whole purpose of the acquisition.

To serve as a functional control point API, the following additional requirements need to be met:

1. BlueZ should close the socket when the active client device disconnects. (This is also necessary to renegotiate the MTU.)
2. BlueZ shouldn't allow write / notify / indicate access to other connected client devices while the respective WriteAcquired / NotifyAcquired property is true.

Vudentz, 2023-01-24:
@beku-epitome this changes quite a bit the design, so we need a socket pair device rather than reusing the same pair, that said one could detect the device has disconnected (via Device.Connected property) and close the fd at the application side until we rework the io support to be per device.

beku-epitome, 2023-01-25:
I didn't think about having a separate socket per client device, but that would make the most sensible design indeed. How would that affect the behavior of the WriteAcquired / NotifyAcquired properties? In this case they won't make any sense as the acquisition is not global, but per device. (On the other hand, it seems to me that BlueZ doesn't care what value these properties have, it only checks if they are present.)

Vudentz, 2023-01-25:
> I didn't think about having a separate socket per client device, but that would make the most sensible design indeed. How would that affect the behavior of the WriteAcquired / NotifyAcquired [...]

For the client WriteAcquired / NotifyAcquired are already on Device object path (per device), for the server we just use its presence to detect if the app is implementing Acquire* methods, but perhaps we can also check if setting it to true we should assume it cannot handle more sockets, so this gives the application ability to tell the daemon if each device individually shall be handled separately or not.

beku-epitome, 2023-01-26:
> perhaps we can also check if setting it to true we should assume it cannot handle more sockets, so this gives the application ability to tell the daemon if each device individually shall be handled separately or not.

This way of operation would be a bit unclear though, so if this path is taken, it must be extensively documented.
In my opinion it never makes sense to share the same socket between multiple client devices:

1. The returned MTU is dependent on the connected client device (I did check the code, and it seems that this returned MTU is discarded, but still the API requires it, and the irrelevance of this value is not documented.)
2. If I understand correctly, this API is added specifically to create a protected single point connection between a client and a server Control Point characteristic. This should not only be reflected in the design of the client side, but also on the server side.

Therefore I see two valid approaches:

1. BlueZ doesn't share the acquired sockets with devices other than the one that was passed in the options, and just returns error when other devices are trying to access the CP while a socket is already open.
2. BlueZ calls the Acquire* API for each device that is trying to access the CP, and lets the server application decide whether to open parallel sockets or return error on the call.

In each case BlueZ should close the socket when the client device disconnects.

beku-epitome, 2023-07-24:
I'm bumping this thread because this bug is a major risk for our secure BLE application, and so is for anyone who also wants to implement a [Bluetooth Authorization Control Service](https://www.bluetooth.com/specifications/specs/authorization-control-service-1-0/) on top of BlueZ. Can we get an ETA for a fix on this?

Vudentz, 2023-07-24:
@beku-epitome while I agree we need to resolve this Im a bit concerned about implementing ACS on top of D-Bus, usually it is _not_ a good idea to implement secure sensitive services on top of D-Bus since a rogue application could also implement them to bypass security, which is why GAP/GATT services resides in bluetoothd to begin with. In this case in particular it seems the control point needs control of the entire database (?), besides you could as well use bt_crypto (which uses the kernel crypto module) instance to implement the procedures.

Vudentz, 2023-07-24:
@beku-epitome please have a look at the patch above if it fixes the problem.

beku-epitome, 2023-07-25:
> Im a bit concerned about implementing ACS on top of D-Bus, usually it is not a good idea to implement secure sensitive services on top of D-Bus since a rogue application could also implement them to bypass security

How is a rogue application any different from a man-in-the-middle attack at any other point of the communication path? ACS defines an application layer security module that achieves end-to-end protection by using an OOB information at the initial key exchange. So the rogue application could only interfere successfully if it can also capture the OOB data. (Correct me if I'm wrong though.)

> it seems the control point needs control of the entire database

My understanding is that only the protected GATT attributes need to be managed by the ACS server, the rest can operate independently.

Vudentz, 2023-07-25:
> How is a rogue application any different from a man-in-the-middle attack at any other point of the communication path? ACS defines an application layer security module that achieves end-to-end [...]

That depends on how the OOB is distributed and stored, specially the later if that is managed by a user application which the rogue application can access as well then it is no good either. Anyway the main aspect here is to make it harder to impersonate the service, since bluetoothd is a system level service one would need to compromise the whole system to be able to do something like a impersonate attack.

> 
> > it seems the control point needs control of the entire database
> 
> My understanding is that only the protected GATT attributes need to be managed by the ACS server, the rest can operate independently.

Well those protected attributes are still part of the GATT database which is controlled by bluetoothd, so how do you plan to share the control? Or the intention is to always have the application that register ACS service to also register attribute that require such controls? While this could work Im afraid you would end up having to duplicate GATT APIs to register these attributes otherwise you are probably confined to a single process having to manage everything related to ACS, which Im not sure it is a good architecture.

Btw, were you able to test with the proposed patch?

Vudentz, 2023-07-27:
@beku-epitome any feedback on the changes?

closed 2023-08-02 by Vudentz