bluez/bluez #461: Problem with Bluez not recognizing bluetooth controller

Zan-XX, 2023-01-28:
I'm using Solus linux with Budgie desktop environment with Linux 6.1.5-229.current x86_64 kernel, but I can't get the bluez bluetooth to work. My wifi works though. Here are some more details:

Output from "rfkill list all":

0: asus-wlan: Wireless LAN

Soft blocked: no

Hard blocked: no
1: asus-bluetooth: Bluetooth

Soft blocked: no

Hard blocked: no
2: hci0: Bluetooth

Soft blocked: no

Hard blocked: no
3: phy0: Wireless LAN

Soft blocked: no

Hard blocked: no
Output from "sudo systemctl status bluetooth":

● bluetooth.service - Bluetooth service

Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; preset: enabled)

Active: active (running) since Fri 2023-01-27 00:31:17 EST; 6min ago

   Docs: man:bluetoothd(8)
Main PID: 4342 (bluetoothd)

 Status: "Running"

  Tasks: 1 (limit: 18240)

 Memory: 600.0K

    CPU: 17ms

 CGroup: /system.slice/bluetooth.service

         └─4342 /usr/lib64/bluez/bluetooth/bluetoothd
Jan 27 00:31:17 asusofzayn systemd[1]: bluetooth.service: Deactivated successfully.

Jan 27 00:31:17 asusofzayn bluetoothd[4211]: Exit

Jan 27 00:31:17 asusofzayn systemd[1]: Stopped Bluetooth service.

Jan 27 00:31:17 asusofzayn systemd[1]: Starting Bluetooth service...

Jan 27 00:31:17 asusofzayn bluetoothd[4342]: Bluetooth daemon 5.66

Jan 27 00:31:17 asusofzayn systemd[1]: Started Bluetooth service.

Jan 27 00:31:17 asusofzayn bluetoothd[4342]: Bluetooth management interface 1.22 initialized

Output from "sudo dmesg | grep -e mt7921e -e btusb":

[ 3.100877] usbcore: registered new interface driver btusb

[ 3.144557] mt7921e 0000:06:00.0: enabling device (0000 -> 0002)

[ 3.164444] mt7921e 0000:06:00.0: ASIC revision: 79220010

[ 3.240413] mt7921e 0000:06:00.0: HW/SW Version: 0x8a108a10, Build Time: 20220908203203a

[ 3.621326] mt7921e 0000:06:00.0: WM Firmware Version: ____000000, Build Time: 20220908203253

[ 4.712873] mt7921e 0000:06:00.0 wlp6s0: renamed from wlan0

Output from "lspci -nnk | grep -i net -A3":

pcilib: Error reading /sys/bus/pci/devices/0000:00:08.3/label: Operation not permitted

06:00.0 Network controller [0280]: MEDIATEK Corp. Device [14c3:7922]

Subsystem: AzureWave Device [1a3b:5300]

Kernel driver in use: mt7921e
07:00.0 Non-Volatile memory controller [0108]: Micron Technology Inc Device [1344:5411] (rev 01)

Output from "sudo bluetoothctl":

Agent registered

[bluetooth]# show

No default controller available

[bluetooth]# list

[bluetooth]# exit

So I've already checked if rfkill is blocking anything, the status of bluetooth, and more. I really want this bluetooth to eventually work. I appreciate any support.

Vudentz, 2023-01-28:
@Zan-XX is it connected over usb or pcie? Perhaps we are missing the driver for the bluetooth side:

`git grep mt7921e drivers/bluetooth/`

Doesn't looks like there is any entry for mt7921 for bluetooth, only net/wireless, so apparently mediatek did not upstream a driver for it yet.

Zan-XX, 2023-01-29:
Below is what I have in "/lib64/firmware/mediatek". This is probably why the wifi works, but I don't see mt7921e in here. And I think this stuff is necessary if you consider https://ask.fedoraproject.org/t/from-5-19-11-to-5-19-14-wireless-doesnt-work-mediatek-mt7921-driver/27546/4 

/lib64/firmware/mediatek:
BT_RAM_CODE_MT7922_1_1_hdr.bin
BT_RAM_CODE_MT7961_1_2_hdr.bin
mt7610e.bin
mt7610u.bin
mt7615_cr4.bin
mt7615_n9.bin
mt7615_rom_patch.bin
mt7622_n9.bin
mt7622pr2h.bin
mt7622_rom_patch.bin
mt7650e.bin
mt7662u.bin
mt7662u_rom_patch.bin
mt7663_n9_rebb.bin
mt7663_n9_v3.bin
mt7663pr2h.bin
mt7663pr2h_rebb.bin
mt7668pr2h.bin
mt7915_eeprom.bin
mt7915_eeprom_dbdc.bin
mt7915_rom_patch.bin
mt7915_wa.bin
mt7915_wm.bin
mt7916_eeprom.bin
mt7916_rom_patch.bin
mt7916_wa.bin
mt7916_wm.bin
mt7986_eeprom_mt7975_dual.bin
mt7986_eeprom_mt7976.bin
mt7986_eeprom_mt7976_dbdc.bin
mt7986_eeprom_mt7976_dual.bin
mt7986_rom_patch.bin
mt7986_rom_patch_mt7975.bin
mt7986_wa.bin
mt7986_wm.bin
mt7986_wm_mt7975.bin
mt7986_wo_0.bin
mt7986_wo_1.bin
mt8173
mt8183
mt8186
mt8192
mt8195
sof
sof-tplg
WIFI_MT7922_patch_mcu_1_1_hdr.bin
WIFI_MT7961_patch_mcu_1_2_hdr.bin
WIFI_RAM_CODE_MT7922_1.bin
WIFI_RAM_CODE_MT7961_1.bin

/lib64/firmware/mediatek/mt8173:
vpu_d.bin
vpu_p.bin

/lib64/firmware/mediatek/mt8183:
scp.img

/lib64/firmware/mediatek/mt8186:
scp.img

/lib64/firmware/mediatek/mt8192:
scp.img

/lib64/firmware/mediatek/mt8195:
scp.img

/lib64/firmware/mediatek/sof:
sof-mt8186.ldc
sof-mt8186.ri
sof-mt8195.ldc
sof-mt8195.ri

/lib64/firmware/mediatek/sof-tplg:
sof-mt8186.tplg
sof-mt8195-mt6359-rt1019-rt5682-dts.tplg
sof-mt8195-mt6359-rt1019-rt5682.tplg

Zan-XX, 2023-01-29:
Also, here is the output of "sudo dmesg | grep Bluetooth":

[    2.856895] Bluetooth: Core ver 2.22
[    2.857002] Bluetooth: HCI device and connection manager initialized
[    2.857007] Bluetooth: HCI socket layer initialized
[    2.857009] Bluetooth: L2CAP socket layer initialized
[    2.857016] Bluetooth: SCO socket layer initialized
[    3.531775] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    3.531782] Bluetooth: BNEP socket layer initialized
[    4.922617] Bluetooth: hci0: Opcode 0x c03 failed: -110

Zan-XX, 2023-01-30:
Oh wait, actually it seems like I use USB instead of pcie. Nonetheless, the issue persists. It might be fixed when a new kernel/firmware gets released.

closed 2025-08-13 by github-actions[bot]