bluez/bluez #462: Setting of io capability not work.

Timfuture, 2023-01-31:
Next logs come from btmon:

_@ MGMT Command: Set IO Capability (0x0018) plen 1                                                                                                                                {0x0001} [hci0] 17.926604
        **Capability: KeyboardDisplay (0x04)**
@ MGMT Event: Command Complete (0x0001) plen 3                                                                                                                                   {0x0001} [hci0] 17.926605
      Set IO Capability (0x0018) plen 0
        Status: Success (0x00)_

...

> HCI Event: IO Capability Response (0x32) plen 9                                                                                                                                     #78 [hci0] 94.155869
        Address: 80:04:5F:74:35:7C (OUI 80-04-5F)
        **IO capability: DisplayYesNo (0x01)**
        OOB data: Authentication data not present (0x00)
        Authentication: General Bonding - MITM not required (0x04)
> HCI Event: IO Capability Request (0x31) plen 6                                                                                                                                      #79 [hci0] 94.158813
        Address: 80:04:5F:74:35:7C (OUI 80-04-5F)
< HCI Command: IO Capability Request Reply (0x01|0x002b) plen 9                                                                                                                       #80 [hci0] 94.159004
        Address: 80:04:5F:74:35:7C (OUI 80-04-5F)
        IO capability: DisplayYesNo (0x01)
        OOB data: Authentication data not present (0x00)
        Authentication: General Bonding - MITM required (0x05)
> HCI Event: Command Complete (0x0e) plen 10                                                                                                                                          #81 [hci0] 94.160871
      IO Capability Request Reply (0x01|0x002b) ncmd 1
        Status: Success (0x00)
        Address: 80:04:5F:74:35:7C (OUI 80-04-5F)

You can see, at the begining, the IO capability be set by MGMT command as 0x04 which is KeyboardDisplay.
But when another device try to connect, got IO capability 0x01 which is DisplayYesNo.
I don't understand why this happened.
The complete log file is attached.
[io-capability-btmon.txt](https://github.com/bluez/bluez/files/10544246/io-capability-btmon.txt)

Vudentz, 2023-01-31:
@Timfuture how are you setting it? Are you using bluetoothctl and are there other pairing agents active in the system?

Timfuture, 2023-02-05:
@Vudentz  Thank you for reply，
Let me detailed my test environment and steps:
I test it on Ubuntu which running on a virtualbox, and the bluetooth device was mapping into ubuntu from host system as a USB device。
The test steps like next:

1. Execute wireshark, and btmon 
2. Run bluetoothctl and next cmds on terminal window:
$:bluetoothctl
$:agent on
$:default-agent
$:power on
$:advertis on
$:scan on
3. Connect ubuntu22(my ubuntu bluetooth device name) with my iphone.

I am a little doubt that it caused by the virtualbox env.
Hope those info help to you.

Vudentz, 2023-02-07:
@Timfuture Make sure that you remove any pairing before you attempt to connect, in both ends, also Im not sure why you are doing both advertise and scan? You advertise if you are acting as peripheral, and scan if you are central, doing both is probably not the right thing to do.