bluez/bluez #465: Failed to add advertisement: Unknown Command (0x01)

yongchang1223, 2023-02-03:
Testing BLE advertising by bluetoothctl

Agent registered
[CHG] Controller 01:BF:77:21:CC:10 Pairable: yes
AdvertisementMonitor path registered
[bluetooth]# advertise on
Failed to register advertisement: org.bluez.Error.Failed

HCI sniffer log:
HCI sniffer - Bluetooth packet analyzer ver 5.53
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff
< HCI Command: Unknown (0x08|0x0036) plen 25
> HCI Event: Command Complete (0x0e) plen 5
    Unknown (0x08|0x0036) ncmd 5
< HCI Command: Unknown (0x08|0x0037) plen 7
> HCI Event: Command Complete (0x0e) plen 4
    Unknown (0x08|0x0037) ncmd 5
< HCI Command: Read Local Name (0x03|0x0014) plen 0   << issue
> HCI Event: Command Complete (0x0e) plen 4
    ****Read Local Name (0x03|0x0014) ncmd 5
    status 0x01 name '.....'
    Error: Unknown HCI Command****

My understanding , the Read Local Name command is only used for BR/EDR controller. Why blueZ send this command when starting advertising?

Vudentz, 2023-02-03:
@yongchang1223 is the controller only supporting LE? If it is dual-mode it may attempt to read the local name to add it in the advertising/scan response.

yongchang1223, 2023-02-06:
> @yongchang1223 is the controller only supporting LE? If it is dual-mode it may attempt to read the local name to add it in the advertising/scan response.

Hi, @Vudentz ,The controller is BLE standalone. 
And base on the following initialize log (hciattach), I think the host already recognize that this is a BLE device.
HCI sniffer - Bluetooth packet analyzer ver 5.53
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff
< HCI Command: Reset (0x03|0x0003) plen 0
> HCI Event: Command Complete (0x0e) plen 4
    Reset (0x03|0x0003) ncmd 5
    status 0x00
< HCI Command: Read Local Supported Features (0x04|0x0003) plen 0
> HCI Event: Command Complete (0x0e) plen 12
    Read Local Supported Features (0x04|0x0003) ncmd 5
    status 0x00
    Features: 0x00 0x00 0x00 0x00 0x60 0x00 0x00 0x00    << BR/EDR NOT SUPPORTED,  LE SUPPORTED

Vudentz, 2023-02-06:
@yongchang1223 something is off, what kernel version are you running?

This is what the current code does:

https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/tree/net/bluetooth/hci_sync.c#n3373
https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/tree/net/bluetooth/hci_sync.c#n3459
https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/tree/net/bluetooth/hci_sync.c#n3623

So the code only really reads the local name if BR/EDR is supported and only at the init stage (e.g. power on), not when you add an advertising instance, when I do add advertise on I don't see this command even on dual-mode.

allen-pasch-vargas, 2023-05-08:
We ran into the same problem when using the 5.x version of the Linux kernel.

The bug appears to be in [mgmt.c:add_ext_adv_data()](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/net/bluetooth/mgmt.c?h=v5.15.110#n8128), which unconditionally tries to read the local name, even if the device does not support BR/EDR.  We compiled the kernel with this line of code commented out and BR/EDR disabled with the CONFIG_BT_BREDR flag, and everything worked.

All of the other code reading or writing the local name in the 5.x Linux kernel is surrounded by a guard making sure it only happens when the device supports BR/EDR.  It doesn't look like this bug exists in the 6.x Linux kernel, since the code was written and the same function no longer tries to read the local name; but we haven't tried it.

avanmalleghem, 2025-03-07:
> We ran into the same problem when using the 5.x version of the Linux kernel.
>
> The bug appears to be in [...]

Well, you definitely saved me some nights ! Thank you very much ! It works on Jetpack 6.1 (NVIDIA Jetson series)

closed 2025-08-13 by github-actions[bot]