bluez/bluez #466: reconnection to HID slow

DrCWO, 2023-02-03:
I run an application using a Microsoft Surface Dial.
After some time not being used it fell asleep disconnecting BT.

Leading to:
`[CHG] Device 70:BC:10:86:85:C5 Connected: no`

Touching the Microsoft Surface Dial I see immediately 
`[CHG] Device 70:BC:10:86:85:C5 Connected: yes`

Then it takes 1.25 seconds till I see
`[CHG] Device 70:BC:10:86:85:C5 ServicesResolved: yes`

and one more  second until the HID device is available and useable again.

**Any idea what happens during this time?
Why have I wait for so long until I can use it again via HID?**

This is very uncomfortable for my users as the device is used in my application as a voloume know 
and the users expected an immediate reaction.

Device --> Raspberry Pi4
Platform --> ARMv8
bluetoothctl -v --> bluetoothctl: 5.55
uname -a --> Linux DietPi 5.15.61-v8+ #1579 SMP PREEMPT Fri Aug 26 11:16:44 BST 2022 aarch64 GNU/Linux

Support would be much appreciated.
Best DrCWO

DrCWO, 2023-02-03:
Retested this on same machine but with:

Device --> Raspberry Pi4
Platform --> ARMv6
bluetoothctl -v --> bluetoothctl: 5.50
uname -a --> Linux Raspbian 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux

Here the time from 
`[CHG] Device 70:BC:10:86:85:C5 Connected: yes`

to 
`[CHG] Device 70:BC:10:86:85:C5 ServicesResolved: yes`

Is 0.25s and the HID device is immediately after hat available.

There is definitely something going wrong with 5.55 or the ARMv8 kernel...

Best DrCWO

Vudentz, 2023-02-03:
@DrCWO 5.55 is sort of old already, any chance to test with something more recent?

DrCWO, 2023-02-04:
@Vudentz 
I did:
`apt-get update`
and
```
root@rooExtend:/home/pi# apt-get install bluez
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
bluez is already the newest version (5.55-3.1+rpt2).
0 upgraded, 0 newly installed, 0 to remove and 57 not upgraded.
```
**Seems to be the latest that I can get?**

Any more suggestions?