bluez/bluez #467: Laggy mouse and connection issues while using headset in mSBC mode

zsolt-donca, 2023-02-03:
I have issues with my bluetooth mouse while using my headset in mSBC mode: several seconds after my headset switches to mSBC mode, my bluetooth mouse will start to lag (starts to have a very annoyingly large input latency). The issue completely disappears the moment the headset switches back to A2DP mode. Also, if I disconnect the mouse while it lags, it will not be able to reconnect while my headset stays in mSBC mode, and I can see in my system settings that it tries to connect about every second but it fails every time. It will also reconnect successfully the moment the headset leaves mSBC and gets back to A2DP.

There is some randomness to this issue, as it does not appear all the time (though it does most of the time), but if the issue appears, it will 100% surely stay until I switch the headset back - it does not get better on its own (even after tens of minutes). Also, after starting a call, sometimes I see my mouse "hiccup" right around the time it would otherwise start to lag, but this is just for a fraction of a second and the mouse will continue working fine afterwards. The headset works perfectly fine in all cases, so there is at least that.

I first reported this issue at [pipewire](https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/2961), but I was redirected here.

I am on Arch Linux. To troubleshoot this issue, I installed [bluez-git](https://aur.archlinux.org/packages/bluez-git), which at the moment of writing is at build `1:5.66.r87.g67395a3b3-1`. Also, I installed the latest firmware using [linux-firmware-git](https://aur.archlinux.org/packages/linux-firmware-git) (since there was a firmware update in the mainline linux kernel for my bluetooth adaptor). I am on the zen kernel `6.1.9-zen1-1-zen`.

Some additional info:
```
❯ lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 27c6:63ac Shenzhen Goodix Technology Co.,Ltd. Goodix USB2.0 MISC
Bus 003 Device 002: ID 0c45:6a15 Microdia Integrated_Webcam_HD
Bus 003 Device 004: ID 8087:0033 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 004: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 006 Device 003: ID 0bda:0413 Realtek Semiconductor Corp. Dell dock
Bus 006 Device 002: ID 0bda:0487 Realtek Semiconductor Corp. Dell dock
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 006: ID 413c:b06e Dell Computer Corp. Dell dock
Bus 005 Device 008: ID 413c:b06f Dell Computer Corp. Dell dock
Bus 005 Device 007: ID 0bda:402e Realtek Semiconductor Corp. USB Audio
Bus 005 Device 005: ID 0bda:5413 Realtek Semiconductor Corp. Dell dock
Bus 005 Device 004: ID 2109:0100 VIA Labs, Inc. USB 2.0 BILLBOARD             
Bus 005 Device 003: ID 0bda:5487 Realtek Semiconductor Corp. Dell dock
Bus 005 Device 002: ID 3297:1969 ZSA Technology Labs Moonlander Mark I
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

❯ bluetoothctl devices
Device F4:B6:88:5D:3C:37 PLT Focus
Device C3:12:31:3D:65:0F MX Master
```

The headset is ["PLT Focus" above](https://www.poly.com/us/en/products/headsets/voyager/voyager-focus-uc), and the mouse it a Logitech MX Master.

Steps to reproduce the lag issue:
1. connect PLT Focus
2. connect MX Master
3. Start an online meeting (this switches the headset to mSBC)
4. Wait 10-20 seconds while moving the mouse
5. Notice that the mouse will start to lag (most likely)

I just did the above while collecting data using `sudo btmon -w scenario1.btsnoop | tee scenario1.log`, and got the files: 
[scenario1.tar.gz](https://github.com/bluez/bluez/files/10582150/scenario1.tar.gz)

Steps to reproduce the connection issue (steps 1-5 are the same as above):
1. connect PLT Focus
2. connect MX Master
3. Start an online meeting (this switches the headset to mSBC)
4. Wait 10-20 seconds while moving the mouse
5. Notice that the mouse will start to lag (most likely)
6. Now turn off the mouse
7. Turn the mouse back on, and notice that it does not successfully connect, even though the system tries every second or so
8. End the meeting, and notice that the mouse connects the moment the headset switched back to A2DP

I just did the above and got these files: 
[scenario2.tar.gz](https://github.com/bluez/bluez/files/10582246/scenario2.tar.gz)

Thanks for looking into this.

Vudentz, 2023-02-03:
@zsolt-donca this might be down to the link-layer, do you have the dmesg out to know exactly which model and firmware it is loading?

zsolt-donca, 2023-02-03:
@Vudentz I hope this is it - if not, please let me know how to figure it out:
```
❯ sudo dmesg | grep -i bluetooth
[    7.904463] Bluetooth: Core ver 2.22
[    7.905637] NET: Registered PF_BLUETOOTH protocol family
[    7.905640] Bluetooth: HCI device and connection manager initialized
[    7.905646] Bluetooth: HCI socket layer initialized
[    7.905649] Bluetooth: L2CAP socket layer initialized
[    7.905654] Bluetooth: SCO socket layer initialized
[    8.225209] Bluetooth: hci0: Device revision is 0
[    8.225213] Bluetooth: hci0: Secure boot is enabled
[    8.225214] Bluetooth: hci0: OTP lock is enabled
[    8.225216] Bluetooth: hci0: API lock is enabled
[    8.225216] Bluetooth: hci0: Debug lock is disabled
[    8.225217] Bluetooth: hci0: Minimum firmware build 1 week 10 2014
[    8.225219] Bluetooth: hci0: Bootloader timestamp 2019.40 buildtype 1 build 38
[    8.283163] Bluetooth: hci0: Found device firmware: intel/ibt-0040-0041.sfi
[    8.283210] Bluetooth: hci0: Boot Address: 0x100800
[    8.283212] Bluetooth: hci0: Firmware Version: 107-51.22
[    8.460165] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    8.460167] Bluetooth: BNEP filters: protocol multicast
[    8.460170] Bluetooth: BNEP socket layer initialized
[    9.860276] Bluetooth: hci0: Waiting for firmware download to complete
[    9.860280] Bluetooth: hci0: Firmware loaded in 1540144 usecs
[    9.860351] Bluetooth: hci0: Waiting for device to boot
[    9.876405] Bluetooth: hci0: Malformed MSFT vendor event: 0x02
[    9.876405] Bluetooth: hci0: Device booted in 15734 usecs
[    9.877634] Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-0040-0041.ddc
[    9.880446] Bluetooth: hci0: Applying Intel DDC parameters completed
[    9.883483] Bluetooth: hci0: Firmware timestamp 2022.51 buildtype 1 build 56683
[    9.953566] Bluetooth: MGMT ver 1.22
[    9.992181] Bluetooth: hci0: offload codecs enable 1 changed 1
[    9.992235] Bluetooth: ISO socket layer initialized
[   25.389568] Bluetooth: RFCOMM TTY layer initialized
[   25.389579] Bluetooth: RFCOMM socket layer initialized
[   25.389587] Bluetooth: RFCOMM ver 1.11
```

zsolt-donca, 2025-09-15:
I would really like if this issue was not closed as "stale", because the issue still persists even though 2 years had passed.

closed 2025-10-31 by github-actions[bot]