bluez/bluez #470: configure: error: D-Bus configuration directory is required

WilliamBehrens, 2023-02-06:
```
./configure --disable-tools --enable-library --disable-client --disable-systemd --disable-monitor --disable-manpages --disable-datafiles --disable-obex
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking how to create a pax tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking whether to enable maintainer-specific portions of Makefiles... no
checking for pkg-config... /usr/bin/pkg-config
checking pkg-config is at least version 0.9.0... yes
checking whether make supports the include directive... yes (GNU style)
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking dependency style of gcc... gcc3
checking for C/C++ restrict keyword... __restrict
checking for gcc... (cached) gcc
checking whether we are using the GNU C compiler... (cached) yes
checking whether gcc accepts -g... (cached) yes
checking for gcc option to accept ISO C89... (cached) none needed
checking whether gcc understands -c and -o together... (cached) yes
checking dependency style of gcc... (cached) gcc3
checking whether gcc accepts -fPIE... yes
checking whether gcc accepts -fsanitize=address... yes
checking whether gcc accepts -fsanitize=leak... yes
checking whether gcc accepts -fsanitize=undefined... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking how to convert x86_64-pc-linux-gnu file names to x86_64-pc-linux-gnu format... func_convert_file_noop
checking how to convert x86_64-pc-linux-gnu file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
checking for dlfcn.h... yes
checking for objdir... .libs
checking if gcc supports -fno-rtti -fno-exceptions... no
checking for gcc option to produce PIC... -fPIC -DPIC
checking if gcc PIC flag -fPIC -DPIC works... yes
checking if gcc static flag -static works... yes
checking if gcc supports -c -o file.o... yes
checking if gcc supports -c -o file.o... (cached) yes
checking whether the gcc linker (/usr/bin/ld -m elf_x86_64) supports shared libraries... yes
checking whether -lc should be explicitly linked in... no
checking dynamic linker characteristics... GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
checking whether stripping libraries is possible... yes
checking if libtool supports shared libraries... yes
checking whether to build shared libraries... yes
checking whether to build static libraries... no
checking for explicit_bzero... yes
checking for getrandom... yes
checking for rawmemchr... yes
checking for signalfd... yes
checking for clock_gettime in -lrt... yes
checking for pthread_create in -lpthread... yes
checking for dlopen in -ldl... yes
checking linux/types.h usability... yes
checking linux/types.h presence... yes
checking for linux/types.h... yes
checking linux/if_alg.h usability... yes
checking linux/if_alg.h presence... yes
checking for linux/if_alg.h... yes
checking linux/uinput.h usability... yes
checking linux/uinput.h presence... yes
checking for linux/uinput.h... yes
checking linux/uhid.h usability... yes
checking linux/uhid.h presence... yes
checking for linux/uhid.h... yes
checking sys/random.h usability... yes
checking sys/random.h presence... yes
checking for sys/random.h... yes
checking for glib-2.0 >= 2.28... yes
checking for dbus-1 >= 1.6... yes
checking D-Bus configuration directory... configure: error: D-Bus configuration directory is required
```
I can confirm that dbus is installed so I'm unsure why the configuration directory can't be found?

tedd-an, 2023-02-06:
Do you have `/etc/dbus-1` in your system? 
Run this command to check the dbus configuration directory:
`$ pkg-config --variable=sysconfdir dbus-1`
It should be `/etc`.

If you have a different path,  `--with-dbusconfidir=` option to specify the dbus config path.

WilliamBehrens, 2023-02-08:
The was exactly the issue

closed 2023-02-08 by WilliamBehrens