bluez/bluez #471: MIDI BLE. More than one note does not work correctly

poedyatel, 2023-02-06:
I am trying to use the BLE MIDI-controller M-vave SMK-25. I compiled Bluez 5.66(and several older versions) from the sources, and the controller worked, but there are problems if the generated events are absolutely simultaneous.
If I play one note, all is well, the target gets 2 midi events:
[image]

If I play two notes at the same time, 2 out of 4 events are lost. These can be events of the same note or different ones:
[image]

If I play 3 notes (C3, D3, E3), then the madness begins. Always comes 4 events, but the notes G9 and E4 are generated out of nowhere:
[image]

USB works fine:
[image]

I am using Ubuntu 22.10. To make sure that all my hardware fit, I additionally tested the controller with Windows and everything worked well

Vudentz, 2023-02-06:
@poedyatel afaik we just pass as is to the kernel, perhaps it could be a timing issue due to connection interval, etc, try using btmon to see how it looks over the air.

poedyatel, 2023-02-06:
```
Bluetooth monitor ver 5.66
= Note: Linux version 5.19.0-29-generic (x86_64)                               0.710959
= Note: Bluetooth subsystem version 2.22                                       0.710960
= New Index: 00:1A:7D:DA:71:13 (Primary,USB,hci0)                       [hci0] 0.710961
= Open Index: 00:1A:7D:DA:71:13                                         [hci0] 0.710961
= Index Info: 00:1A:7D:DA:71:13 (Cambridge Silicon Radio)               [hci0] 0.710961
= New Index: 4C:1D:96:5C:42:BA (Primary,USB,hci1)                       [hci1] 0.710962
@ MGMT Open: bluetoothd (privileged) version 1.22                     {0x0001} 0.710962
```

play 1 note

```
> ACL Data RX: Handle 69 flags 0x02 dlen 12                       #13 [hci0] 137.419004
      ATT: Handle Value Notification (0x1b) len 7
        Handle: 0x0072
          Data: 808090307f
> ACL Data RX: Handle 69 flags 0x02 dlen 12                       #14 [hci0] 137.480209
      ATT: Handle Value Notification (0x1b) len 7
        Handle: 0x0072
          Data: 8080803040
```

play 2 notes
          
```
> ACL Data RX: Handle 69 flags 0x02 dlen 14                       #15 [hci0] 164.563041
      ATT: Handle Value Notification (0x1b) len 9
        Handle: 0x0072
          Data: 808090307f327f
> ACL Data RX: Handle 69 flags 0x02 dlen 14                       #16 [hci0] 164.683699
      ATT: Handle Value Notification (0x1b) len 9
        Handle: 0x0072
          Data: 80808032403040
```

play 3 notes

```
> ACL Data RX: Handle 69 flags 0x02 dlen 16                       #17 [hci0] 177.423167
      ATT: Handle Value Notification (0x1b) len 11
        Handle: 0x0072
          Data: 808090327f347f307f
> ACL Data RX: Handle 69 flags 0x02 dlen 16                       #18 [hci0] 177.587819
      ATT: Handle Value Notification (0x1b) len 11
        Handle: 0x0072
          Data: 808080344030403240
```

Vudentz, 2023-02-07:
@poedyatel from the looks of it it seems to try to process everything received, even if there is multiple events:

https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/profiles/midi/midi.c#n115

smerkley, 2024-02-26:
I'm having the same issue with a similar controller, the M-VAVE SMC-PAD. The issue happens not just with notes but with CC values as well. If you turn more than 2 knobs at once you start getting wild CC values. I've confirmed the device works properly with midi monitoring software on my Android phone.

harrison3000, 2024-03-20:
I'm also having the same issue with a SMK-25, the easiest way to reproduce the issue is to enable chord mode

By capturing the bluetooth traffic using wireshark I noticed the chord packets are bigger
Since each key adds 2 more bytes, I would guess this is related to the "Running Status" feature of the MIDI specification, but I might be wrong

single key, 17 bytes: 
```
0000   02 01 2e 0c 00 08 00 04 00 1b 72 00 80 80 80 3c
0010   40
```

2 keys chord, 19 bytes:
```
0000   02 01 2e 0e 00 0a 00 04 00 1b 72 00 80 80 80 3c
0010   40 39 40
```

3 keys chord, 21 bytes:
```
0000   02 01 2e 10 00 0c 00 04 00 1b 72 00 80 80 80 3b
0010   40 3f 40 42 40
```

harrison3000, 2024-03-20:
The bug is very easy to reproduce by using the chord function in the controller
I'm not very good at using C, but I did the following changes to the code

``` diff
diff --git a/profiles/midi/midi.c b/profiles/midi/midi.c
index bab309bca..2ed1114fa 100644
--- a/profiles/midi/midi.c
+++ b/profiles/midi/midi.c
@@ -105,6 +105,8 @@ static void midi_io_value_cb(uint16_t value_handle, const uint8_t *value,
                return;
        }
 
+       warn("Len: %u", length);
+
        snd_seq_ev_clear(&ev);
        snd_seq_ev_set_source(&ev, midi->seq_port_id);
        snd_seq_ev_set_subs(&ev);
@@ -115,6 +117,8 @@ static void midi_io_value_cb(uint16_t value_handle, const uint8_t *value,
        while (i < length) {
                size_t count = midi_read_raw(&midi->midi_in, value + i, length - i, &ev);
 
+               warn("Count: %zu, evt: %x, note: %x", count, ev.type, ev.data.note.note);
+
                if (count == 0)
                        goto _err;

```

That resulted in the following:

Single key:
```
Len: 5
Count: 5, evt: 6, note: 40
```

2 key at the same time:
```
Len: 7
Count: 5, evt: 6, note: 43
Count: 3, evt: ff, note: 43
```

3 keys: 
```
Len: 9
Count: 5, evt: 7, note: 3b
Count: 3, evt: 7, note: 40
Count: 2, evt: ff, note: 40
```

Seems like the first event is correct, but for the running status its reading too much, 3 bytes instead of 2
If I manage to make breakpoints work I'll investigate further, and maybe even try to make a PR, who knows

harrison3000, 2024-03-20:
I found the cause:
The controller is sending the timestamps as all zeros (the `0x80 0x80` at the begining of the messages), making this if enter multiple times for the same message:
```C
	if (parser->timestamp_high == 0)
		parser->timestamp_high = data[i++] & 0x3F;
```

just to test this hipotesis, I fixed it the worst way possible and it worked:
```C
	if (parser->timestamp_high == 0){
		int8_t ts = data[i++] & 0x3F;
		if(ts == 0){
			ts = 1;
		}
		parser->timestamp_high = ts;
	}
```

smerkley, 2024-04-01:
I made a bit more of an elegant solution with a boolean keeping track of the timestamp_high byte being read. My PR: https://github.com/bluez/bluez/pull/801

SunsBR, 2025-08-05:
The PR #801 works really well.
I did try to fix it before finding the information here. there is a way to vote to this pull request to be added to the main distribution?

SunsBR, 2025-08-06:
As smerkley made this pullrequest more than an year ago, I created a patch file.
Not sure if this is a proper patch, I created it using git.

The patch for the 801 pull request is attached.

Vudentz, 2025-09-15:
> Is the patch I sent correct or do you need more information or another type of switches when generating the patch?

For sending the patch please follow:

https://github.com/bluez/bluez/blob/master/HACKING#L98