bluez/bluez #472: "Received error event" multiple brands of bluetooth headphones causing media and audio playback to entirely hang in multiple programs

ipaqmaster, 2023-02-09:
I've noticed this issue on the Airpod Pros and also the WH1000-XM5 series. I can't fully tell if this is a bluez problem or only a pipewire problem (Given only pipewire daemons are complaining in the output below).

Video content in any browser or video playback application will hang for about 30 seconds while it panics and the user journalctl logs will fill up with lines such as:

```
Feb 09 14:48:29 mypc wireplumber[54384]: (bluez_output.A0_A3_09_04_33_33.1-19) running -> error (Received error event)
Feb 09 14:48:29 mypc pipewire[54382]: pw.node: (bluez_output.A0_A3_09_04_33_33.1-181) running -> error (Received error event)
Feb 09 14:48:29 mypc pipewire[54382]: pw.node: (bluez_output.A0_A3_09_04_33_33.1-181) running -> error (error changing node state: Bad address)
```

If I go to PulseAudio Volume Control (Through pipewire-pulse) and change these bluetooth headphone devices from whichever bluetooth audio profile they're currently set to to another, the media playback unfreezes with audio. (Followed by a quick shameful return to the desired bluetooth audio codec they were initially using).

This seems to be happening on all my Archlinux machines the past month.

The airpods use AAC and the XM5s use LDAC by default. It doesn't seem to matter what they use; just that they must have their codec flipped to regain control of a program struggling to resume media playback. Firefox also likes to completely hang during this which renders other tabs unclickable during this headset problem too.

ipaqmaster, 2023-02-09:
To add, disconnecting them or just shutting them off instantly restores media playback through the wired headphones. Something funny with pipewire and bluetooth audio lately it seems. It mentions bluez_output there but I suspect that's just the sink's name and not actually bluez' fault. I've made this issue anyway to search for some helpful input.

pv, 2023-02-09:
Unlikely to be directly bluez related, you should open an issue at https://gitlab.freedesktop.org/pipewire/pipewire --- for best results, fill in the information asked in the bluetooth issue template and take the debug logs

closed 2023-02-09 by ipaqmaster