bluez/bluez #475: A2DP readiness takes a very long time

tymokvo, 2023-02-10:
Hello,

I've been having issues with bluetooth headphones on Fedora 36 intermittently and it seems to have started in the last several weeks. 

I posted an issue on [AskFedora](https://ask.fedoraproject.org/t/fedora-36-a2dp-bluetooth-audio/27843/7). The issue ended up resolving itself (My WH-1000XM3 headphones are _now_ working as an A2DP sink), but it took about an hour of waiting for that to happen.

`bluetoothctl devices`:
```console
Device CC:98:8B:93:D0:18 WH-1000XM3
```

The tail end of `journalctl -b -u bluetooth` shows:
```console
Feb 10 07:12:46 freibergite bluetoothd[1370]: Failed to set mode: Failed (0x03)
Feb 10 08:13:45 freibergite bluetoothd[1370]: /org/bluez/hci0/dev_CC_98_8B_93_D0_18/sep1/fd1: fd(34) ready
```
The first entry here is from immediately after I re-paired my headphones.
If you check the timestamps, you'll see that `bluez` seems to have created a file descriptor matching the device's MAC about an hour afterwards.

Vudentz, 2023-02-10:
@tymokvo what version are you running? Perhaps there is something with the cache, have you tried removing the device and repairing it?

tymokvo, 2023-02-10:
Hm, I'm not sure how to check the `bluez` version. I've removed and re-paired the device multiple times through the GUI. Should I try doing it with `bluetoothctl`?

`bluetoothctl version`: `5.66`

`pipewire --version`:
```console
pipewire
Compiled with libpipewire 0.3.64
Linked with libpipewire 0.3.64
```

Vudentz, 2023-02-10:
@tymokvo the version is the latest it seems, try collecting the HCI traces using btmon.

closed 2025-08-13 by github-actions[bot]