bluez/bluez #478: Xbox one S controller lastest firmware + AMD RZ608 bluetooth = regression

deathxxx123, 2023-02-15:
On Linux
Regression with lastest controller firmware + AMD RZ608 (a real MediaTek MT7921K chip) bluetooth - combo.
So, problem is the lastest firmware v5.xxx that cannot be seen in my BT devices list and also i cannot connect my controller.
With firmware v3.xxx and v4.xxx is no problem. With cable - no problem too with any firmware.

On Win11:
No problem with my BT and Xbox controller combo.
Win11 can see my controller and connect with it.

My experience:
I try with different BT device - a BT 4.0: CSR 8510 A10 chip, usb dobgle.
With this dongle and lastest firmware is working as should on Linux.

So to use this controller on linux, i need to revert firmware version to factory v3.xxx and now controller is working with my BT AMD RZ608.

In short, lastest controller update breaks working on Linux with my BT AMD RZ608, that is integrated in my Gigabyte X570S AORUS ELITE AX

deathxxx123, 2023-02-15:
I wonder: Can this be fixed in this driver, or MS itself need fix firmware?

Vudentz, 2023-02-15:
@deathxxx123 Please adds dmesg and btmon traces so we can check what is going on.

deathxxx123, 2023-02-16:
> @deathxxx123 Please adds dmesg and btmon traces so we can check what is going on.

Hi,
i'm affraid that dmesg shows just noting.
And how to btmon trace?
Or i can provide a remote desktop to my PC, so you can check better.
It's hard to trace, because any program just not detect anything. Just like controller does not exist.

deathxxx123, 2023-02-16:
I will refpash controller to lastest firmware again and try trace with `btmon`

deathxxx123, 2023-02-16:
That is btmon log with already registered controller and try to connect.
[xbox.log](https://github.com/bluez/bluez/files/10753978/xbox.log)

deathxxx123, 2023-02-17:
So i understand that firmware v5.xxx support only BLE, witch don't supported by linux driver for my BT model.

deathxxx123, 2023-02-23:
It is working with my another Wi-Fi-BT combo adapter:
Intel Wireless-AC 9260 Bluetooth Adapter

Detected and connected in no time.

deathxxx123, 2023-02-23:
And still connected to steam games, but stops working untill disconnect and connect again. That happens several times and i moved to PS4 controller without any issues.
So with no fix with Xbox controller and my BT devices, i will use it on Windows, where all are working as should and not even one stop working or disconnect.
PS4 controlled in other hand working best with linux out of box, but not working on Win out of box.

deathxxx123, 2023-02-23:
So after lastest firmware update, Xbox controller are in no use on linux, and when i use on proper BT to connect, still play games with me. Without XpadNeo driver, some buttons are replaced with another one's. What is happening with newest updates? Why all is breaking? Xbox controllers was best support on linux, now is just a useless ... i dont want to say what.

deathxxx123, 2023-02-23:
And why we still need xpadneo driver, and still not supported by the kernel? Wow, just wow.

Vudentz, 2023-02-23:
> And why we still need xpadneo driver, and still not supported by the kernel? Wow, just wow.

Not sure what this has to do with BlueZ though? Are you sure you don't want to contact MediaTek folks, since it appears this is related to their firmware there is nothing we can do in the stack.

deathxxx123, 2023-02-25:
> > And why we still need xpadneo driver, and still not supported by the kernel? Wow, just wow.
> 
> Not sure what this has to do with BlueZ though? Are you sure you don't want to contact MediaTek folks, since it appears this is related to their firmware there is nothing we can do in the stack.

MediaTek just don't have any support options. The whole site is confusing and no place to download drivers also.
I guess, i will not mess anymore with MediaTek then.

deathxxx123, 2023-03-05:
I found something intresting:
https://www.reddit.com/r/Amd/comments/n90uv3/anandtech_amds_wifi_6e_module_the_rz608/ 
So it's already problematic with linux.

deathxxx123, 2023-03-05:
And i'm really really confused, where i should report this issue...
1. To linux kernel
2. To MediaTek
3. To GigaByte
4. To AMD
 Really who is making the drivers for linux?

deathxxx123, 2023-03-06:
Wow there is so hard to contact MediaTek and send feedback for any issues....
How they develop drivers without any feedback?
I search for any contact to report issues, and guess what: there is no contact for report issues.
You only can report security issues.
What a shadow company. They don't have a drivers page too. Only blogs and press media with no contact info.

deathxxx123, 2023-08-04:
Now with kernel 6.4.6 BT is working and connected/

gottaeat, 2024-02-05:
issue still persists with `mt7921e` on linux 6.7.3 and bluez 5.72 on my arch box here.

i have the rz608 sku with the bluetooth + wireless combo on an msi b550m pro-vdh mobo.

the xbox series x|s controller simply does not show up when scanned. the controller firmware got updated on a windows 10 machine and have successfully paired with the same windows 10 machine. also pairs without any hassle on several android devices.

the bluetooth bits are presented as such in lspci:
```sh
Bus 001 Device 006: ID 0e8d:0608 MediaTek Inc. Wireless_Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 [unknown]
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0e8d MediaTek Inc.
  idProduct          0x0608 Wireless_Device
  bcdDevice            1.00
  iManufacturer           5 MediaTek Inc.
  iProduct                6 Wireless_Device
  iSerial                 7 000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00fe
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          8 Config_01
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              100mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass        224 Wireless
      bFunctionSubClass       1 Radio Frequency
      bFunctionProtocol       1 Bluetooth
      iFunction               4 BT_FUNCTION
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       224 Wireless
      bInterfaceSubClass      1 Radio Frequency
      bInterfaceProtocol      1 Bluetooth
      iInterface              1 BT_ACL_If
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
[... 342 lines cut ...]
        Capabilities: [f8] Power Management version 3
                Flags: PMEClk- DSI+ D1+ D2+ AuxCurrent=0mA PME(D0+,D1+,D2+,D3hot+,D3cold+)
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [100 v1] Vendor Specific Information: ID=1556 Rev=1 Len=008 <?>
        Capabilities: [108 v1] Latency Tolerance Reporting
                Max snoop latency: 0ns
                Max no snoop latency: 0ns
        Capabilities: [110 v1] L1 PM Substates
                L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
                          PortCommonModeRestoreTime=3us PortTPowerOnTime=28us
                L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
                           T_CommonMode=0us LTR1.2_Threshold=0ns
                L1SubCtl2: T_PwrOn=10us
        Capabilities: [200 v2] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
                AERCap: First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap- ECRCChkEn-
                        MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
                HeaderLog: 00000000 00000000 00000000 00000000
        Kernel driver in use: mt7921e
        Kernel modules: mt7921e
```

closed 2025-08-12 by github-actions[bot]