bluez/bluez #479: Broken AVRCP Volume synchronization

BenKaufmann, 2023-02-16:
Starting with bluez version 5.65, we often observe broken volume synchronization in particular with iOS devices.

The situation looks like this:

- Connect with an iOS phone (tested with iPhone 8 and iOS version 16.3.1)
- Start playback
- Change volume on the phone (-> avrcp event is received and volume is propagated on MediaTransport1 interface)
- Change volume via dbus on MediaTransport1
- Expected: Volume is propagated to phone
- :x:  Actual: Often, volume is not propagated (no avrcp event is sent)

I debugged the issue and from what I can see, it is a side-effect of [this recent commit](https://github.com/bluez/bluez/commit/fa7828bddd2164408535a9ac45095564e9ebbeea). 

 In particular, what happens is:

- During the connection attempt, `avrcp_handle_register_notification()` is called for event `AVRCP_EVENT_VOLUME_CHANGED`
- `avrcp_handle_register_notification()`  calls `media_transport_get_device_volume()` but for some reason (and not always), no transport exists (yet) at this point.
- Previous versions of bluez (< 5.65) always returned `0` in this case but since the above mentioned fix, the function returns `btd_device_get_volume()`, which is typically still set to the initial value of `-1`.
- A returned volume of -1 is treated as error by `avrcp_handle_register_notification()`  and thus  the `AVRCP_EVENT_VOLUME_CHANGED` event is not added to `registered_events`
- Future transport volume changes are then not propagated because `avrcp_set_volume()/avrcp_event()`  detect that `AVRCP_EVENT_VOLUME_CHANGED` is not among the registered events.

I'm sure there are better fixes, but as a quick workaround, I adjusted `media_transport_get_device_volume()` such that it returns the maximum of `0` and `btd_device_get_volume(dev)` and with this, volume synchronization again seems to work reliably in both directions.

dkarnikis, 2023-02-24:
I have the exact same issue. The avrcp spec 1.6.2 mentions that the ranges of volume should be in range of 0x0 - 0x7F (0 - 100%). And the top bit (7) is reserved for future addition. I think -1 is undefined behavior, so initializing to 0 might be considered fine?

Asikbct, 2025-07-21:
Has a fix been provided for this issue?