bluez/bluez #480: Unnecessary fragmentation of packets slows down communication.

zapta, 2023-02-18:
When connecting to the a BLE device, the bluez client forces packet fragmentation which significantly slows down the communication to a point that the app fails. This doesn't happen with other BLE client we tested such as WinRT on windows or Nordic nRF connect on iPhone. 

Is there a way to enable DLE (Data Length Extension) with Bluez?

The connection starts with bluez (master) sending an LL_FEATURE_REQUEST:
[image]

That LL_FEATURE_REQUEST has the data extension feature disabled
[image]

Later the master and device exchange MTU and agree on 247 
[image]

But because bluez forced data fragmentation, each packet takes multiple slots to complete which slows the communication to a point that the app fails:
[image]

The full wireshark packets log is available here
[mtu_linux_filtered.zip](https://github.com/bluez/bluez/files/10775299/mtu_linux_filtered.zip)

(This issue, is a continuation of this Bleak issue https://github.com/hbldh/bleak/issues/1227 )

zapta, 2023-02-19:
If it helps, here is the hciconfig info.

```
$ hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: EC:63:D7:F1:C7:8B  ACL MTU: 1021:5  SCO MTU: 96:6
	UP RUNNING 
	RX bytes:812 acl:0 sco:0 events:67 errors:0
	TX bytes:5420 acl:0 sco:0 commands:67 errors:0
	Features: 0xff 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'linux'
	Class: 0x7c0104
	Service Classes: Rendering, Capturing, Object Transfer, Audio, Telephony
	Device Class: Computer, Desktop workstation
	HCI Version: 4.2 (0x8)  Revision: 0x1000
	LMP Version: 4.2 (0x8)  Subversion: 0x1000
	Manufacturer: Intel Corp. (2)
```

Vudentz, 2023-02-19:
@zapta we are setting using the value we read as maximum the controller supports:

https://github.com/bluez/bluetooth-next/blob/master/net/bluetooth/hci_event.c#L2113
https://github.com/bluez/bluetooth-next/blob/master/net/bluetooth/hci_sync.c#L4366

Well except if the controller doesn't really support DLE then we have no way to use it.

zapta, 2023-02-19:
Thanks Luiz.

1. I will try to figure out what BT adapter is for my computer and check
its docs. Any suggestions on how to do that? E.g. a command line that will
print an adapter model?  It's this one
https://www.bee-link.com/catalog/product/index?id=302

2. Can I plug in a USB BT adapter that will be recognized automatically?
Hopefully I can find one that will support packet length extension. E.g. I
can try this one  https://www.amazon.com/gp/product/B009ZIILLI

Vudentz, 2023-02-21:
> Thanks Luiz. 1. I will try to figure out what BT adapter is for my computer and check its docs. Any suggestions on how to do that? E.g. a command line that will print an adapter model? It's this one https://www.bee-link.com/catalog/product/index?id=302 2. Can I plug in a USB BT adapter that will be recognized automatically? Hopefully I can find one that will support packet length extension. E.g. I can try this one https://www.amazon.com/gp/product/B009ZIILLI EDIT: This is the radio module I found in the computer [image]. Seems to be an old model even though the computer is new. I also ordered this https://www.amazon.com/gp/product/B08M1VJHVD from Amazon to try, I believe it's based on RTL8763B.

You can check what the controller supports via btmon during the initialization (e.g. bluetoothctl> power off/power on), and yes it shall be possible to plug another controller over USB.

zapta, 2023-02-22:
Thanks @Vudentz.  I got this adapter https://www.amazon.com/dp/B08M1VJHVD, installed in my ubuntu box, disabled the builtin one and now everything works with no fragmentation.  It's a much newer adapter and it supports the data extension feature.

Please feel free to close this issue.

closed 2023-02-23 by Vudentz