bluez/bluez #481: Hearing Aids not recognized as streaming devices

barolo, 2023-02-25:
I have a Pairr of ASHA protocol enabled Hearing Aids [Signia Styletto 7 AX], which connect successfully with kernel 6.2.1 and bluez 5.66, but aren't recognized as streaming/audio devices.

bluetoothctl info:
```
        Name: greg's hearing aids
        Alias: greg's hearing aids
        Paired: yes
        Bonded: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Google Inc.               (0000fdf0-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (0a23ae62-c4c2-43d1-87b1-e8c83839a063)
        UUID: Vendor specific           (56772eaf-2153-4f74-acf3-4368d99fbf5a)
        UUID: Vendor specific           (7d74f4bd-c74a-4431-862c-cce884371592)
        UUID: Vendor specific           (8341f2b4-c013-4f04-8197-c4cdb42e26dc)
        UUID: Vendor specific           (c8f777d0-21b2-45b8-87f8-bd49a13eff49)
        UUID: Vendor specific           (c8f7a831-21b2-45b8-87f8-bd49a13eff49)
        UUID: Vendor specific           (d1d4dc2a-215f-44d2-b44c-0f4de3c91af2)
        ManufacturerData Key: 0x0295
        ManufacturerData Value:
  01 00                                            ..              
        ServiceData Key: 0000fdf0-0000-1000-8000-00805f9b34fb
        ServiceData Value:
  01 02 35 43 42 44                                ..5CBD          
        AdvertisingFlags:
  06
```

kernel spits out this

```
[  753.467751] Bluetooth: hci0: Device setup in 147751 usecs
[  753.467759] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  753.749197] Bluetooth: hci0: AOSP extensions version v1.00
[  753.749204] Bluetooth: hci0: AOSP quality report is supported
[  789.176169] Bluetooth: hci0: ACL packet for unknown connection handle 3837
[  812.336628] Loading firmware: mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin
[  812.478636] Bluetooth: hci0: Device setup in 146985 usecs
[  812.478643] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[  812.754954] Bluetooth: hci0: AOSP extensions version v1.00
[  812.754961] Bluetooth: hci0: AOSP quality report is supported
[  837.887729] Bluetooth: hci0: Opcode 0x 401 failed: -16
[ 1030.562205] Bluetooth: hci0: ACL packet for unknown connection handle 3837
```

paulmenzel, 2023-07-27:
Similar question asked on the mailing list on July 25th, 2023 in thread [Bluetooth for hearing aids support (ASHA)](https://lore.kernel.org/linux-bluetooth/CABBYNZ+Rs8XLy9=BSJfTUPeJe69Fw0bVCpb1KQmNw305ATH-0w@mail.gmail.com/T/#t).

Please contact the vendor and ask for GNU/Linux/BlueZ driver.

barolo, 2023-07-29:
AFAIK these devices are driverless and use vendor neutral, open, ASHA stack over BT with Android based mobile devices. And similarly MFi within Apple's ecosystem.
Waiting for LE Audio is not a solution since it's highly unlikely that they'll ever get updated firmware supporting it.

paulmenzel, 2023-07-30:
Yes, still somebody needs to implement it for BlueZ. The “hardware vendors” should do it or contract the work. That’s how FLOSS works.

barolo, 2023-07-30:
Currently it seems that implementation lies on the side of OS vendor, namely Google-Android, Apple-iOS, soon MacOS, and later on Microsoft-Windows.
Leaving Linux user with what? Hopes that a commercial distro will pick the feature up?

tazz4843, 2023-08-30:
I have a pair of Cochlear Nucleus 7 sound processors that also support ASHA. Given that Cochlear has released the Nucleus 8, I expect no more updates to the firmware on these devices, which will likely be stuck with exclusively ASHA forever.

bluetoothctl output
```
Device 84:77:78:02:C2:69 (public)
        Name: -redacted-
        Alias: -redacted-
        Paired: yes
        Bonded: yes
        Trusted: no
        Blocked: no
        Connected: yes
        LegacyPairing: no
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
        UUID: Google Inc.               (0000fdf0-0000-1000-8000-00805f9b34fb)
        UUID: Vendor specific           (3c3c6c6b-c9bc-4704-bb14-2ffd32b5aa31)
        UUID: Vendor specific           (7d74f4bd-c74a-4431-862c-cce884371592)
        UUID: Vendor specific           (8341f2b4-c013-4f04-8197-c4cdb42e26dc)
        Modalias: bluetooth:v0497p0001d000E
        ManufacturerData Key: 0x0497 (1175)
        ManufacturerData Value:
  4c 60 41 3c 03                                   L`A<.     
```

Paired and connected successfully on kernel 6.1.48-1-lts on Arch Linux, with bluez 5.69.

I would be willing to give this a stab if given a few hints on where to go to implement this, as there's, to put it bluntly, zero chance any company will implement this, not to mention there is no one vendor to ask for a driver, as ASHA is an open protocol.

ford-prefect, 2024-05-06:
We have been taking a stab at this (had to pause work on this over the last year, taking this up again now). I've redrafted the API spec to make it simpler, comments welcome @Vudentz (and please let me know if there's a better place to post this for feedback).

https://asymptotic.notion.site/ASHA-Spec-for-Bluez-61a46027fd46458d8235e14d6bedb2ce

Vudentz, 2024-05-06:
> We have been taking a stab at this (had to pause work on this over the last year, taking this up again now). I've redrafted the API spec to make it simpler, comments welcome @Vudentz (and please let me know if there's a better place to post this for feedback).
> 
> [asymptotic.notion.site/ASHA-Spec-for-Bluez-61a46027fd46458d8235e14d6bedb2ce](https://asymptotic.notion.site/ASHA-Spec-for-Bluez-61a46027fd46458d8235e14d6bedb2ce)

Looks good, there are few new properties that perhaps you just reuse the old one, like Capabilities vs DeviceCapabilities for example, but I guess we can discuss that later when it comes to the actual code.

ford-prefect, 2024-05-07:
Thank you, I've started work on the BlueZ plugin and will publish a draft PR from [this branch](https://github.com/asymptotic-io/bluez/tree/asha-support) when there's something testable.

Once this is working, the plan is to extend the PipeWire BlueZ module to support ASHA as well. The first pass addresses streaming to a single device, once that's all in place I want to tackle streaming to a pair.

closed 2024-06-13 by BluezTestBot