bluez/bluez #482: RTL8761B USB devices is recognized by ubuntu but doesn't work.

zapta, 2023-02-28:
I am trying to use this BLE adapter https://www.amazon.com/dp/B09KGVFX1K on Ubuntu 22.04.1 LTS Release 22.04. When I run hciconfig I get the information below which right to me (?), but I can't get anything to work with it. For example, when I try to scan for devices using the BT widget I get no devices, or, when I try to run a python/bleak app it failed to connect. 

1. Any suggestion what the problem may be?

2. Alternatively, is there a similar USB adapter with an external antenna that is know to work on ubuntu?  I need BLE 4.2+ that supports the Packet Data Length Extension feature.

BTW, when I installed it on Windows (see below), the driver file names suggested that this is a RTL9761B device.

```
$ hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: 8C:88:2B:45:0B:5E  ACL MTU: 1021:6  SCO MTU: 255:12
	UP RUNNING PSCAN ISCAN 
	RX bytes:1744 acl:0 sco:0 events:192 errors:0
	TX bytes:33509 acl:0 sco:0 commands:192 errors:0
	Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'RTK_BT_5.0'
	Class: 0x7c0104
	Service Classes: Rendering, Capturing, Object Transfer, Audio, Telephony
	Device Class: Computer, Desktop workstation
	HCI Version: 5.1 (0xa)  Revision: 0x9a9
	LMP Version: 5.1 (0xa)  Subversion: 0x8a6b
	Manufacturer: Realtek Semiconductor Corporation (93)
``

To  diagnose I also tried these two configurations:

i. Same adapter but with Windows 10. Windows didn't recognize it so I installed the Windows driver that came with the BT adapter and it worked.  (Does the fact that Windows 10 didn't recognize it out of the box means something?)

ii. Same ubuntu computer but with this Edimax adapter and it works. https://www.amazon.com/dp/B08M1VJHVD

zapta, 2023-03-01:
I found this article https://fosspost.org/fix-bluetooth-rtl8761b-problem-on-linux-ubuntu-22-04/ , aliased rtl8761bu firmware to rtl8761bu and now the adapter works well.

It seems to be a known issue. Anything can be done about it, to have the adapters recognized out of the box?  Is there an open issue for it?

Vudentz, 2023-03-02:
@zapta there is already a bug open for this: https://bugzilla.kernel.org/show_bug.cgi?id=214287 that said it might be better to contact Realtek directly as they should be responsible for submitting proper firmware upstream.

closed 2023-03-02 by Vudentz