bluez/bluez #485: LE Audio transport not created

claudiadraghi, 2023-03-02:
Hello,
I have a setup running Pipewire over BlueZ on both central and Peripheral and trying to send an audio sample file with paplay. 
The issue I encounter very often is that the transport path is not created. From the bluetoothd logs I gathered that this happens because the BAP stream does not go into the QoS config state. It stops after idle -> codec config transition. I cannot find any error in the logs why it stops there.
Can you help with this?
I am using Bluez master with Kernel Experimental enabled, Pipewire master compiled with lc3, Kernel 6.2.0. Default endpoint configuration for BAP Sink and BAP Source.
[no_transport_created.zip](https://github.com/bluez/bluez/files/10872227/no_transport_created.zip)

Vudentz, 2023-03-02:
@claudiadraghi can you try with the above patch? What is the peripheral you are trying to connect btw?

claudiadraghi, 2023-03-03:
Hi,
I applied the patch on https://github.com/tedd-an/bluez-upstream-test/commits/726232 but I didn't see any improvement, it stopped even sooner in the service discovery process. 
In the attached logs I tried the connection several times, switching the GAP roles, but with the same result.
Both central and peripheral are development boards, each connected to an Ubuntu virtual machine, running BlueZ + Pipewire.
Thank you!
[w_patch.zip](https://github.com/bluez/bluez/files/10883449/w_patch.zip)

Vudentz, 2023-03-03:
@claudiadraghi it doesn't seem to be even attempting to config a stream, are you sure pipewire is registering the necessary endpoints? I could only find A2DP endpoints in the logs, perhaps try with bluetoothctl -e, then use endpoint.list after connecting, if that comes out empty that means we couldn't find any matching capabilities, otherwise you should see the transports using transport.list.

claudiadraghi, 2023-03-07:
Hi,
Indeed it was an issue with the endpoints. It did not pass the check in experimental_endpoint_supported(), it seems I need a Kernel update. But after I bypassed this check, the transport is created almost every time, so it works way much better.
Thank you!

closed 2023-03-13 by BluezTestBot