bluez/bluez #486: LE Audio transport state change issue

silviubarbulescu, 2023-03-03:
Hello all,
I have an issue testing the LE audio support in Pipewire. I have a setup with two virtual machines(central and peripheral) configured with Ubuntu 22.04 (Kernel 6.2), BlueZ master, enabled BlueZ Experimental ISO socket, and Pipewire master with LC3 support enabled and LE controller. Both central and peripheral have PAC_SINK_UUID and PAC_SOURCE_UUID endpoints loaded.  I created a Bluetooth connection using bluetoothctl and the transports are created with success. I use the paplay command to play a wav from the central to peripheral, I see that a bidirectional  CIS is created and data sent from the central to the peripheral but I see a lot of data (mostly 0x00) from the peripheral to the central (and the microphone pictogram appears on the peripheral PC as it tries to record data ) as it can be seen on the Helvum graph (peripheral)
[image]

I asked this question to the pipewire community and they were wondering if this is ok.
"The reason seems to be that BlueZ changes both source and sink to streaming state, when one of them is started. This is probaly connected to it being bidirectional CIS stream. This probably comes as we use auto-allocation of the CIS ids, but it's not clear BlueZ should do it like this.
The peripheral side Pipewire works correctly: we see BlueZ moving the two transports to pending state, and then we have to serve them."

Looking at the logs(pipewire and BlueZ) I can see that both transports change the state together because they are linked. 
[log_bluez_cen.zip](https://github.com/bluez/bluez/files/10882233/log_bluez_cen.zip)
[log_bluez_pe.zip](https://github.com/bluez/bluez/files/10882234/log_bluez_pe.zip)
[log_pipewire_cen.zip](https://github.com/bluez/bluez/files/10882235/log_pipewire_cen.zip)
[log_pipewire_pe.zip](https://github.com/bluez/bluez/files/10882236/log_pipewire_pe.zip)
 

In my case, because I have PAC_SINK_UUID and PAC_SOURCE_UUID endpoints loaded on both central and peripheral. And because linked transports change their state at the same time, data is sent from both sides. I don't think this case is handled correctly.

Am I wrong?

Vudentz, 2023-03-03:
@silviubarbulescu the linked transports means they use the same CIS, not sure if you have a problem with that and would like to have each direction using a different CIS? When transport use the same CIS the linked transport would attempt to enable when any of the transport is Acquired, note that it appears that some controllers seem to produce empty packets perhaps that is what you appear to be seeing?

pv, 2023-03-05:
Yes, it's the linked transports. Pipewire as BAP client can select different CIS in SelectProperties to avoid it.

One question here is whether it is required behavior that ASEs configured to a bidirectional CIS shall or should be enabled/disabled at the same time. I did not find it said directly in BAP on one reading, only two related points are mentioned in BAP v1.0 Sec. 5.6.5 and 5.6.5.1, which appear to imply server should permit it: "Termination of a bidirectional CIS should not occur if an ASE configured with the CIS_ID for that CIS is in the Enabling state or the Streaming state." But that's maybe a bit weak to rely on.

pv, 2023-03-05:
However, multiple CIS connections to the same device don't seem to work right now in kernel. I get `Bluetooth: conn->sk already set` from __iso_chan_add. It happens because hci_bind_bis only considers the destination address when looking up whether there is an existing connection. Something naive like [0001-Bluetooth-ISO-fix-multiple-CIS-connections-to-same-d.patch.txt](https://github.com/bluez/bluez/files/10892598/0001-Bluetooth-ISO-fix-multiple-CIS-connections-to-same-d.patch.txt) is also not enough, would need to take a closer look why...

Vudentz, 2023-03-05:
> However, multiple CIS connections (which is what you get if you give different CIS ids to bluez in the SelectProperties for sink and the one for source) to the same device don't seem to work right now in kernel. I get `Bluetooth: conn->sk already set` from __iso_chan_add. It happens because hci_bind_bis only considers the destination address when looking up whether there is an existing connection. Something naive like [0001-Bluetooth-ISO-fix-multiple-CIS-connections-to-same-d.patch.txt](https://github.com/bluez/bluez/files/10892598/0001-Bluetooth-ISO-fix-multiple-CIS-connections-to-same-d.patch.txt) is also not enough, would need to take a closer look why...

Yep, currently is no possible to setup multiple CIS to the same device, it is on my TODO list.

Vudentz, 2023-03-05:
> Pipewire as BAP client can select different CIS for sink/source in SelectProperties to avoid this, and it maybe should always do it, as server support for that appears mandatory in BAP and on application level the linked behavior is not useful for PW.
> 
> One question here is whether it is required behavior that ASEs configured to a bidirectional CIS shall or should be enabled/disabled at the same time. I did not find it said directly in BAP on one reading, only two related points are mentioned in BAP v1.0 Sec. 5.6.5 and 5.6.5.1, which appear to imply server should permit it: "Termination of a bidirectional CIS should not occur if an ASE configured with the CIS_ID for that CIS is in the Enabling state or the Streaming state." But that's maybe a bit weak to rely on.

That is more of an optimization since a CIS maps to a single socket so it is normally better that Acquire would return the fd which is capable of doing read/write operations, that is actually the intention of saying the transports are linked so the client don't have to acquire the same fd twice.

pv, 2023-03-05:
> That is more of an optimization since a CIS maps to a single socket so it is normally better that Acquire would return the fd which is capable of doing read/write operations, that is actually the intention of saying the transports are linked so the client don't have to acquire the same fd twice.

I think if there is no need to do it on protocol level, it would better to leave acquiring both transports to the client. The issue is that setting both ASEs to streaming state requests BAP server to start recording even though the client is not interested in reading the audio data it sends.

Vudentz, 2023-03-05:
> > That is more of an optimization since a CIS maps to a single socket so it is normally better that Acquire would return the fd which is capable of doing read/write operations, that is actually the intention of saying the transports are linked so the client don't have to acquire the same fd twice.
> 
> I think if there is no need to do it on protocol level, it would better to leave acquiring both transports to the client. The issue is that setting both ASEs to streaming state when acquiring only eg. sink requests BAP server to start recording even though the client is not interested in reading the audio data it sends.

Well HCI_Create_CIS, which is what Acquire will end up, doing _must_ setup for both direction if that is how it was programmed in the CIG, otherwise when you acquire only one direction we would have to reprogram the CIG to only direction and once the other direction is acquires we would then have to teardown the socket so it would be even harder to keep things working at the socket level, we can however do enable one ASE at time, but if they are still using the same CIS it once again will use HCI_Create_CIS that enables both directions even though the ASE may not be enabled. So is really a problem of having the microphone open in such situation, note that at CIS level both direction would be enabled simultaneously, but you can always mute if you don't want to use it.

pv, 2023-03-05:
> Well HCI_Create_CIS, which is what Acquire will end up, doing _must_ setup for both direction if that is how it was programmed in the CIG, otherwise when you acquire only one direction we would have [...]

Yes, the CIS is bidirectional and running if either ASE is streaming that is clear. However, whether the ASE is enabled or not signals to the peripheral what the central is doing, and so the peripheral can know e.g. if it needs to activate a mic and light up a microphone light indicating the mic is active, and the peripheral might not have a mute button. (For the convenience aspect, I think requiring separate acquires would be more convenient for client since it does not need special casing.)

But I don't see strong statements in the specs that this is something that is supposed to be supported way to use these, so may not be safe to rely on.