bluez/bluez #488: Bluetooth joystick pairs, connects but not reconnect

ericitaquera, 2023-03-13:
Hello!

I have a wireless bluetooth joystick. 

I´m able to discover pair and connect to it. But, if I turn it off and turn it on again, it connects and disconnects right after.

Steps:

I turn my controller bluetooth discover, go bluetoothctl > scan on

```
root@raspberrypi:~# bluetoothctl
Agent registered
[CHG] Controller B8:27:EB:3F:7A:73 Pairable: yes
[bluetooth]# scan on
Discovery started
[CHG] Controller B8:27:EB:3F:7A:73 Discovering: yes
[NEW] Device 5C:73:72:B8:D4:DF 5C-73-72-B8-D4-DF
[NEW] Device F7:D9:CA:15:F6:9C Amazfit GTS2 mini
[NEW] Device 30:2C:74:EE:98:16 Wireless Controller
```
I´m able to pair and connect to it

```
[bluetooth]# pair 30:2C:74:EE:98:16
Attempting to pair with 30:2C:74:EE:98:16
[CHG] Device 30:2C:74:EE:98:16 Connected: yes
[CHG] Device 30:2C:74:EE:98:16 Bonded: yes
[CHG] Device 30:2C:74:EE:98:16 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device 30:2C:74:EE:98:16 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 30:2C:74:EE:98:16 ServicesResolved: yes
[CHG] Device 30:2C:74:EE:98:16 Paired: yes
Pairing successful
[CHG] Device 30:2C:74:EE:98:16 WakeAllowed: yes
[CHG] Device 30:2C:74:EE:98:16 ServicesResolved: no
[CHG] Device 30:2C:74:EE:98:16 Connected: no
[bluetooth]# connect 30:2C:74:EE:98:16
Attempting to connect to 30:2C:74:EE:98:16
[CHG] Device 30:2C:74:EE:98:16 Connected: yes
Connection successful
[CHG] Device 30:2C:74:EE:98:16 ServicesResolved: yes
[Wireless Controller]#
```
Then I trust it

```
[Wireless Controller]# trust 30:2C:74:EE:98:16
[CHG] Device 30:2C:74:EE:98:16 Trusted: yes
Changing 30:2C:74:EE:98:16 trust succeeded
[Wireless Controller]# info 30:2C:74:EE:98:16
Device 30:2C:74:EE:98:16 (public)
        Name: Wireless Controller
        Alias: Wireless Controller
        Class: 0x00002508
        Icon: input-gaming
        Paired: yes
        Bonded: yes
        Trusted: yes
        Blocked: no
        Connected: yes
        WakeAllowed: yes
        LegacyPairing: no
        UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
        UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v054Cp09CCd0100
        RSSI: -53
[Wireless Controller]#
```
It works like a charm, until I disconnect it. It never reconnects. As follows:

[... 197 lines cut ...]
        Handle: 11 Address: 30:2C:74:EE:98:16 (OUI 30-2C-74)
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                            {0x0001} [hci0] 14.542618
        BR/EDR Address: 30:2C:74:EE:98:16 (OUI 30-2C-74)
        Reason: Connection terminated by local host (0x02)
< HCI Command: Write Scan Enable (0x03|0x001a) plen 1                                                             #32 [hci0] 14.633701
        Scan enable: Page Scan (0x02)
> HCI Event: Command Complete (0x0e) plen 4                                                                       #33 [hci0] 14.634151
      Write Scan Enable (0x03|0x001a) ncmd 1
        Status: Success (0x00)
```

I´m bluez compiled on a raspberry pi 3
```
[bluetooth]# version
Version 5.66
[bluetooth]#

root@raspberrypi:~# cat /sys/firmware/devicetree/base/model
Raspberry Pi 3 Model B Rev 1.2
```

Please let me know what further info I could provide.

Thanks in advance

grinya007, 2023-03-20:
Was just about to report the same issue. My two game controllers are clones of Sony Dualshock 4. Both work perfectly when initially paired (recognized by steam and games, so no complaints on that matter). But after I disconnect them or reboot my linux box, they never connect again (connect briefly and then disconnect). I tried all combinations of two different bluetooth adapters (intel ax200 and the onboard broadcom of a Macbook Air 2013) and some linux distros (Fedora 37, Fedora 38b, Ubuntu 18.04, 20.04, 22.04). The behavior is the same (as it is described in the original issue by @ericitaquera). I also have just built the latest release of bluez to try that out (to see if these changes that now appear to be merged helped https://patchwork.kernel.org/project/bluetooth/patch/20220802163441.2522333-1-luiz.dentz@gmail.com/), no difference. Please help! I built a steam machine and bought two controllers as a BD present for my son (which is on the 6th of April, the time is running out).

grinya007, 2023-03-21:
I discovered a very weird workaround that works for me
@ericitaquera please try this:
# pair/trust/connect a gamepad (either in bluetoothctl or through the UI) make sure it works as you would always expect it to
# disconnect the gamepad or restart bluetooth (`sudo systemctl restart bluetooth`)
# try to connect and verify that the original problem exists (i.e. that it doesn't connect)
# `vim /etc/bluetooth/main.conf` -> uncomment `ControllerMode = dual` and change it to `ControllerMode = le` -> save config
# `sudo systemctl restart bluetooth`
# `vim /etc/bluetooth/main.conf` -> change `ControllerMode = le` to `ControllerMode = bredr` -> save config
# `sudo systemctl restart bluetooth`
# Now try to connect the gamepad again

grinya007, 2023-03-23:
So, I spent about 6 hours hacking the C code of bluez, findings:
1. The only place where the information about the previous `ControllerMode` is stored is the bluetooth adapter itself.
2. When the daemon has been started and stopped with `ControllerMode = dual` (which is the default), during the subsequent start the state of the stored adapter settings is:
```
0 MGMT_SETTING_ADVERTISING
1 MGMT_SETTING_BONDABLE
1 MGMT_SETTING_BREDR
0 MGMT_SETTING_CIS_CENTRAL
0 MGMT_SETTING_CIS_PERIPHERAL
0 MGMT_SETTING_CONFIGURATION
0 MGMT_SETTING_CONNECTABLE
0 MGMT_SETTING_DEBUG_KEYS
0 MGMT_SETTING_DISCOVERABLE
0 MGMT_SETTING_FAST_CONNECTABLE
0 MGMT_SETTING_HS
1 MGMT_SETTING_LE
0 MGMT_SETTING_LINK_SECURITY
0 MGMT_SETTING_PHY_CONFIGURATION
0 MGMT_SETTING_POWERED
0 MGMT_SETTING_PRIVACY
1 MGMT_SETTING_SECURE_CONN
0 MGMT_SETTING_SSP
0 MGMT_SETTING_STATIC_ADDRESS
0 MGMT_SETTING_WIDEBAND_SPEECH
```
3. When it's set to `ControllerMode = le`, started, stopped, set back to `dual`, during the subsequent start the state of the stored adapter settings is:
```
0 MGMT_SETTING_ADVERTISING
1 MGMT_SETTING_BONDABLE
0 MGMT_SETTING_BREDR
0 MGMT_SETTING_CIS_CENTRAL
0 MGMT_SETTING_CIS_PERIPHERAL
0 MGMT_SETTING_CONFIGURATION
0 MGMT_SETTING_CONNECTABLE
0 MGMT_SETTING_DEBUG_KEYS
0 MGMT_SETTING_DISCOVERABLE
0 MGMT_SETTING_FAST_CONNECTABLE
0 MGMT_SETTING_HS
1 MGMT_SETTING_LE
0 MGMT_SETTING_LINK_SECURITY
0 MGMT_SETTING_PHY_CONFIGURATION
0 MGMT_SETTING_POWERED
0 MGMT_SETTING_PRIVACY
1 MGMT_SETTING_SECURE_CONN
0 MGMT_SETTING_SSP
0 MGMT_SETTING_STATIC_ADDRESS
0 MGMT_SETTING_WIDEBAND_SPEECH
```
4. The only difference is MGMT_SETTING_BREDR which is turned off in the second case (which is rather obvious). But, for whatever reason, when the daemon starts in `dual` mode and the adapter is still set to `le` (which is updated right when the process starts), the connection of my previously paired gamepads works as it is supposed to.
5. I tried hard to tell the adapter to always remain in that second state when the daemon is not running (set_mode before termination), so that every time the daemon starts, the adapter appears as if its previous usage was in `le` mode but I couldn't do that. I tried setting mode before poweroff, after poweroff, in the callback when poweroff is confirmed, but none worked. 
 
So I ended up with a super dirty workaround:
`sudo vim /etc/bluetooth/starter.sh`
```bash
#!/bin/bash

/usr/lib/bluetooth/bluetoothd -f /etc/bluetooth/main_le.conf &
export CPID=$!
sleep 2
kill $CPID
sleep 1
echo Starting...
exec /usr/lib/bluetooth/bluetoothd
```
`sudo chmod +x /etc/bluetooth/starter.sh`
`sudo systemctl stop bluetooth`
`sudo systemctl disable bluetooth`
`sudo mkdir /etc/systemd/system/bluetooth.service.d/`
`sudo vim /etc/systemd/system/bluetooth.service.d/override.conf`
```service
[Service]
ExecStart=
ExecStart=/etc/bluetooth/starter.sh
Type=notify
```
`sudo systemctl daemon-reload`
`sudo systemctl enable bluetooth`
`sudo systemctl start bluetooth`

I hope this helps
PS I'm not an expert in systemd so some of the above might be redundant but at least is works

ericitaquera, 2023-03-23:
> 

Hello @grinya007.

On my system the workarouds didn´t work (both manually and scripted resulted the same previous behaviors: the controllers pair and connect but never reconnects after disconnection).

New info: now I this error on btmon:

`< HCI Command: Disconnect (0x01|0x0006) plen 3                                                                                                       #27 [hci0] 10.069454
        Handle: 12 Address: 30:2C:74:EE:98:16 (OUI 30-2C-74)
        Reason: Authentication Failure (0x05)
`

Regards

grinya007, 2023-03-23:
I see, this is instead of `Result: Connection refused - security block (0x0003)` that you saw before. I observed both variants while digging this out. Both originate from the bluetooth adapter, and bluetooth daemon only reports these messages post-factum (i.e. authorisation request doesn't even make its way through to the daemon). Anyway, I suggest trying once more:
1. `bluetoothctl`
2. `remove 30:2C:74:EE:98:16`
3. `scan on`
4. turn your gamepad into pairing mode
5. `trust 30:2C:74:EE:98:16`
6. `pair 30:2C:74:EE:98:16`
7. `connect 30:2C:74:EE:98:16`
This order of commands is important for whatever reason.
Now restart the service into le mode and back to dual mode and try to connect.
I'm sure it will eventually work

ericitaquera, 2023-03-24:
Thanks for the effort, but I´m having no luck. =)

With bluetooth in dual mode, I removed the device.

```
[bluetooth]# remove 30:2C:74:EE:98:16
[DEL] Device 30:2C:74:EE:98:16 Wireless Controller
Device has been removed
[bluetooth]#
```

Put the controller in pairing mode and turned scan on and the controller was found eventually.

```
[bluetooth]# scan on
Discovery started
[CHG] Controller B8:27:EB:3F:7A:73 Discovering: yes
[NEW] Device 44:B8:9F:D6:87:63 44-B8-9F-D6-87-63
[CHG] Device 44:B8:9F:D6:87:63 Connected: yes
[CHG] Device 44:B8:9F:D6:87:63 Connected: no
[DEL] Device 44:B8:9F:D6:87:63 44-B8-9F-D6-87-63
[NEW] Device 44:B8:9F:D6:87:63 44-B8-9F-D6-87-63
[NEW] Device F7:D9:CA:15:F6:9C Amazfit GTS2 mini
[CHG] Device F7:D9:CA:15:F6:9C UUIDs: 00001530-0000-3512-2118-0009af100700
[CHG] Device F7:D9:CA:15:F6:9C UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Device F7:D9:CA:15:F6:9C UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Device F7:D9:CA:15:F6:9C UUIDs: 00001802-0000-1000-8000-00805f9b34fb
[CHG] Device F7:D9:CA:15:F6:9C UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Device F7:D9:CA:15:F6:9C UUIDs: 0000180d-0000-1000-8000-00805f9b34fb
[CHG] Device F7:D9:CA:15:F6:9C UUIDs: 0000180f-0000-1000-8000-00805f9b34fb
[CHG] Device F7:D9:CA:15:F6:9C UUIDs: 00001811-0000-1000-8000-00805f9b34fb
[CHG] Device F7:D9:CA:15:F6:9C UUIDs: 00001812-0000-1000-8000-00805f9b34fb
[CHG] Device F7:D9:CA:15:F6:9C UUIDs: 0000fee0-0000-1000-8000-00805f9b34fb
[CHG] Device F7:D9:CA:15:F6:9C UUIDs: 0000fee1-0000-1000-8000-00805f9b34fb
[CHG] Device F7:D9:CA:15:F6:9C ServicesResolved: yes
[CHG] Device F7:D9:CA:15:F6:9C Paired: yes
[CHG] Device F7:D9:CA:15:F6:9C Appearance: 0x03c1
[CHG] Device F7:D9:CA:15:F6:9C Icon: input-keyboard
[CHG] Device F7:D9:CA:15:F6:9C Modalias: bluetooth:v0157p0049d0101
[CHG] Device 44:B8:9F:D6:87:63 Connected: yes
[CHG] Device 44:B8:9F:D6:87:63 Connected: no
[DEL] Device 44:B8:9F:D6:87:63 44-B8-9F-D6-87-63
[NEW] Device 7A:74:39:5A:9C:F4 7A-74-39-5A-9C-F4
[NEW] Device 44:B8:9F:D6:87:63 44-B8-9F-D6-87-63
[CHG] Device 44:B8:9F:D6:87:63 Connected: yes
[CHG] Device 44:B8:9F:D6:87:63 Connected: no
[DEL] Device 44:B8:9F:D6:87:63 44-B8-9F-D6-87-63
[CHG] Device 7A:74:39:5A:9C:F4 Connected: yes
[CHG] Device 7A:74:39:5A:9C:F4 Connected: no
[DEL] Device 7A:74:39:5A:9C:F4 7A-74-39-5A-9C-F4
[NEW] Device 44:B8:9F:D6:87:63 44-B8-9F-D6-87-63
[NEW] Device 5D:1A:26:F7:D4:6B 5D-1A-26-F7-D4-6B
[CHG] Device 44:B8:9F:D6:87:63 RSSI: -74
[CHG] Device 5D:1A:26:F7:D4:6B RSSI: -82
[NEW] Device 30:2C:74:EE:98:16 Wireless Controller  <<<<<<<<<<<<<<<<<<<<<<<<
```

So i trusted/pair/connected it successfully

```
[bluetooth]# trust 30:2C:74:EE:98:16
[CHG] Device 30:2C:74:EE:98:16 Trusted: yes
Changing 30:2C:74:EE:98:16 trust succeeded
[bluetooth]# pair 30:2C:74:EE:98:16
Attempting to pair with 30:2C:74:EE:98:16
[CHG] Device 30:2C:74:EE:98:16 Connected: yes
[CHG] Device 30:2C:74:EE:98:16 UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device 30:2C:74:EE:98:16 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 30:2C:74:EE:98:16 ServicesResolved: yes
[CHG] Device 30:2C:74:EE:98:16 Paired: yes
[... 206 lines cut ...]
        Reason: Authentication Failure (0x05)
> HCI Event: Command Status (0x0f) plen 4                                                                                                                                                                             #137 [hci0] 102.422614
      Disconnect (0x01|0x0006) ncmd 1
        Status: Success (0x00)
> HCI Event: Disconnect Complete (0x05) plen 4                                                                                                                                                                        #138 [hci0] 102.510897
        Status: Success (0x00)
        Handle: 11
        Reason: Connection Terminated By Local Host (0x16)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                                                                 {0x0002} [hci0] 102.511026
        BR/EDR Address: 30:2C:74:EE:98:16 (OUI 30-2C-74)
        Reason: Connection terminated by local host (0x02)
@ MGMT Event: Device Disconnected (0x000c) plen 8                                                                                                                                                                 {0x0001} [hci0] 102.511026
        BR/EDR Address: 30:2C:74:EE:98:16 (OUI 30-2C-74)
        Reason: Connection terminated by local host (0x02)
< HCI Command: Write Scan Enable (0x03|0x001a) plen 1                                                                                                                                                                 #139 [hci0] 102.582549
        Scan enable: Page Scan (0x02)
> HCI Event: Command Complete (0x0e) plen 4                                                                                                                                                                           #140 [hci0] 102.583954
      Write Scan Enable (0x03|0x001a) ncmd 1
        Status: Success (0x00)
@ MGMT Command: Pair Device (0x0019) plen 8                                                                                                                                                                       {0x0001} [hci0] 109.689126
        LE Address: F7:D9:CA:15:F6:9C (OUI F7-D9-CA)
        Capability: DisplayYesNo (0x01)
``` 

Thanks for the support.

tuxrinku, 2023-04-21:
I have the same issue, except I have both an Xbox one controller and a Dualshock 4 controller. There is no problem with the xbox controller, but the dualshock 4 keeps disconnecting after being turned on. After trying the mentioned workaround, the dualshock 4 stays connected but the xbox controller won't connect anymore. 

So basically :
bredr mode = Dualshock 4 works fine, but not the xbox controller
dual mode = Xbox controller works fine, but not the dualshock 4

rodericktaylor, 2023-07-03:
I just purchased a Cooler Master Storm Controller and was having the same issue.  Using Ubuntu 22.04 with the official bluez 5.64, I can pair and it would work fine until I disconnected after pair. As soon as it was turned off and back on again, it would not automatically connect.  Had to pair again to get a connection.

grinya007's wierd workaround seems to work for this controller as well. Change ControllerMode from dual, to le restart bluetooth.service, then change it back to dual and restart again and it connects fine.

yahyaomer, 2024-03-30:
I am having the same issues with my gamepad (knock-off DualShock 4 v2) as well.
I pair and connect my device using `bluetoothctl` with the following steps
```
scan on
trust <MAC_ADDRESS>
pair <MAC_ADDRESS>
connect <MAC_ADDRESS>
```
and everything works fine... until I disconnect the device and try to reconnect it.

When I try to reconnect I get the `Reason: Authentication Failure (0x05)` error (monitoring with `btmon`).

The workaround suggested by @grinya007 did not resolve the issue for me. After numerous hours of trial and error, I found a (potentially inconvenient) workaround.

First, modify the configurations as follows:
```
#/etc/bluetooth/main.conf
[General]
JustWorksRepairing = always
```
and
```
#/etc/bluetooth/input.conf
[General]
ClassicBondedOnly=false
LEAutoSecurity=false
```

Then, pair and connect the device using the aforementioned steps. Disconnect and attempt to reconnect by turning on the gamepad (which fails as usual).

Then execute:
```
sudo systemctl restart bluetooth
sudo btmgmt ssp off
sudo btmgmt le off
```
and power on the controller. In my case the controller reconnects immediately without any issues.

BUT, when `ssp (Secure Simple Pairing)` is off, i cannot pair any new devices.

I am not actually sure if `btmgmt le off` is necessary.

These are my observations and after many hours of trial and error, and I could not find any permanent solutions to this problem.
Hope this workaround works for you.

Maybe this  problem is related with other open issues #673 OR #777.

I am using TP-Link UB500 bluetooth adapter, but tried this on 2 other computers with built in bluetooth adapters, the results are the same.
My bluez version is 5.72 and I am on Fedora 39 with latest updates.

nfp0, 2024-05-11:
Setting `FastConnectable = true` on `/etc/bluetooth/main.conf` fixed that problem for me on my Nintendo Switch controllers and now they always reconnect instantaneously with no need to re-pair.

capsyweb, 2024-09-25:
I was also having this issue, tried all of the above with no sucess, then I searched some more and found this [](https://gist.github.com/AndBondStyle/bc178789a441ce73b51e1e0637393aac) it worked for me

privatecore, 2025-01-16:
Have the same issue like @grinya007 and workaround from comment https://github.com/bluez/bluez/issues/488#issuecomment-1480463191 is worked:
1. `sudo sed -i -E 's/#?ControllerMode = [a-z]+/ControllerMode = le/g' /etc/bluetooth/main.conf`
2. `sudo systemctl restart bluetooth.service`
3. `sudo sed -i -E 's/#?ControllerMode = [a-z]+/ControllerMode = dual/g' /etc/bluetooth/main.conf`
4. `sudo systemctl restart bluetooth.service`

And after that controller (JPD-Wireless-Thrillershock PC/PS4) reconnects w/o problem. Tried with Ubuntu 22.04 and SteamOS 3.6.20.

jacobwhall, 2025-04-10:
The workaround described by @yahyaomer above worked for my DualShock 4 v2 controller. I'm learning it must be a clone, which is probably why it was so cheap on eBay 😂

Here is the VID and PID for this device, which match those listed for the official DualShock 4 online:
- VID: 0x054C
- PID: 0x09CC

I agree that keeping Secure Simple Pairing off is not a viable long-term solution.

baracoder, 2025-06-09:
Still experiencing it with a dualsense controller, Intel Corp. AX210 Bluetooth module, kernel 6.15.0 and bluez 5.80

thexwarr1orsxnew, 2025-10-01:
status update, by the moment disabling Secure Simple Pairing, which means the Bluetooth controller will fall back to legacy PIN-based pairing works to reconnect controller model DATA FROG P03 but does not solve the issue of BlueZ

it can be done with this command works until you reboot or reload Bluetooth:

sudo btmgmt ssp off

Disabling SSP reduces security, since legacy pairing with fixed PINs is much easier to brute-force. so is not recommended to make it persistent adding udev rules.

closed 2025-11-09 by github-actions[bot]

danmilon, 2026-08-15:
On a knockoff controller (mine is branded as Vermillion Mingpin MB-S820) that was experiencing this problem, after a firmware upgrade through the companion android app (Shooting Plus V3) reconnections work fine.