bluez/bluez #490: Having problems creating LE Audio transport

silviubarbulescu, 2023-03-16:
Hello,
I have a setup running Pipewire over BlueZ on both central and Peripheral and trying to send an audio sample file with paplay.

Often, I encounter the issue of no transport created on central. During debugging, I found that the problem is that in bap_config at line ep->id = bt_bap_stream_config(ep->stream, &ep->qos, ep->caps, config_cb, ep); the endpoint id is set with the value returned by bt_bap_stream_config. And the expectation is that when the  ATT_MULTIPLE_HANDLE_VALUE_NTF is received from the peripheral the ep->id is set to 0 in config_cb.
In my case, as you can see in the attached log when the ATT_MULTIPLE_HANDLE_VALUE_NTF  is received () the config_cb function is not called. 
So in bap_state  on BT_BAP_STREAM_STATE_CONFIG ep->id is not 0. 

Is my my analysis correct?

I am using Bluez master () with Kernel Experimental enabled, Pipewire master 
compiled with lc3, and Kernel 6.3.0 rc2. Default endpoint configuration for BAP Sink and BAP Source.

When I used the branch suggested in ticket #485 I had better results in creating the transport.

Attached Bluez logs from central and peripheral  [no trasnport.zip](https://github.com/bluez/bluez/files/10991707/no.trasnport.zip)

silviubarbulescu, 2023-03-16:
Another thing I've seen is that in the case when the transport is created, the write command is sent after all the write requests (to enable notifications) have been sent and responses received. In the case of the transport, not being created the write command and write requests/responses are interspersed.

Vudentz, 2023-03-16:
@silviubarbulescu Hmm I though I fixed this with the likes of:

https://github.com/bluez/bluez/commit/3842320f450e0b40b205b8fd0ce13b8821d49b51

Does this happens after reconnection as well? It not it could be that there is still this bug where the address is updated and generating different instances of bt_att/bt_gatt_client that don't work together since they believe to be communicating with different devices.

Vudentz, 2023-03-16:
@silviubarbulescu looks like there was a bug with the ordering of GATT request, GATT subscription (aka. Write to CCCD) was done later than BAP Codec Configuration so it misses the notification, the above should fix this behavior.

silviubarbulescu, 2023-03-16:
Thanks, @Vudentz, for the prompt response. I will test your fix shortly.

closed 2023-03-17 by BluezTestBot