bluez/bluez #491: Opcode 0x200d failed  /  request failed to create LE connection: err -110

Mausy5043, 2023-03-19:
This issue looks very similar to #340. But I'm creating a new issue in order not to hijack the other one. I'm also not sure if it belongs here. So, feel free to close if off-topic / off-repo. ;-)

I'm using a Raspberry Pi (`Linux rbair 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64 GNU/Linux`) running [DietPi](https://github.com/MichaIng/DietPi) (a Debian flavour; bullseye) with Bluetooth enabled.  
I'm trying to communicate with a number of Xiaomi LYWSD03MMC devices. To do that I'm using [bluepy3](https://github.com/Mausy5043/bluepy3) which is, basically, a [bluepy](https://github.com/IanHarvey/bluepy) rip-off that has some updates & fixes to run under Python3.7+ and compiles the `bluepy(3)-helper` against tag 5.66 of [bluez](https://github.com/bluez/bluez).

Excerpt of logging:
```
$ journalctl | grep -i bluetooth | more
Mar 19 11:31:26 rbair systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Mar 19 11:31:33 rbair kernel: Bluetooth: Core ver 2.22
Mar 19 11:31:33 rbair kernel: NET: Registered PF_BLUETOOTH protocol family
Mar 19 11:31:33 rbair kernel: Bluetooth: HCI device and connection manager initialized
Mar 19 11:31:33 rbair kernel: Bluetooth: HCI socket layer initialized
Mar 19 11:31:33 rbair kernel: Bluetooth: L2CAP socket layer initialized
Mar 19 11:31:33 rbair kernel: Bluetooth: SCO socket layer initialized
Mar 19 11:31:33 rbair kernel: Bluetooth: HCI UART driver ver 2.3
Mar 19 11:31:33 rbair kernel: Bluetooth: HCI UART protocol H4 registered
Mar 19 11:31:33 rbair kernel: Bluetooth: HCI UART protocol Three-wire (H5) registered
Mar 19 11:31:33 rbair kernel: Bluetooth: HCI UART protocol Broadcom registered
Mar 19 11:31:33 rbair systemd[1]: Started Configure Bluetooth Modems connected by UART.
Mar 19 11:31:34 rbair systemd[1]: Starting Raspberry Pi bluetooth helper...
Mar 19 11:31:34 rbair systemd[1]: Finished Raspberry Pi bluetooth helper.
Mar 19 11:31:34 rbair systemd[1]: Starting Bluetooth service...
Mar 19 11:31:34 rbair bluetoothd[684]: Bluetooth daemon 5.55
Mar 19 11:31:34 rbair systemd[1]: Started Bluetooth service.
Mar 19 11:31:34 rbair systemd[1]: Reached target Bluetooth.
Mar 19 11:31:34 rbair bluetoothd[684]: Starting SDP server
Mar 19 11:31:34 rbair kernel: Bluetooth: BNEP (Ethernet Emulation) ver 1.3
Mar 19 11:31:34 rbair kernel: Bluetooth: BNEP filters: protocol multicast
Mar 19 11:31:34 rbair kernel: Bluetooth: BNEP socket layer initialized
Mar 19 11:31:34 rbair dbus-daemon[394]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service'
requested by ':1.3' (uid=0 pid=684 comm="/usr/libexec/bluetooth/bluetoothd ")
Mar 19 11:31:34 rbair bluetoothd[684]: Bluetooth management interface 1.22 initialized
Mar 19 11:31:34 rbair kernel: Bluetooth: MGMT ver 1.22
Mar 19 11:31:34 rbair bluetoothd[684]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Mar 19 11:31:34 rbair bluetoothd[684]: sap-server: Operation not permitted (1)
Mar 19 11:31:35 rbair bluetoothd[684]: Failed to set privacy: Rejected (0x0b)
Mar 19 11:37:06 rbair kernel: Bluetooth: hci0: Opcode 0x200d failed: -110
Mar 19 11:37:06 rbair kernel: Bluetooth: hci0: request failed to create LE connection: err -110
```

The last two lines repeatedly popup in the logging. I've been able to trace this to when the connection to a device fails to be made. Mostly upon retry the connection succeeds.

Some additional info:
```
$ hciconfig
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:90:4F:F5  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING
	RX bytes:523997 acl:15607 sco:0 events:11238 errors:0
	TX bytes:252027 acl:14876 sco:0 commands:2493 errors:0
```

```
$ bluetoothctl
Agent registered
[CHG] Controller B8:27:EB:90:4F:F5 Pairable: yes
[bluetooth]# version
Version 5.55
[bluetooth]# show
Controller B8:27:EB:90:4F:F5 (public)
	Name: rbair
	Alias: rbair
	Class: 0x00000000
	Powered: yes
	Discoverable: no
	DiscoverableTimeout: 0x000000b4
	Pairable: yes
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v1D6Bp0246d0537
	Discovering: no
	Roles: central
	Roles: peripheral
Advertising Features:
	ActiveInstances: 0x00 (0)
	SupportedInstances: 0x05 (5)
	SupportedIncludes: tx-power
	SupportedIncludes: appearance
	SupportedIncludes: local-name
```

Vudentz, 2023-03-21:
@Mausy5043 try capturing the HCI trace with btmon, perhaps there is some error with the command itself.

Mausy5043, 2023-03-23:
`btmon` generates an enormous amount of output.   
```
$ sudo btmon -a btmon.txt
Bluetooth monitor ver 5.55
Creating new device for unknown index
Unknown opcode 14
Unknown opcode 14
Unknown opcode 16
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 15
Unknown opcode 14
Unknown opcode 16
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 15
Unknown opcode 14
Unknown opcode 16
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 15
Unknown opcode 14
Unknown opcode 16
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 15
Unknown opcode 14
Unknown opcode 16
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 15
Unknown opcode 14
Unknown opcode 16
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 15
Unknown opcode 14
Unknown opcode 16
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 15
Unknown opcode 14
Unknown opcode 16
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 15
Unknown opcode 14
Unknown opcode 16
Unknown opcode 17
Unknown opcode 17
Unknown opcode 17
Unknown opcode 15
Unknown opcode 17
Trace contains 1530 packets

Found BR/EDR controller with index 65535
  BD_ADDR 00:00:00:00:00:00
  0 commands
  0 events
  0 ACL packets
  0 SCO packets
  0 vendor diagnostics
  2 system notes
  0 user logs
  37 unknown opcodes

Found BR/EDR controller with index 0
  BD_ADDR B8:27:EB:90:4F:F5 (Cypress Semiconductor)
  83 commands
  375 events
  1005 ACL packets
  0 SCO packets
  0 vendor diagnostics
  0 system notes
  0 user logs
  25 unknown opcodes
```
I've put the output for one device [here](https://gist.github.com/Mausy5043/e961359d83f6dff576a8e8d40b510157)   
Anything to look out for?

Mausy5043, 2023-03-24:
@kroese : I was also starting to suspect the device. But troubleshooting is very hard as I can make heads nor tails of the (debugging) info I get returned from the bluetooth stack via my application nor from `btmon`. I've already replaced the battery of some of the devices. With not much success.

Mausy5043, 2023-03-26:
Opcode 0x200d is defined here:
https://github.com/bluez/bluez/blob/fb1c694100b2aff6ee1ad69d2a07c0f122e2e929/monitor/bt.h#L2066-L2080

https://github.com/bluez/bluez/blob/fb1c694100b2aff6ee1ad69d2a07c0f122e2e929/monitor/packet.c#L9534-L9535

This is used to create the BLE connection to the device using `le_create_conn_cmd` which is here:
https://github.com/bluez/bluez/blob/fb1c694100b2aff6ee1ad69d2a07c0f122e2e929/monitor/packet.c#L7055-L7089

It's not clear to me how or where the errors are generated
```
kernel: Bluetooth: hci0: Opcode 0x200d failed: -110
kernel: Bluetooth: hci0: request failed to create LE connection: err -110
```

But it seems most likely that these errors are due to the remote device not being able to connect, refusing to connect or taking too much time to answer the request. Usually the connection succeeds on retry and if it doesn't the higher-level application throws an error. For me, it would be nice if the kernel errors could be suppressed, or only generated if the connection fail persists after retrying.

Vudentz, 2023-03-26:
Ive sent a series taking care of these errors: https://patchwork.kernel.org/project/bluetooth/list/?series=733699

Mausy5043, 2023-03-26:
@Vudentz : Thanks. What are typical throughput times for such a change? When can I test this? 
```
$ apt list |grep installed |grep blue

bluetooth/stable,now 5.55-3.1+rpt2 all [installed]
bluez-firmware/stable,now 1.2-4+rpt10 all [installed,automatic]
bluez/stable,now 5.55-3.1+rpt2 arm64 [installed,automatic]
libbluetooth-dev/stable,now 5.55-3.1+rpt2 arm64 [installed]
libbluetooth3/stable,now 5.55-3.1+rpt2 arm64 [installed,automatic]
pi-bluetooth/stable,now 0.1.19 all [installed]
```

I'm assuming it's an update of the `bluetooth` package I'll need to keep an eye out for?

Vudentz, 2023-03-26:
@Mausy5043 it will have to go thru the stable process to reach the stable trees.

Mausy5043, 2023-03-30:
I've replaced a number of batteries and the amount of errors has gone down. But the batteries drain like crazy. The changes proposed by @Vudentz [in the series referneced here](https://github.com/bluez/bluez/issues/491#issuecomment-1484170779) seem a good direction. Hopefully they improve the situation further.

[image]

closed 2025-08-12 by github-actions[bot]