bluez/bluez #496: advertising headers added to wrong place in the "Row data"

bwomsm1, 2023-03-29:
I am sending an advertising ble message and successfully manage to see the adverting in android `nRF Connect` mobile app.

But when I open the blue `RAW` button in the app and looking at the `Raw data:` section, 
I noticed that the packed is being sent when the advertising headers are added to the end of the raw data instead of the beginning 
0x15FFA703641E610001BE000A5832D964030409000207**020106**

The correct string which I should see in the `Raw data:` is 
0x**020106**15FFA703641E610001BE000A5832D964030409000207

[image]

Vudentz, 2023-03-29:
> @grinya007 @Vudentz do you mind taking a look on the bug/question above? THANKS!

The advertising data is in LTV format, so it doesn't matter in which order it is placed, so len 0x02 type 0x01 (BT_AD_FLAGS) value 0x06 (BT_AD_FLAG_GENERAL | BT_AD_FLAG_NO_BREDR), that said it is probably a good idea to prepend in case of the flags but there is nothing wrong if we append either.

bwomsm1, 2023-03-31:
Thanks @Vudentz. we can close the ticket from my side

closed 2023-03-31 by bwomsm1