bluez/bluez #502: Connection Failed: br-connection-profile-unavailable

Susoiduts, 2023-04-07:
Hey,
I have problems establishing a Bluetooth connection with my Xiao Mi A3 smartphone using an ASUS Zen Book 13 with Linuxmint 21.1 Cinnamon. The pairing seems to work but when I try to connect the phone I get the error message:
`Connection Failed: br-connection-profile-unavailable`
Do you know how I can solve this?
If you need further information/logs let me know what you need and I will provide the information.

Vudentz, 2023-04-07:
@Susoiduts this normally happens when there is no matching profiles to be used, if you are planning to use for example A2DP make use if you have endpoints to act as Audio Sink, etc.

Susoiduts, 2023-04-07:
@Vudentz 
Hey, thank you for your quick response. I am not sure if am getting it correctly: Where can I configure/select the profile?
Is the problem on the side of my smartphone or the side of my laptop?

TheBizii, 2023-04-14:
I had the same issue, this seems to have fixed it for me:

```shell
systemctl --user start pulseaudio
```

UltraBlackLinux, 2023-08-19:
same  problem here, but I'm on pipewire and wireplumber...

Vulpesx, 2023-11-22:
I'm on Artix with pipewire and wireplumber.

Got this issue after a system update. I've had it before, but it magically resolved itself, but this time I've had it for like a month. 
I only get this error with ds4 controller, nothing else (that I've tried).

cmvanb, 2024-06-26:
I also have this problem with pipewire on Arch.

```
sudo systemctl restart bluetooth
```

Restarting the bluetooth service solves the issue, but this should be automatic.

Penguine998, 2024-08-15:
No luck for me. I'm having the same trouble on Kubuntu 24.04, pairing a HaritoraX 1.1B device- bluez detects it fine, but fails to connect with
`[HaritoraX1.1B-<serial#>]# Failed to connect: org.bluez.Error.NotAvailable br-connection-profile-unavailable`

Restarting the bluetooth service hasn't worked.

whale2186, 2025-04-21:
did anyone managed to fix it? (wireplumber pipewire)

reinux, 2025-07-08:
Likewise having this issue on a Pi 5 with pipewire/wireplumber.