bluez/bluez #504: systemd service fails when /var/lib/bluetooth is a symlink

jgero, 2023-04-14:
Hi,
I'd like to change the location of the directory where the connected devices are stored.
I'm currently playing around with a temporary root file system, which means I lose the bluetooth connections on reboot. Is there a setting to allow /var/lib/bluetooth to be a symlink or another way to move the location of that directory?

Thanks in advance for your time

Vudentz, 2023-04-17:
> Hi, I'd like to change the location of the directory where the connected devices are stored. I'm currently playing around with a temporary root file system, which means I lose the bluetooth connections on reboot. Is there a setting to allow /var/lib/bluetooth to be a symlink or another way to move the location of that directory?
> 
> Thanks in advance for your time
> 
> Edit: Here is the error:
> 
> ```
> Apr 14 23:00:28 nixps (uetoothd)[25990]: bluetooth.service: Failed to set up special execution directory in /var/lib: Not a directory
> Apr 14 23:00:28 nixps (uetoothd)[25990]: bluetooth.service: Failed at step STATE_DIRECTORY spawning /nix/store/sl9xzfjrrs3k47apx1zs533cji9625y4-bluez-5.66/libexec/bluetooth/bluetoothd: Not a directory
> ```

This is actually setup by systemd, not sure if there is something we can do to allow things like STATE_DIRECTORY to be a symlink.

jgero, 2023-04-30:
I ended up using the [NixOS impermanence module](https://nixos.wiki/wiki/Impermanence) which uses bind-mounts instead of symlinks. This works for me, so I'll close the issue.

closed 2023-04-30 by jgero