bluez/bluez #508: Unable to pair with bluetooth pedal

ChrisKinne, 2023-04-25:
Hello everyone,

I have an issue, i'm unable to pair my computer with blueooth music pedals. They identifies as a bluetooth keyboard and bluetoothctl ask me to type a PIN code. There is of course no way to type anything. 

I tried to run bluetoothctl with NoInputNoOutput agent with no luck the PIN is still asked (agent NoInputNoOutput before pair <DEVICEMAC>). I've also disabled/uninstalled all other bluetooth managing apps on the system to be sure there is no unwanted interaction caused by anything else like gnome control center or blueman.

On windows pairing just works out of the box, it takes 5 seconds to pair the device.  

Is there a way to really disable this PIN security ?

I'm currently using bluez 5.64 on Ubuntu 22.04 LTS

bgix, 2023-04-25:
"Just Works" generally just use 0000 as the PIN.   Have you tried that?

ChrisKinne, 2023-04-26:
Yes i've tried 0000, 000000 and 1234. When I do a pair with bluetoothctl it returns a PIN even with the NoInputNoOutput agent. I've tried to use it when i do a connect (after pair) and it ask for a PIN (it's there that I did try the previous other PINs) 
On windows nothing was asked.

ChrisKinne, 2023-04-28:
The problem is now solved. I changed the adapter for a new bluetooth 5 adapter, the old one was bt 2.0 and did not support SSP. Everything works fine, the device paired immediately. 
Hope this could help other people with the same issue.