bluez/bluez #509: BlueZ 5.64 has some bugs related to connection and rx-notify.

navinreddy23, 2023-04-27:
Hello,

I would like to report that a program I wrote to connect to a custom peripheral works fine on `BlueZ version 5.53` (Ubuntu 20.04) and `BlueZ version 5.55` (RPi 4B). But the same program fails to work as expected on `BlueZ version 5.64` (Ubuntu 22.04).

The problems I faced: 
1. Uninitiated disconnects (very frequent)
2. GATT Rx notify not working on some services.

After I removed `BlueZ 5.64` on Ubuntu and replaced with `5.53` (Compiled from source, it worked well)
This way I could verify that the issue was in 5.64 version.

Vudentz, 2023-04-28:
@navinreddy23 do you have the HCI trace (btmon)? Also what are you reference to as GATT RX notify? GATT Notification? Its probably a good idea to test with the latest BlueZ release since 5.64 is already quite old as well.

navinreddy23, 2023-04-29:
@Vudentz Thank you for the response!
Yes, I mean GATT notification. 
I verified with version 5.66 and it works as expected.

Thank you again for the suggestion! I will close the ticket.

closed 2023-04-29 by navinreddy23