bluez/bluez #510: Discovery stops after 10 seconds

TatriX, 2023-04-27:
Hi!
Sometime after upgrading from linux 5.19 bluetooth discovery started timing out for me.
Here's roughly what I do :
```
$ bluetoothctl
# menu scan
# uuids 0x0612 ;; set filter for my specific beacon
# back
# scan on
```
Discovery starts and everything works just fine. But after about 10 seconds discovery stops, meaning that I'm not seeing any data coming. Discovering property still is true, I see no errors or anything really in bluetoothd logs.
It seems I'm hitting some kind of timeout.
Essentially I'd like discovery to be actively runnig until I call StopDiscovery.

Thanks!

Vudentz, 2023-04-28:
@TatriX please include the HCI traces using btmon, also are you it stops or it just not able to find anything since you are setting a filter, also try using scan.pattern to see if that works when you provide a name/address.

TatriX, 2023-04-29:
Hi! Thank you for checking this out!

Please check attached files.
I've just recorded two sessions on two machines with the same ble device.
Both use the same distro.

`working-*` files are recorded on a machine with older kernel.
`broken-*` are recorded on a machine with newer kernel.

I've started getting this issue after I've upgraded from 5.19. If I
downgrade kernel on either machine everything works again.

To summarize what I observe:

- on older kernel with uuids filter enabled, discovery keeps going and
  when I send data from my ble device I can see it in bluetoothctl as
  well as I can get the data via bluetooth socket.

- on newer kernels data stops arriving after about 10
  seconds. Discovery seems to be still going because if I try to start
  it again via "scan on" I'm getting "Failed to start discovery:
  org.bluez.Error.InProgress"

I've also tried using
  [bluetooth]# pattern A4:34:F1:79:8A:F6
instead of
  [bluetooth]# uuids 0x0612
but the symptoms are the same - data stops coming.

I've tried stepping through bluetoothd code, but coudn't find anything
that could help me. I've also tried looking at the kernel counterpart,
but that's beyond my capabilites.

Hope that helps!
Thank you again :)

TatriX, 2023-04-29:
Hi! Thank you for checking this out!

Please check attached files.
I've just recorded two sessions on two machines with the same ble device.
Both use the same distro.

`working-*` files are recorded on a machine with older kernel.
`broken-*` are recorded on a machine with newer kernel.

I've started getting this issue after I've upgraded from 5.19. If I downgrade kernel on either machine everything works again.

To summarize what I observe:

- on older kernel with uuids filter enabled, discovery keeps going and when I send data from my ble device I can see it in bluetoothctl as well as I can get the data via bluetooth socket.

- on newer kernels data stops arriving after about 10 seconds. Discovery seems to be still going because if I try to start
  it again via "scan on" I'm getting "Failed to start discovery: org.bluez.Error.InProgress"

I've also tried using
```
[bluetooth]# pattern A4:34:F1:79:8A:F6
```
instead of
```
[bluetooth]# uuids 0x0612
```
but the symptoms are the same - data stops coming.

I've tried stepping through bluetoothd code, but coudn't find anything that could help me. I've also tried looking at the kernel counterpart, but that's beyond my capabilities.

Hope that helps!
Thank you again :)

*upd*: rewrote the comment directly on github
[working-btmon.log](https://github.com/bluez/bluez/files/11358930/working-btmon.log)
[working-bluetoothctl.log](https://github.com/bluez/bluez/files/11358931/working-bluetoothctl.log)
[broken-bluetoothctl.log](https://github.com/bluez/bluez/files/11358932/broken-bluetoothctl.log)
[broken-btmon.log](https://github.com/bluez/bluez/files/11358933/broken-btmon.log)

TatriX, 2023-05-05:
Perhaps I should write an email to linux-bluetooth mailing list instead, since this problem seems to be about kernel, not bluez?

Vudentz, 2023-05-05:
@TatriX you could try to bissect to check when this behavior started, that said it would be great to confirm if that still reproducible with a more recent kernel.

attilio-oliva, 2023-06-15:
I noticed this behavior sometimes as well and began quite some time ago. Currently, I am using linux 6.3.7 and this still occurs.

TatriX, 2023-08-07:
No changes in bluez 5.68 and kernel 6.4.7
At the same time here's another seemingly related issue:

On the old kernel 5.19.12, bluez 5.66, one of my appers when I do `bluetoothctl > scan on` with the name.
Looking at the `btmon` logs, it seems that device name is fetched from the `LE Extended Advertising Report `
```
> HCI Event: LE Meta Event (0x3e) plen 33                  #118 [hci0] 5.607028
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0013
            Props: 0x0013
              Connectable
              Scannable
              Use legacy advertising PDUs
            Data status:  [0;32mComplete [0m
          Legacy PDU Type: ADV_IND (0x0013)
          Address type: Random (0x01)
          Address: 68:71:DD:73:97:D5 (Resolvable)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -54 dBm (0xca)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x07
        02 01 06 03 02 f0 ff                             .......         
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        16-bit Service UUIDs (partial): 1 entry
          Unknown (0xfff0)
> HCI Event: LE Meta Event (0x3e) plen 49                  #119 [hci0] 5.608029
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x001b
            Props: 0x001b
              Connectable
              Scannable
              Scan response
              Use legacy advertising PDUs
            Data status:  [0;32mComplete [0m
          Legacy PDU Type: SCAN_RSP to an ADV_SCAN_IND (0x001b)
          Address type: Random (0x01)
          Address: 68:71:DD:73:97:D5 (Resolvable)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -54 dBm (0xca)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x17
        0d 09 50 6c 61 79 66 69 6e 69 74 79 2d 32 02 0a  ..Playfinity-2..
        00 05 12 50 00 68 00                             ...P.h.         
        Name (complete): Playfinity-2
        TX power: 0 dBm
        Peripheral Conn. Interval: 0x0050 - 0x0068
```

On the newer kernel name is not found:
```
> HCI Event: LE Meta Event (0x3e) plen 33                  #130 [hci0] 9.180207
      LE Extended Advertising Report (0x0d)
        Num reports: 1
        Entry 0
          Event type: 0x0013
            Props: 0x0013
              Connectable
              Scannable
              Use legacy advertising PDUs
            Data status: [0;32mComplete[0m
          Legacy PDU Type: ADV_IND (0x0013)
          Address type: Random (0x01)
          Address: 4D:18:19:A8:63:B5 (Resolvable)
          Primary PHY: LE 1M
          Secondary PHY: No packets
          SID: no ADI field (0xff)
          TX power: 127 dBm
          RSSI: -53 dBm (0xcb)
          Periodic advertising interval: 0.00 msec (0x0000)
          Direct address type: Public (0x00)
          Direct address: 00:00:00:00:00:00 (OUI 00-00-00)
          Data length: 0x07
        02 01 06 03 02 f0 ff                             .......         
        Flags: 0x06
          LE General Discoverable Mode
          BR/EDR Not Supported
        16-bit Service UUIDs (partial): 1 entry
          Unknown (0xfff0)
```

Vishal8523, 2023-08-17:
@TatriX : Facing the same issue with Kernel 5.10.184 and BlueZ 5.66
Any suggestions for resolution ?

Vishal8523, 2023-08-21:
@TatriX : Is the communication stable with Kernel 5.9.X ?

TatriX, 2023-08-21:
> Is the communication stable with Kernel 5.9.X ?

I'm not sure what you specifically mean by "communication", but I don't have any issues with kernel 5.9.12 for my use cases: listening for broadcast data, detecting devices by their name, connecting and reading notification data.

Vudentz, 2023-09-06:
Try with the following change:

https://github.com/bluez/bluetooth-next/commit/52bf4fd43f759ac4698f041fff2cb27691698265

TatriX, 2023-09-07:
Looks like it works now!
Wonderful! Thanks you so much for the fix :)

closed 2023-09-07 by TatriX