bluez/bluez #511: No way to define which transport to use (BREDR or LE) on per-device basis.

zalexua, 2023-05-01:
H i!

I've a device with supports both transports and uses the same ADDR.
It's BT-04A. If you want, you can find it here https://www.aliexpress.com/item/32674284442.html

The problem is that I cannot control when I want to use LE namely.

There are 2 libraries:  https://github.com/Jakeler/ble-serial which uses https://github.com/hbldh/bleak/ which use BlueZ and makes connection by Dbus.

There is related discussion in https://github.com/hbldh/bleak/issues/701

I could find this in documentation 
https://github.com/bluez/bluez/blob/9be85f867856195e16c9b94b605f65f6389eda33/doc/device-api.txt#L24-L36
As I see it comes from a function select_conn_bearer() https://github.com/bluez/bluez/blob/master/src/device.c#L2464-L2508

I tried hard but could not get that to work to connect my BT-04A with LE transport.

I tried to change globally /etc/bluetooth/main.conf and set "ControllerMode = le" and that worked, but that is not acceptable as then I cannot work with BR/EDR devices.

So, **per-device** method is needed to use LE transport.

Vudentz, 2023-05-01:
@zalexua I assume you are saying the some devices connect over BR/EDR when you want to connect over LE, which is probably due to them having BR/EDR supported flag set, so you want some means to force connecting over LE instead. In the past there was some debate on how we could support such functionality, there was even a proposal for that some years ago:

https://www.spinics.net/lists/linux-bluetooth/msg66817.html

zalexua, 2023-05-02:
@Vudentz thanks for letting me know about the proposal. It's dated back in 2016, so quite long time passed already and I do not see it was considered seriously.
Yes, on LE discovery my device says than both flags are set and BR/EDR is supported. And the BR/EDR is really working too, on the same ADDRess!

I have multiple other BT devices which support BR/EDR and LE, but those are using different ADDResses and NAMEs, so they look like independent and do not cause any issues for BlueZ.

zalexua, 2025-06-04:
This is wrong to mark this as a stale.
It's still valid and should remain opened.