bluez/bluez #513: Build warnings on 32-bit/musl due to %lu and %zd formats

macmpi, 2023-05-04:
[As per this log](https://pastebin.com/4FvPUeRa), compiling on 32-bit armhf/musl (time64), we can see quite a bit of:
`tools/parser/parser.h:: warning: format '%lu' expects argument of type 'long unsigned int', but...`
and few:
`client/player.c:: warning: format '%zd' expects argument of type 'signed size_t', but...`

Any hint on fixing these?

nekopsykose, 2023-05-10:
if it's not an issue to potentially be too wide when time_t/off_t are actually 32-wide (they should never be anyway; you should set `-D_FILE_OFFSET_BITS=64 -D_TIME_BITS=64` to make them 64 on 32-bit platforms too.), then you can just make both instances `%ll` instead of `%l`/`%z`

bgix, 2023-05-11:
Try searching the BLUEZ code base for "grep -r PRI.64 .", specifically for PRIx64 and PRIu64 to in place of %z and other less portable formaters.

For instance, if you know the data type is 64 bit unsigned, use PRIu64 for the decimal equivilant and PRIx64 for the hexidecimal equivilent.

closed 2025-08-10 by github-actions[bot]