bluez/bluez #514: Pairing AirPods Pro requires ControllerMode to be explicitly set to BR/EDR

GabeChiarelli, 2023-05-05:
When attempting to pair AirPods Pro, the default `dual` setting for `ControllerMode` in main.conf does not seem to work. When the service is restarted with the value set to `bredr`, pairing is successful. After pairing, the `ControllerMode` value can be restored to the default and the service reset, and the AirPods will connect normally.

There doesn't seem to be a parameter to specify which mode to pair with in dual mode, which would potentially solve this problem (assuming the device is misleadingly specifying that it supports both modes).

atimofeev, 2025-03-27:
This is still actual and also includes similar devices, such as Marshall Motif 2.
Only operational on `bredr` mode. Dual may work for a fisrt-time pairing, but upon reconnect to paired device it will get into a connect/disconnect loop.

Here's a pretty popular gh gist, describing the procedure of switching the controller mode in order to make AirPods operational: https://gist.github.com/aidos-dev/b49078c1d8c6bb1621e4ac199d18213b

atimofeev, 2025-09-18:
Not stale

atimofeev, 2025-10-23:
Still not stale

atimofeev, 2025-12-27:
not

stale

dirkf, 2026-01-22:
https://github.com/bluez/bluez/blob/c24f0b487ca3d4e0ca315114a889dad4d2c3bb26/src/main.c#L428 shows `dual` being the default but that should also come with a log message `Unknown controller mode`, which I don't see. Maybe the build or Debian packaging, or the _blueman-manager_ client (v2.3.5), changes the default.