bluez/bluez #515: BAP playback to TWS headset is desynchronized

pv, 2023-05-08:
I'm seeing BAP playback to a TWS headset (Samsung Galaxy Buds2 Pro + Intel AX210) left and right earbuds to be desynchronized immediately at playback start. Looking at audio recorded by external mic, it seems the playback on left and right earbuds is off by 10ms.

This occurs sometimes, quite often when the devices connect after taking them out of their case. Disconnect + reconnect fixes the issue, but sometimes it needs to be done more than once to get synchronized playback.

Looking at the HCI traces, I did not see the sound server (Pipewire) doing something strange, and BlueZ/kernel problem also seems unlikely, but uploading here mainly to have some place to refer to for the files.

Headset: Samsung Galaxy Buds2 Pro
Controller: Intel AX210
Bluez: 6bafe1c66515e57
Kernel: 6.3.1 + https://git.kernel.org/pub/scm/linux/kernel/git/bluetooth/bluetooth-next.git/commit/?id=d4b20f0b8491bfb3238ebf7df4b13243189620d2
Pipewire: https://gitlab.freedesktop.org/pvir/pipewire/-/commit/068a132203a9896ff69c0f561406355c86fbdcd5 (master/0.3.70 is also same, this is just to make absolutely sure exactly same data is written to both CIS)

This has occurred also on all previous versions of the software/firmware I've tested.

```
Bluetooth: hci1: Minimum firmware build 1 week 10 2014
Bluetooth: hci1: Bootloader timestamp 2019.40 buildtype 1 build 38
Bluetooth: hci1: Found device firmware: intel/ibt-0041-0041.sfi
Bluetooth: hci1: Boot Address: 0x100800
Bluetooth: hci1: Firmware Version: 98-13.23
Bluetooth: hci1: Waiting for firmware download to complete
Bluetooth: hci1: Firmware loaded in 1599630 usecs
Bluetooth: hci1: Waiting for device to boot
Bluetooth: hci1: Device booted in 25438 usecs
Bluetooth: hci1: Malformed MSFT vendor event: 0x02
Bluetooth: hci1: Found Intel DDC parameters: intel/ibt-0041-0041.ddc
Bluetooth: hci1: Applying Intel DDC parameters completed
Bluetooth: hci1: Firmware timestamp 2023.13 buildtype 1 build 62562
```

Input signal: [ax210-samsung-gbuds2pro-tws-desync-3-input.flac.log](https://github.com/bluez/bluez/files/11423692/ax210-samsung-gbuds2pro-tws-desync-3-input.flac.log)

HCI trace 1 (desynchronized): [ax210-samsung-gbuds2pro-tws-desync-3.log](https://github.com/bluez/bluez/files/11423696/ax210-samsung-gbuds2pro-tws-desync-3.log)
External mic (close to both earbuds) recording 1: [ax210-samsung-gbuds2pro-tws-desync-3.flac.log](https://github.com/bluez/bluez/files/11423698/ax210-samsung-gbuds2pro-tws-desync-3.flac.log)

HCI trace 2 (desynchronized): [ax210-samsung-gbuds2pro-tws-desync-4.log](https://github.com/bluez/bluez/files/11423701/ax210-samsung-gbuds2pro-tws-desync-4.log)
External mic (close to both earbuds) recording 2: [ax210-samsung-gbuds2pro-tws-desync-4.flac.log](https://github.com/bluez/bluez/files/11423703/ax210-samsung-gbuds2pro-tws-desync-4.flac.log)

You can hear in the recordings that the blip sounds are slightly separated, and it is obvious in stereo. Inspecting the waveforms one can see there's a 10ms offset.

For comparsion (I don't see any suspicious difference however in the traces however): 

HCI trace 3 (synchronization ok): [ax210-samsung-gbuds2pro-tws-ok.log](https://github.com/bluez/bluez/files/11423742/ax210-samsung-gbuds2pro-tws-ok.log)

Vudentz, 2023-06-21:
@pv Im reading the following part of the spec and it looks like we need to sync the sequence number somehow:

```
BLUETOOTH CORE SPECIFICATION Version 5.3 | Vol 6, Part G
page 3041

SDUs sent to the upper layer by the ISOAL shall be given a sequence number
which is initialized to 0 when the CIS or BIS is created. SDUs received by the
ISOAL from the upper layer shall be given a sequence number which is
initialized to 0 when the CIS or BIS is created. The upper layer may
synchronize its sequence number with the sequence number in the ISOAL
once the Datapath is configured and the link is established.
The sequence number shall be incremented for each SDU_Interval, whether or
not an SDU was received from or sent to the upper layer.
```

That said normally we would use HCI_LE_Read_ISO_TX_Sync to sync up the sequence number but I don't think that is implemented in AX210 firmware, note that we need to increment the sequence number even when there is no data.

pv, 2023-06-21:
The user application probably could set some running sequence number in a CMSG, and we could then use those to align the packets for different CIS in same CIG on kernel side, and also assign new sequence numbers for the controller.

I'm not sure if AX210 does something with the numbers. IIRC I tried aligning the sequence numbers in the two CIS (passing them from userspace in cmsg) earlier, but this had no effect. Maybe need to try this again, it was some time ago.

Vudentz, 2023-06-21:
@pv Id leave this for the kernel to resolve since it might revolve into HCI traffic, and in case we move the socket around the stack so it is probably not a good idea to use CMSG for something so fundamental as sequence number, anyway if you tried that and it didn't make any difference, then maybe it is not related to that.

pv, 2023-06-21:
Yeah, for the CMSG I mean that it would be useful to have some mechanism for indicating to the kernel which packets correspond to the same time in different CIS, to make it easier to synchronize the streams on socket level. Running sequence numbering for the packets (unrelated to whatever kernel decides to send to adapter) same in each CIS could be one way.

Vudentz, 2023-06-22:
@pv can you give it a try with following patch to see if it makes a difference:

https://patchwork.kernel.org/project/bluetooth/patch/20230621205230.1006939-1-luiz.dentz@gmail.com/

pv, 2023-06-23:
Doesn't seem to change it. BlueZ and Pipewire I think do not make use of the getsockopt ISO QoS latency/interval values obtained after connection, but use the ASE QoS values.

I guess a suspicion was that user application should stuff N>0 extra packets to the sockets immediately when playback starts, so that kernel+adapter do not run out of data to send if there are timing glitches. I tried this earlier with different numbers of extra frames, but didn't seem to help with the desync.

Writing many initial packets also seems to make it a bit worse. The kernel seems to do this:
```
bluetoothd[5840]: < ISO Data TX: Handle 2560 flags 0x02 dlen 124                                        #252 [hci0] 53.405433
bluetoothd[5840]: < ISO Data TX: Handle 2560 flags 0x02 dlen 124                                        #253 [hci0] 53.405848
bluetoothd[5840]: < ISO Data TX: Handle 2560 flags 0x02 dlen 124                                        #254 [hci0] 53.406037
bluetoothd[5840]: < ISO Data TX: Handle 2560 flags 0x02 dlen 124                                        #255 [hci0] 53.406211
bluetoothd[5840]: < ISO Data TX: Handle 2561 flags 0x02 dlen 124                                        #256 [hci0] 53.406409
bluetoothd[5840]: < ISO Data TX: Handle 2561 flags 0x02 dlen 124                                        #257 [hci0] 53.406624
bluetoothd[5840]: < ISO Data TX: Handle 2561 flags 0x02 dlen 124                                        #258 [hci0] 53.406819
bluetoothd[5840]: < ISO Data TX: Handle 2561 flags 0x02 dlen 124                                        #259 [hci0] 53.407072
```
even though the socket writes are in interleaved CIS1 CIS2 CIS1 CIS2 ... order. Probably there's no ordering guarantee. The ordering can also swap spontaneously as kernel sends ... CIS1 CIS2 CIS2 CIS1 ... or CIS1 CIS1 CIS2 CIS2. However, I have also HCI traces with fully CIS1 CIS2 CIS1 CIS2 ... interleaved traffic that is desynchronized, and traces with interleaving order swaps that is synchronized. So it's not clear this has something to do with it.

Vudentz, 2023-06-23:
> Doesn't seem to change it. BlueZ (except transport.send) and Pipewire I think do not make use of the getsockopt ISO QoS latency/interval values obtained after connection, but use the ASE QoS values.

We probably need to fix that though so the CIS Established values are used since that afaik is what the controller expects.

> I guess a suspicion was that user application should stuff N>0 extra packets to the sockets immediately when playback starts, so that kernel+adapter do not run out of data to send if there are [...]

Well these seems to be sent at the same iteration though, Im thinking on doing some changes to the how we schedule since the current scheduler don't take into account the timestamps of the packets.

> even though the socket writes are in interleaved CIS1 CIS2 CIS1 CIS2 ... order. Probably there's no ordering guarantee. The ordering can also swap spontaneously as kernel sends ... CIS1 CIS2 CIS2 CIS1 ... or CIS1 CIS1 CIS2 CIS2. However, I have also HCI traces with fully CIS1 CIS2 CIS1 CIS2 ... interleaved traffic that but desynchronized playback, and traces with interleaving order swaps giving synchronized playback. So it's not clear this has something to do with it.

This is because of the current queuing discipline, it is basically a priority based one which in this case probably means there are 2 channels with the same priority so they will get each half of the buffers/credits at each scheduler iteration, which is not that great for timing sensitive packets.

Vudentz, 2023-07-28:
@pv Im using these files to with new btmon -a/--analyze code:

```
  Found LE-ISO connection with handle 2560
        Address: 00:00:00:00:00:00 (OUI 00-00-00)
        Connection setup missing
        1878 RX packets
        2102 TX packets
        2100 TX completed packets
        9-53 msec (~13 msec) TX Latency
        124-124 octets (~124 octets) TX packet size

                                                                               
  10000 +-+----------------------------------------------------------------+   
         +|                                                                |   
         +|                                                Packets +-----+ |   
         +|         +                                                      |   
   1000 +-|         |                                                      |   
         +|         |                                                      |   
         +|         |                                                      |   
         +|         |                                                      |   
    100 +-|         |                                                      |   
         +|         |                         +                            |   
         +|         |                         |                            |   
         +|         |            +            |                            |   
          |    +   +|            |            |            +               |   
     10 +-|    |   ||            |            |            |            +  |   
         +|    |   ||            |            |            |            |  |   
         +|    |   ||           +|       +    |          ++|      +     |  |   
          |    |   ||           ||       |    |          |||      |     |  |   
      1 +-|    |   ||           ||       |    |          |||      |     |  |   
         ++----------------------------------------------------------------+   
          +      +     +      +     +      +     +      +     +      +     +   
          5      10    15     20    25     30    35     40    45     50    55  
                                    Latency (ms)
```

```
  Found LE-ISO connection with handle 2561
        Address: 00:00:00:00:00:00 (OUI 00-00-00)
        Connection setup missing
        1896 RX packets
        2102 TX packets
        2100 TX completed packets
        3-33 msec (~13 msec) TX Latency
        124-124 octets (~124 octets) TX packet size

                                                                               
  10000 +-+----------------------------------------------------------------+   
         +|                                                                |   
         +|                                                Packets +-----+ |   
         +|                       +                                        |   
   1000 +-|                       |                                        |   
         +|                       |                                        |   
         +|                       |                                        |   
         +|                       |                                        |   
    100 +-|                       |                                        |   
         +|     +                 |                                        |   
         +|     |                 |                  +                     |   
         +|     |                 |                  |                     |   
          |     |          +    + |                  |                     |   
     10 +-|     |          |    | |                  |                     |   
         +|     |          |    | |                + |                     |   
         +|     |          |    | |                | |                 +   |   
          |     |          |    | |                | |                 |   |   
      1 +-|     |          |    | |                | |                 |   |   
         ++----------------------------------------------------------------+   
          +        +         +        +        +        +         +        +   
          0        5         10       15       20       25        30       35  
                                    Latency (ms)
```

No wonder they get out of sync, although the vast majority are around 12-13 ms there are quite a few that are spread.

Vudentz, 2023-07-28:
I suspect this is when start to derail:

```
> HCI Event: Number of Completed Packets (0x13) plen 9                                                                                                                                                                                                           #1608 [hci1] 09:56:06.744196
        Num handles: 2
        Handle: 2560 Address: 00:00:00:00:00:00 (OUI 00-00-00)
        Count: 3
        Latency: 53 msec (9-53 msec ~43 msec)
        Latency: 43 msec (9-53 msec ~43 msec)
        Latency: 33 msec (9-53 msec ~38 msec)
        Handle: 2561 Address: 00:00:00:00:00:00 (OUI 00-00-00)
        Count: 3
        Latency: 23 msec (3-33 msec ~13 msec)
        Latency: 13 msec (3-33 msec ~13 msec)
        Latency: 3 msec (3-33 msec ~8 msec)
```

I will probably need to add the frame number to make it clearer which are these responses.

pv, 2023-07-29:
From what I recall, and can still reproduce (on AX210 Firmware timestamp 2023.24 buildtype 1 build 67068, Found device firmware: intel/ibt-0041-0041.sfi, Firmware Version: 252-24.23) is that the playback sync does not "slip" during playback and there are no sound glitches. Instead, the left/right streams are desynchronized immediately at playback start.

pv, 2023-07-29:
IIRC, the playback was desynchronized in these cases immediately at the beginning, and I think it does not "slip" later on and there were no sound glitches. This is also how it is when reproduced now. So if there is some issue with the packet scheduling most likely is within a few seconds from stream start.

If user pauses writing to the socket for a while, and then resumes sending packets, this can change the synchronization (either make it synchronized or break the synchronization).

pv, 2024-04-12:
With the TX latency patches, this is now mitigated on application side.