bluez/bluez #516: BAP playback breaks if transports are released

pv, 2023-05-08:
With Samsung Galaxy Buds2 Pro + BlueZ 6bafe1c66515e57636 (and all previous versions I've tested):

The BAP transports can be acquired the first time. However, if they are released and go idle, the next acquire always fails.

In the BlueZ log, I see a transition (for the remote source ASE) from streaming -> disabling. The remote sink ASE seems to go streaming -> qos fine. However, in the logs it seems the source ASEs seem to get stuck in the disabling state, and this maybe prevents reacquiring the released transports.

BlueZ log: [samsung2-gbuds2-ase-stuck.bluez.log](https://github.com/bluez/bluez/files/11423952/samsung2-gbuds2-ase-stuck.bluez.log)
HCI trace: [samsung2-gbuds2-ase-stuck.log](https://github.com/bluez/bluez/files/11423953/samsung2-gbuds2-ase-stuck.log)

Vudentz, 2023-05-10:
@pv looks like btmon cannot resolve the attributes definitions so this makes it a little hard to find when we are sending ASE Disable to confirm we are dropping the CIS. Btw, what firmware version of the gbuds are you using? I recall this working properly when doing bluetoothctl> transport.release.

pv, 2023-05-10:
The gbuds2 firmware is R510XXUOAWA5, the first one with the LE audio support, afaik there's no newer one.

In the bluetoothd logs above we have 0x05 (ASCS_DISABLE) queued
```
May 08 21:10:38 bluetoothd[37059]: src/shared/bap.c:bap_queue_req() req 0x6060001a2d60 (op 0x05) queue 0x603000056770
May 08 21:10:38 bluetoothd[37059]: src/shared/bap.c:bap_queue_req() req 0x6060001a2dc0 (op 0x05) queue 0x6030002ee730
```
but I don't see op 0x06 (ASCS_STOP) being queued. This probably has to do with the code path the above patch touches, i.e. linked transport release where it can depend on whether source or sink gets released first.

With the above patch 0x06 gets queued, but only one earbud seems to transition to qos, probably it's not quite right yet. 
[samsung2-gbuds2-ase-stuck-w-patch.bluez.log](https://github.com/bluez/bluez/files/11446190/samsung2-gbuds2-ase-stuck-w-patch.bluez.log)

pv, 2023-05-12:
The behavior with that patch seems to be similar as with the other one: [samsung2-gbuds2-ase-stuck-w-patch.bluez.log](https://github.com/bluez/bluez/files/11467837/samsung2-gbuds2-ase-stuck-w-patch.bluez.log) the source ASE of one of the devices doesn't transition back to qos.

Vudentz, 2023-05-13:
> The behavior with that patch seems to be similar as with the other one: [samsung2-gbuds2-ase-stuck-w-patch.bluez.log](https://github.com/bluez/bluez/files/11467837/samsung2-gbuds2-ase-stuck-w-patch.bluez.log) the source ASE of one of the devices doesn't transition back to qos despite the stop commands, but the other does. Not clear to me why.

Yeah, I tried with the gbus2 so even when we send ASE Receiver Stop Ready (0x06) it still don't respond, seems to be a problem with their firmware or something, but as far as not sending the command to stop that should have been fixed.