bluez/bluez #517: LE connections failing to Samsung Galaxy Buds2 Pro, but connects on power off/on

pv, 2023-05-08:
LE connections to the Galaxy Buds2 Pro TWS headset seem to fail in a predictable pattern:

- power device on (take out from case) -> connects immediately
- disconnect using bluetoothctl, then try to reconnect -> fails to connect with le-connection-abort-by-local (timeout)
- controller power off, power on -> connects immediately

HCI trace: [ax210-samsung-gbuds2pro-connect-problem.log](https://github.com/bluez/bluez/files/11424262/ax210-samsung-gbuds2pro-connect-problem.log) -- try to connect using bluetoothctl, but fails. then power off/on controller and it connects immediately

Vudentz, 2023-05-08:
@pv Ive seem this bug before, looks like there is something preventing the controller to connect when we perform bluetoothct> disconnect manually, it does connect properly via auto-connect method so I suspect there is something at link-layer that is not quite right.

Vudentz, 2023-05-08:
This seems where the problem appears:

```
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 58                                                       #269 [hci0] 27.250855
        Filter policy: Accept list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Random (0x01)
        Peer address: 63:91:92:1F:E5:37 (Resolvable)
        Initiating PHYs: 0x07
        Entry 0: LE 1M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 7.50 msec (0x0006)
          Max connection interval: 12.50 msec (0x000a)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
        Entry 1: LE 2M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 7.50 msec (0x0006)
          Max connection interval: 12.50 msec (0x000a)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
        Entry 2: LE Coded
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 7.50 msec (0x0006)
          Max connection interval: 12.50 msec (0x000a)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                                                #270 [hci0] 27.252752
      LE Extended Create Connection (0x08|0x0043) ncmd 2
        Status: Success (0x00)
< HCI Command: LE Create Connection Cancel (0x08|0x000e) plen 0                                                          #271 [hci0] 31.290826
> HCI Event: Command Complete (0x0e) plen 4                                                                              #272 [hci0] 31.304821
      LE Create Connection Cancel (0x08|0x000e) ncmd 2
        Status: Success (0x00)
@ MGMT Event: Connect Failed (0x000d) plen 8                                                                         {0x0001} [hci0] 31.304876
        LE Address: 63:91:92:1F:E5:37 (Resolvable)
        Status: Timeout (0x08)

```

vs

```
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 58                                                       #505 [hci0] 60.438795
        Filter policy: Accept list is not used (0x00)
< HCI Command: LE Extended Create Connection (0x08|0x0043) plen 58                                                       #505 [hci0] 60.438795
        Filter policy: Accept list is not used (0x00)
        Own address type: Public (0x00)
        Peer address type: Random (0x01)
        Peer address: 63:91:92:1F:E5:37 (Resolvable)
        Initiating PHYs: 0x07
        Entry 0: LE 1M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 7.50 msec (0x0006)
          Max connection interval: 12.50 msec (0x000a)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
        Entry 1: LE 2M
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 7.50 msec (0x0006)
          Max connection interval: 12.50 msec (0x000a)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
        Entry 2: LE Coded
          Scan interval: 60.000 msec (0x0060)
          Scan window: 60.000 msec (0x0060)
          Min connection interval: 7.50 msec (0x0006)
          Max connection interval: 12.50 msec (0x000a)
          Connection latency: 0 (0x0000)
          Supervision timeout: 420 msec (0x002a)
          Min connection length: 0.000 msec (0x0000)
          Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                                                #506 [hci0] 60.440736
      LE Extended Create Connection (0x08|0x0043) ncmd 2
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 31                                                                                #507 [hci0] 60.502876
      LE Enhanced Connection Complete (0x0a)
        Status: Success (0x00)
        Handle: 3585
        Role: Central (0x00)
        Peer address type: Random (0x01)
        Peer address: 63:91:92:1F:E5:37 (Resolvable)
        Local resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Peer resolvable private address: 00:00:00:00:00:00 (Non-Resolvable)
        Connection interval: 7.50 msec (0x0006)
        Connection latency: 0 (0x0000)
        Supervision timeout: 420 msec (0x002a)
        Central clock accuracy: 0x00
```

The addresses are actually the same so it really beats me why it can't connect sometimes.

closed 2025-08-10 by github-actions[bot]