bluez/bluez #523: Galaxy Buds2 Pro - unpairable on modern (LE) adapters(with Bluez)

AndKe, 2023-06-02:
after much troubleshooting I finally know why *some* of my laptops/computers can pair with Buds Pro 2
OS: Ubuntu 23.10  
Bluez 5.66-0ubuntu1
kernel 6.2.0-20-generic

method:
I open the clamshell case, long press both buds until they blink RGB.

Working device:
BT Device:  (BT2.0 + EDR)
Bus 001 Device 017: ID 0a5c:2121 Broadcom Corp. BCM2210 Bluetooth

I see  - as expected, it connects, and works. 

[image]

Failure device:
BT Device:  BT4LE
Bus 001 Device 018: ID 0a5c:21e8 Broadcom Corp. BCM20702A0 Bluetooth 4.0
or: Dell:  413c:8197 Dell BCM20702A0 

Then I see: SmartTags come and go, one or more "Buds Pro"  - there may even once in several attempts be a "Galaxy Buds2 Pro" 
regardless of what I click on - only occasionally can it result in a connection, and even then - no audio playback device is spawned.
It may be impossible to remove such a phantom "device" using GUI  if partially connected.
[image]

It all seems to be related to newer BT devices. My

AndKe, 2023-06-15:
additional info:
This is how I managed to pair them from terminal using bluetoothctl
```
[bluetooth]# scan bredr
Discovery started
[CHG] Controller 24:FD:52:31:D4:45 Discovering: yes
[NEW] Device 44:CE:20:A6:8E:A6 44-CE-20-A6-8E-A6
[NEW] Device 6D:E2:F6:9D:92:81 6D-E2-F6-9D-92-81
[NEW] Device 64:5D:F4:88:49:89 Galaxy Buds2 Pro
[bluetooth]# trust 64:5D:F4:88:49:89
[CHG] Device 64:5D:F4:88:49:89 Trusted: yes
Changing 64:5D:F4:88:49:89 trust succeeded
[bluetooth]# pair 64:5D:F4:88:49:89
Attempting to pair with 64:5D:F4:88:49:89
[CHG] Device 64:5D:F4:88:49:89 Connected: yes
[CHG] Device 64:5D:F4:88:49:89 Bonded: yes
[CHG] Device 64:5D:F4:88:49:89 Modalias: bluetooth:v0075pA013d0001
[CHG] Device 64:5D:F4:88:49:89 UUIDs: 00001101-0000-1000-8000-00805f9b34fb
[CHG] Device 64:5D:F4:88:49:89 UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Device 64:5D:F4:88:49:89 UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Device 64:5D:F4:88:49:89 UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Device 64:5D:F4:88:49:89 UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Device 64:5D:F4:88:49:89 UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 64:5D:F4:88:49:89 UUIDs: 2e73a4ad-332d-41fc-90e2-16bef06523f2
[CHG] Device 64:5D:F4:88:49:89 UUIDs: a23d00bc-217c-123b-9c00-fc44577136ee
[CHG] Device 64:5D:F4:88:49:89 UUIDs: b4a9d6a0-b2e3-4e40-976d-a69f167ea895
[CHG] Device 64:5D:F4:88:49:89 UUIDs: e7ab2241-ca64-4a69-ac02-05f5c6fe2d62
[CHG] Device 64:5D:F4:88:49:89 UUIDs: f8620674-a1ed-41ab-a8b9-de9ad655729d
[CHG] Device 64:5D:F4:88:49:89 ServicesResolved: yes
[CHG] Device 64:5D:F4:88:49:89 Paired: yes
Pairing successful
[CHG] Device 64:5D:F4:88:49:89 ServicesResolved: no
[CHG] Device 64:5D:F4:88:49:89 Connected: no
[DEL] Device 44:CE:20:A6:8E:A6 44-CE-20-A6-8E-A6
[CHG] Device 64:5D:F4:88:49:89 RSSI: -65
[NEW] Device 72:03:EE:85:5B:79 72-03-EE-85-5B-79
[NEW] Device 44:CE:20:A6:8E:A6 44-CE-20-A6-8E-A6

```

Vudentz, 2023-06-22:
@AndKe do you have experimental flag enabled? I suspect this has to do with attempting to do LE Audio and failing since 5.66 is probably not as stable as master in this respect and you also need the a more recent release of pipewire as well in order to get LE Audio working properly with true wireless buds.

AndKe, 2023-06-22:
@Vudentz  I do not know how/where to check for "experimental flag" I guess it's a bluez feature?
I did not change it, so AFAIK - it is at it's default setting.
Please note that when paired with the proper name, (via CLI) - audio works flawlessfly.  As I see it, the issue is somehow limited to the pairing when using GUI / and discovery results seen in GUI.

Vudentz, 2023-06-22:
@AndKe maybe it should be reported to Gnome then, it is probably the same device showing as LE and BR/EDR because until they are paired we have no idea they are the same.

AndKe, 2023-06-22:
@Vudentz  Any suggestions on where to report?  this does not seem like the place: https://gitlab.gnome.org/GNOME/gnome-bluetooth
I do not know if this is a GUI /settings panel issue, or some backend system.

akhilerm, 2023-12-20:
The solution mentioned by @AndKe above using `bluetoothctl` worked for me. I am using Galaxy Buds 2 Pro, Thinkpad L14, with ubuntu 23.10
```bash
akhil@akhil-ThinkPad-L14:~ $ uname -a
Linux akhil-ThinkPad-L14 6.5.0-14-generic #14-Ubuntu SMP PREEMPT_DYNAMIC Tue Nov 14 14:59:49 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
akhil@akhil-ThinkPad-L14:~ $ cat /etc/os-release 
PRETTY_NAME="Ubuntu 23.10"
NAME="Ubuntu"
VERSION_ID="23.10"
VERSION="23.10 (Mantic Minotaur)"
VERSION_CODENAME=mantic
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=mantic
LOGO=ubuntu-logo
akhil@akhil-ThinkPad-L14:~ $ hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: AC:50:DE:F4:B9:2E  ACL MTU: 1021:6  SCO MTU: 240:8
	UP RUNNING PSCAN ISCAN INQUIRY 
	RX bytes:1182384 acl:168 sco:11439 events:21830 errors:0
	TX bytes:9215654 acl:12850 sco:10914 commands:3260 errors:0
	Features: 0xbf 0x3e 0x8d 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'akhil-ThinkPad-L14'
	Class: 0x7c010c
	Service Classes: Rendering, Capturing, Object Transfer, Audio, Telephony
	Device Class: Computer, Laptop
	HCI Version: 5.2 (0xb)  Revision: 0x2714
	LMP Version: 5.2 (0xb)  Subversion: 0x2306
	Manufacturer: MediaTek, Inc. (70)
```

After connecting once via the `bluetoothctl` CLI, later connections could be done via the GUI itself. 

While scanning it actually detected two devices 
```
[NEW] Device 64:5D:F4:82:6F:F2 Galaxy Buds2 Pro
[NEW] Device 64:5D:F4:82:6A:4C Galaxy Buds2 Pro
[CHG] Device 64:5D:F4:82:6F:F2 Class: 0x00244404 (2376708)
[CHG] Device 64:5D:F4:82:6F:F2 Icon: audio-headset
```
and I connected to the one with device address `64:5D:F4:82:6F:F2` since it was shown as `audio-headset`

Vudentz, 2023-12-20:
> The solution mentioned by @AndKe above using `bluetoothctl` worked for me. I am using Galaxy Buds 2 Pro, Thinkpad L14, with ubuntu 23.10
> 
> ```shell
> akhil@akhil-ThinkPad-L14:~ $ uname -a
> Linux akhil-ThinkPad-L14 6.5.0-14-generic #14-Ubuntu SMP PREEMPT_DYNAMIC Tue Nov 14 14:59:49 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
> akhil@akhil-ThinkPad-L14:~ $ cat /etc/os-release 
> PRETTY_NAME="Ubuntu 23.10"
> NAME="Ubuntu"
> VERSION_ID="23.10"
> VERSION="23.10 (Mantic Minotaur)"
> VERSION_CODENAME=mantic
> ID=ubuntu
> ID_LIKE=debian
> HOME_URL="https://www.ubuntu.com/"
> SUPPORT_URL="https://help.ubuntu.com/"
> BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
> PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
> UBUNTU_CODENAME=mantic
> LOGO=ubuntu-logo
> akhil@akhil-ThinkPad-L14:~ $ hciconfig -a
> hci0:	Type: Primary  Bus: USB
> 	BD Address: AC:50:DE:F4:B9:2E  ACL MTU: 1021:6  SCO MTU: 240:8
> 	UP RUNNING PSCAN ISCAN INQUIRY 
> 	RX bytes:1182384 acl:168 sco:11439 events:21830 errors:0
> 	TX bytes:9215654 acl:12850 sco:10914 commands:3260 errors:0
> 	Features: 0xbf 0x3e 0x8d 0xfe 0xdb 0xff 0x7b 0x87
> 	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
> 	Link policy: RSWITCH SNIFF 
> 	Link mode: PERIPHERAL ACCEPT 
> 	Name: 'akhil-ThinkPad-L14'
> 	Class: 0x7c010c
> 	Service Classes: Rendering, Capturing, Object Transfer, Audio, Telephony
> 	Device Class: Computer, Laptop
> 	HCI Version: 5.2 (0xb)  Revision: 0x2714
> 	LMP Version: 5.2 (0xb)  Subversion: 0x2306
> 	Manufacturer: MediaTek, Inc. (70)
> ```
> 
> After connecting once via the `bluetoothctl` CLI, later connections could be done via the GUI itself.
> 
> While scanning it actually detected two devices
> 
> ```
> [NEW] Device 64:5D:F4:82:6F:F2 Galaxy Buds2 Pro
> [NEW] Device 64:5D:F4:82:6A:4C Galaxy Buds2 Pro
> [CHG] Device 64:5D:F4:82:6F:F2 Class: 0x00244404 (2376708)
> [CHG] Device 64:5D:F4:82:6F:F2 Icon: audio-headset
> ```
> 
> and I connected to the one with device address `64:5D:F4:82:6F:F2` since it was shown as `audio-headset`

This is normal with LE Audio devices that implements Coordinated Sets they will show up multiple times, that said one of those shall be printed in grey to indicate that they are not normally visible, perhaps that is missing on the gnome discovery and if you attempt to pair the invisible one that might cause problems.

AndKe, 2024-05-06:
@BloodDrag0n may having trouble because the Ubuntu community is not interested in helping people report issues like this.
Please see the video in my attempt to report it:

https://askubuntu.com/questions/1511447/where-to-report-this-ubuntu-24-04-issue-bluetooth-not-connecting-to-a2dp-device

to be more precise: https://youtube.com/shorts/8QAAoIWGdF8?feature=share

I hope this get fixed, because, as you see ... it does actually work.

MPW1412, 2025-03-14:
@AndKe, your askubuntu link is dead. Any updates on this?

AndKe, 2025-03-15:
well, I have upgraded the laptop, and also connected to PC with USB adapter.
I have some impression that the BT version of the audio source affects how it works.
When binding, I often see two BT devices  - like the second screenshot in the first post here, and pairing *one* of them using GUI may work just fine - but it seems that if I select the incorrect one first, then the process need to be restarted.
So the issue remains, and  it is possible to succeed without bluetoothctl . It works fine once paired. - regardless method. 
- so one can summarise this as "janky pairing process" that may give the user the impression that it won't work.

MPW1412, 2025-03-15:
Hm, thanks for your reply. Surprisingly, my older Buds Live 3708 show up as just one device. The issue only happens with the lasted Buds Pro 2. 

If I pair both and connect them both, it at least works.

ValeZAA, 2025-03-28:
>several attempts be a "Galaxy Buds2 Pro" and even "Galaxy Buds2 Pro LE"

This says that you cannot connect to LE audio subdevice without forcing LE only in /etc/bluetooth/main.conf 

I will just say that pairing on Windows 11 with my arrow lake works, but the sound does not play, on Linux so far I could not even pair LE audio

pkit, 2025-06-04:
@AndKe no, it's not working fine at all in Ubuntu 24.10
Buds 2 Pro show as two devices and there is no sound if I connect a "wrong" one.
More than that, there is no sound if I connect only one too.
There is an audio device only if I connect both in a particular order.
Yes, the behavior seems strange. I will investigate what actually happens and report back.

AndKe, 2025-06-04:
reopened.

closed 2025-06-06 by BluezTestBot

pkit, 2025-06-10:
The behavior is as following:
```
[Galaxy Buds2 Pro #338]# info A8:BA:69:85:E4:5E
Device A8:BA:69:85:E4:5E (public)
	Name: Galaxy Buds2 Pro #338
	Alias: Galaxy Buds2 Pro #338
	Appearance: 0x0941 (2369)
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Volume Control            (00001844-0000-1000-8000-00805f9b34fb)
	UUID: Coordinated Set Identif.. (00001846-0000-1000-8000-00805f9b34fb)
	UUID: Microphone Control        (0000184d-0000-1000-8000-00805f9b34fb)
	UUID: Audio Stream Control      (0000184e-0000-1000-8000-00805f9b34fb)
	UUID: Broadcast Audio Scan      (0000184f-0000-1000-8000-00805f9b34fb)
	UUID: Published Audio Capabil.. (00001850-0000-1000-8000-00805f9b34fb)
	UUID: Common Audio              (00001853-0000-1000-8000-00805f9b34fb)
	UUID: Telephony and Media Audio (00001855-0000-1000-8000-00805f9b34fb)
	UUID: Telephony and Media Aud.. (00002b51-0000-1000-8000-00805f9b34fb)
	UUID: Unknown                   (00008fe1-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (a7a473e9-19c6-491b-aea6-7ea92b8f043a)
	Modalias: bluetooth:v0075pA013d0100
	ManufacturerData.Key: 0x0075 (117)
	ManufacturerData.Value:
  02 09 01 00 00 00 15 03 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 03 01 46 01 5b  .............F.[
  48 65 61 64 70 68 6f 6e 65 5d 20                 Headphone]      
	ServiceData.00002b51-0000-1000-8000-00805f9b34fb:
  2a 00                                            *.              
	ServiceData.00001853-0000-1000-8000-00805f9b34fb:
  00                                               .               
	ServiceData.0000184e-0000-1000-8000-00805f9b34fb:
  00 6f 00 6b 00 00                                .o.k..          
	ServiceData.a7a473e9-19c6-491b-aea6-7ea92b8f043a:
  01 46 0d e4 01                                   .F...           
	Battery Percentage: 0x64 (100)
```
```
info A8:BA:69:85:E4:0D
Device A8:BA:69:85:E4:0D (public)
	Name: Galaxy Buds2 Pro #338
	Alias: Galaxy Buds2 Pro #338
	Class: 0x00244404 (2376708)
	Icon: audio-headset
	Paired: yes
	Bonded: yes
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: Advanced Audio Distribu.. (0000110d-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree                 (0000111e-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	UUID: Device Information        (0000180a-0000-1000-8000-00805f9b34fb)
	UUID: Battery Service           (0000180f-0000-1000-8000-00805f9b34fb)
	UUID: Volume Control            (00001844-0000-1000-8000-00805f9b34fb)
	UUID: Coordinated Set Identif.. (00001846-0000-1000-8000-00805f9b34fb)
	UUID: Microphone Control        (0000184d-0000-1000-8000-00805f9b34fb)
	UUID: Audio Stream Control      (0000184e-0000-1000-8000-00805f9b34fb)
	UUID: Broadcast Audio Scan      (0000184f-0000-1000-8000-00805f9b34fb)
	UUID: Published Audio Capabil.. (00001850-0000-1000-8000-00805f9b34fb)
	UUID: Common Audio              (00001853-0000-1000-8000-00805f9b34fb)
	UUID: Telephony and Media Audio (00001855-0000-1000-8000-00805f9b34fb)
	UUID: Telephony and Media Aud.. (00002b51-0000-1000-8000-00805f9b34fb)
	UUID: Unknown                   (00008fe1-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (2e73a4ad-332d-41fc-90e2-16bef06523f2)
	UUID: Vendor specific           (a23d00bc-217c-123b-9c00-fc44577136ee)
	UUID: Vendor specific           (a7a473e9-19c6-491b-aea6-7ea92b8f043a)
	UUID: Vendor specific           (b4a9d6a0-b2e3-4e40-976d-a69f167ea895)
	UUID: Vendor specific           (d908aab5-7a90-4cbe-8641-86a553db0146)
	UUID: Vendor specific           (e74beac1-ab40-423f-b746-fb9869a982dd)
	UUID: Vendor specific           (e7ab2241-ca64-4a69-ac02-05f5c6fe2d62)
	UUID: Vendor specific           (f8620674-a1ed-41ab-a8b9-de9ad655729d)
	Modalias: bluetooth:v0075pA013d0001
	ManufacturerData.Key: 0x0075 (117)
	ManufacturerData.Value:
  02 09 01 00 00 00 15 03 00 00 00 00 00 00 00 00  ................
  00 00 00 00 00 00 00 00 00 00 00 03 01 46 01 5b  .............F.[
  48 65 61 64 70 68 6f 6e 65 5d 20                 Headphone]      
	ServiceData.00002b51-0000-1000-8000-00805f9b34fb:
  2a 00                                            *.              
	ServiceData.00001853-0000-1000-8000-00805f9b34fb:
  01                                               .               
	ServiceData.0000184e-0000-1000-8000-00805f9b34fb:
  01 6f 00 6b 00 00                                .o.k..          
	ServiceData.a7a473e9-19c6-491b-aea6-7ea92b8f043a:
  01 46 0d e4 0c                                   .F...           
	Battery Percentage: 0x64 (100)
                       
```

First I need to connect `A8:BA:69:85:E4:5E` then `A8:BA:69:85:E4:0D`.
But only if it was connected to other device previously.
If it was connected to the same device only `A8:BA:69:85:E4:5E` is enough.

ValeZAA, 2025-06-11:
>Windows 11 with my arrow lake works, but the sound does not play, on Linux so far I could not even pair LE audio

I was finally able to pair LE audio BAP and music plays but only on one earbud. LE Audio uses LE mac address, not Bluetooth Legacy MAC address.

Anyway, I was testing on commit before https://github.com/bluez/bluez/commit/4a7ee30978da919cc820116be6ced323b35e4ebd now on master it does not work. Will test further.