bluez/bluez #524: Apple Magic Trackpad (A1339) turns off every few minutes.

ScherbakovAl, 2023-06-02:
Apple Magic Trackpad (A1339) turns off every few minutes. I am using Ubuntu 23.04.
libinput --version: 1.22.1.
The trackpad periodically turns off and then, after a dozen seconds, reconnects. I'll attach a report from hciconfig, bluetoothctl, syslog, sudo btmon:
https://pastebin.com/3A9EXusp

I love using the trackpad, but these reconnections make interacting with it impossible. This trackpad works great with android phone and doesn't have any reconnects. Connectivity is constantly lost with Ubuntu. This also happens with different usb bluetooth dongles.
This issue has been reported here:
https://bugs.launchpad.net/linux/+bug/1834085

I booted Ubuntu 23.04(kernel 6.2.0) from a flash drive - the same error:
https://pastebin.com/5KtUjCHQ

I booted from Fedor's(kernel 6.2.9) flash drive - error:
https://pastebin.com/4LJC7NJQ

When using Fedora, communication breaks also occur.

What other logs and reports do I need to provide in order to sort out the problem?
Where else can you write about it?
Thank you!

Vudentz, 2023-06-06:
@ScherbakovAl Looks like it is a link layer issue as the connection just times out when we attempt to exit sniff mode:

> HCI Event: Disconnect Complete (0x05) plen 4                                                                     #18605 [hci0] 358.578948
        Status: Success (0x00)
        Handle: 256
        Reason: Connection Timeout (0x08)

Perhaps the device doesn't like that we are exitting sniff mode on our own or something like that.

shootheart, 2023-07-31:
I found the same issue at Arch. And my bluetooth earphone worked incorrectly when trackpad disconnected.
libinput version 1.23.0
kernel 6.4.7

nstwfdev, 2023-10-06:
Can someone describe: is it fixable? 
I found around 10 big topics with same issues from 2014...

closed 2025-08-10 by github-actions[bot]