bluez/bluez #527: BLE advertisement does not work after enabling experimental option

gradimir-lj, 2023-06-05:
I need to set min and max advertise intervals using BlueZ dbus interface. In order to do this I enabled bluetoothd experimental option and when I set these values to 120ms for example and then turn on advertise with advertise on using bluetoothctl I get an error: Failed to register advertisement: org.bluez.Error.NotPermitted.
BlueZ version is 5.61, Linux kernel 5.15.5 and extended advertise option should be supported here.
Do you know what might be the cause to this error?

Vudentz, 2023-06-05:
@gradimir-lj please add the HCI trace (btmon), also afaik you do need the controller to support the so called extended advertising in order to set certain intervals.

Vudentz, 2023-06-05:
@gradimir-lj perhaps it is a bug in bluetoothd, try updating to see if there is any difference.

zalexua, 2023-08-14:
> 
I can confirm that including BlueZ 5.62, when "bluetoothd" with experimental option:
`-E, --experimental          Enable experimental features/interfaces`
 you will get the same error: 
`Failed to register advertisement: org.bluez.Error.NotPermitted`
when trying to start "advertise peripheral" command in bluetoothctl.

In versions 5.63-5.64 you will get another error:
`LEAdvertisingManager not found`
That will happen because ... LEAdvertisingManager is not created in Dbus tree IF you stated daemon with the experimental option.

Only in versions 5.65+ (including tested recent 5.68) the experimental really works and does not break related things.
I really was able to set custom advertisement interval using bluetoothctl tool.

Here is a nice solution to simplify testing. Install a tool named "expect", which should exist on any Linux distro.
Create a file named "auto-ble-adv-100ms.exp" with content:
```
#!/usr/bin/expect -f

set timeout -1

spawn bluetoothctl
sleep .1
expect "\[bluetooth\]"

send -- "menu advertise\r"
expect "export"

send -- "name CTL-TEST-100ms\r"
expect "\[bluetooth\]"

send -- "interval 100 100\r"
expect "\[bluetooth\]"

send -- "back\r"
expect "export"

send -- "advertise peripheral\r"
expect "Interval*"

interact
```
set the file attribute executable and just start it.
It will do all the magic to start advertising!
When you finished to look at you smartphone, just interrupt it by pressing Ctrl+C.

So, this issue has been fixed !

Now question is - how to do that using Dbus, instead bluetoothctl ?

Vudentz, 2023-08-14:
> I can confirm that including BlueZ 5.62, when "bluetoothd" with experimental option: `-E, --experimental Enable experimental features/interfaces` you will get the same error: `Failed to register [...]

You can create a set of startup command and pass to bluetoothctl with -i/--init-script, that way it will wait each command to complete before starting the next.

> set the file attribute executable and just start it. It will do all the magic to start advertising! When you finished to look at you smartphone, just interrupt it by pressing Ctrl+C.
> 
> So, this issue has been fixed !
> 
> Now question is - how to do that using Dbus, instead bluetoothctl ?

bluetoothctl uses the D-Bus APIs, so it shouldn't be that hard to figure out what interface.method it is using behind scenes.

gradimir-lj, 2023-08-15:
I forgot to update the issue. Sorry. Yes, starting from BlueZ version 5.65 experimental features really work. We also implemented full control of BLE part of BlueZ stack using sdbus-cpp library. We have project fully implemented using C++ and we decided to use this library because it has nice interface and it is easy to use. Documentation for BlueZ D-Bus interface can be found in doc folder, in GitHub repository.

zalexua, 2023-08-15:
> You can create a set of startup command and pass to bluetoothctl with -i/--init-script, that way it will wait each command to complete before starting the next.

Looking on --help, I see the -i/--init-script options are supported for bluetoothctl v5.68, but not existing in v5.66

I could not find any hint in sources to understand it.
Is there any documentation which describes how the feature may be used?

zalexua, 2023-08-15:
Yay, I managed to decrease default interval using Dbus, When using @kevincar/bless library.
Currently it's available in this fork https://github.com/zalexua/bless/commit/affb7a2688c3ffc1040d170b9ad063bcc022193e

closed 2025-08-10 by github-actions[bot]