bluez/bluez #528: The following error occurs when bluez5.53 starts bluetooth-meshd

guchunqi-git, 2023-06-06:
~$ sudo /usr/lib/bluetooth/bluetooth-meshd -nd
Mesh Crypto functions unavailable

Explanation:
        CONFIG_CRYPTO_USER
	CONFIG_CRYPTO_USER_API
	CONFIG_CRYPTO_USER_API_AEAD
	CONFIG_CRYPTO_USER_API_HASH
	CONFIG_CRYPTO_USER_API_SKCIPHER

	CONFIG_CRYPTO_AES
	CONFIG_CRYPTO_CCM
	CONFIG_CRYPTO_AEAD
	CONFIG_CRYPTO_CMAC
The above configuration is already open.
system:ubuntu 20.04 kernel 4.4.194

bgix, 2023-06-06:
Those CONFIG_CRYPTO_* functions (and any others in the bluez README files)
need to be enabled when compiling the kernel being used by the daemon. If
the “Mesh Crypto functions unavailable” error comes up, that mean the
kernel you are using doesn’t support everything the daemon needs.

guchunqi-git, 2023-06-06:
If I want to use bluez Bluetth-meshd, is there a specific requirement for the Linux kernel version, since all the configurations are already open in the README.

bgix, 2023-06-06:
Also:

1. A minimum of kernel version 4.9 or later is required

It looks like you are using kernel version 4.4

guchunqi-git, 2023-06-06:
I tried running bluez5.53 bluetooth-meshd with the 5.10 kernel version with the same error

bgix, 2023-06-06:
Did you rebuild the kernel (v5.10) yourself making sure that the required CONFIG_CRYPTO_* settings were all enabled?  Some standard distributions do not enable all of the kernal crypto versions by default.  The error being reported is basically a daemon startup test that calls into the kernel to encrypt a block of data with a specific NONCE and 128 bit key with AEAD confirmation, and expects a specific result.  Something in the kernel is not working as expected if this test fails, and mesh (which depends heavily on crypto) will not be able to communicate with remote devices if any of the required crypto isn't working.

bgix, 2023-06-06:
For instance, I know that Raspberry PI v3 and v4 kernal distro dose not come with all the needed crypto enabled.

guchunqi-git, 2023-06-06:
CONFIG_CRYPTO_USER
CONFIG_CRYPTO_USER_API
CONFIG_CRYPTO_USER_API_AEAD
CONFIG_CRYPTO_USER_API_HASH
CONFIG_CRYPTO_USER_API_SKCIPHER
CONFIG_CRYPTO_AES
CONFIG_CRYPTO_CCM
CONFIG_CRYPTO_AEAD
CONFIG_CRYPTO_CMAC
I confirmed that all the configurations included above have been enabled, recompiled the kernel, and the burning phenomenon is consistent

guchunqi-git, 2023-06-06:
How can I get more information to determine exactly which encryption library is missing?

guchunqi-git, 2023-06-06:
I install (sudo apt-get -y install bluz-meshd) bluz-meshd by adding kernel configuration to the rock pi 4 plus V1.7.3 development board and then start bluetooth-meshd. Same phenomenon?

kernel Version: 5.10
bluez version: 5.64

closed 2023-06-06 by guchunqi-git