bluez/bluez #529: Logitech MX Master 3s Connected but it doesn't works

blacklistperformer, 2023-06-08:
I have tried everything. I bought the mouse six months ago, and it used to work very well on Ubuntu. But today, as I booted up my machine, I saw the mouse wasn't working, so I did the fresh pairing it still didn't work. It works on the same device with windows os, so hardware isn't the issue, and I don't think the driver is the issue. I tried all the articles and other solutions that I found on the internet using Bluez and Bluman etc... Also, connecting from the command line with bluetoothctl.

**Further details on the issue:**
the mouse is connected successfully via Bluetooth, and it shows the same too, but when I run lsusb there is no entry for the device

neeraj@neeraj:~$ lsusb

Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 003: ID 8087:0032 Intel Corp. AX210 Bluetooth
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 13d3:56a2 IMC Networks USB2.0 HD UVC WebCam
Bus 003 Device 003: ID 0b05:194b ASUSTek Computer, Inc. TUF Gaming K3
Bus 003 Device 002: ID 2109:0100 VIA Labs, Inc. USB-C dongle
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
neeraj@neeraj:~$ 

**bluez versions**

neeraj@neeraj:~$ sudo apt install bluez
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
bluez is already the newest version (5.64-0ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
neeraj@neeraj:~$ bluetoothctl 
Agent registered
[bluetooth]# version
Version 5.66
[bluetooth]# 

Now when I manually try to connect both from setting (GUI method) and cli using bluetoothctl commands, even with restarting the Bluetooth by other commands it still doesn't work.

Here are some other kernel messages 
neeraj@neeraj:~$ sudo journalctl -b0 -g 'Error setting Report value: Unexpected error code'
Jun 09 03:02:43 neeraj sudo[6007]:   neeraj : TTY=pts/0 ; PWD=/home/neeraj ; USER=root ; COMMAND=/usr/bin/journalctl >
Jun 09 03:07:37 neeraj sudo[6159]:   neeraj : TTY=pts/0 ; PWD=/home/neeraj ; USER=root ; COMMAND=/usr/bin/journalctl >
Jun 09 03:07:56 neeraj sudo[6203]:   neeraj : TTY=pts/0 ; PWD=/home/neeraj ; USER=root ; COMMAND=/usr/bin/journalctl >
Jun 09 03:52:11 neeraj sudo[9325]:   neeraj : TTY=pts/0 ; PWD=/home/neeraj ; USER=root ; COMMAND=/usr/bin/journalctl >
lines 1-4/4 (END)

this was suggested by @InternetUnexplorer so do crating a new issue
there is  a similar issue on the same https://github.com/bluez/bluez/issues/220
#220 

and also, I thought these data might be of some importance

it is the output f this command
neeraj@neeraj:~$ sudo journalctl -b0 -g 'bluetooth'

it is in the text file below

[issue.txt](https://github.com/bluez/bluez/files/11696711/issue.txt)

blacklistperformer, 2023-06-08:
I also have posted a question on askubuntu
https://askubuntu.com/questions/1471555/logitech-mx-master-3s-connected-but-it-doesnt-works

kshamameena, 2023-06-08:
This evening I too started to have same issue.

Vudentz, 2023-06-08:
@blacklistperformer please attach the HCI logs with use of btmon

gustaphe, 2024-08-17:
I'm seeing the same issue on Arch linux with MX Master 3s and bluez 5.77-1

<details>
<summary>btmon output</summary>
```
btmon[59162]: @ RAW Close: btmon                            {0x0002} 14.065570
> HCI Event: LE Meta Event (0x3e) plen 4                   #9 [hci0] 14.066284
      LE Channel Selection Algorithm (0x14)
        Handle: 3585 Address: DF:18:2E:C5:D8:AB (Static)
        Algorithm: #2 (0x01)
< HCI Command: LE Read Remote Use.. (0x08|0x0016) plen 2  #10 [hci0] 14.098854
        Handle: 3585 Address: DF:18:2E:C5:D8:AB (Static)
> HCI Event: Command Status (0x0f) plen 4                 #11 [hci0] 14.100309
      LE Read Remote Used Features (0x08|0x0016) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 12                 #12 [hci0] 14.101269
      LE Read Remote Used Features (0x04)
        Status: Success (0x00)
        Handle: 3585 Address: DF:18:2E:C5:D8:AB (Static)
        Features: 0x05 0x41 0x00 0x00 0x00 0x00 0x00 0x00
          LE Encryption
          Extended Reject Indication
          LE 2M PHY
          Channel Selection Algorithm #2
< HCI Command: LE Start Encryption (0x08|0x0019) plen 28  #13 [hci0] 14.101488
        Handle: 3585 Address: DF:18:2E:C5:D8:AB (Static)
        Random number: 0x0000000000000000
        Encrypted diversifier: 0x0000
        Long term key[16]: 1fc33b9f877cdda4cea553937030dc2c
bluetoothd[555]: = profiles/input/hog-lib.c:uhid_destroy() bt_uh..   14.102817
bluetoothd[555]: = profiles/input/hog-lib.c:uhid_destroy() bt_uh..   14.102839
bluetoothd[555]: = src/service.c:service_accept() input-hog prof..   14.102847
> HCI Event: Command Status (0x0f) plen 4                 #14 [hci0] 14.103265
      LE Start Encryption (0x08|0x0019) ncmd 1
        Status: Success (0x00)
> HCI Event: Encryption Change (0x08) plen 4              #15 [hci0] 14.147555
        Status: Success (0x00)
        Handle: 3585 Address: DF:18:2E:C5:D8:AB (Static)
        Encryption: Enabled with AES-CCM (0x01)
< HCI Command: Write Authenticate.. (0x03|0x007c) plen 4  #16 [hci0] 14.147620
        Handle: 3585 Address: DF:18:2E:C5:D8:AB (Static)
        Timeout: 30000 msec (0x0bb8)
bluetoothd[555]: < ACL Data TX: Hand.. flags 0x00 dlen 7  #17 [hci0] 14.147760
      ATT: Exchange MTU Request (0x02) len 2
        Client RX MTU: 517
> HCI Event: Command Complete (0x0e) plen 6               #18 [hci0] 14.149310
      Write Authenticated Payload Timeout (0x03|0x007c) ncmd 1
        Status: Success (0x00)
        Handle: 3585 Address: DF:18:2E:C5:D8:AB (Static)
> HCI Event: Number of Completed Packets (0x13) plen 5    #19 [hci0] 14.155263
        Num handles: 1
        Handle: 3585 Address: DF:18:2E:C5:D8:AB (Static)
        Count: 1
        #17: len 7 (8 Kb/s)
        Latency: 7 msec (7-7 msec ~7 msec)
> ACL Data RX: Handle 3585 flags 0x02 dlen 11             #20 [hci0] 14.158792
      ATT: Read By Group Type Request (0x10) len 6
        Handle range: 0x0001-0xffff
        Attribute group type: Primary Service (0x2800)
bluetoothd[555]: < ACL Data TX: Han.. flags 0x00 dlen 24  #21 [hci0] 14.159073
      ATT: Read By Group Type Response (0x11) len 19
        Attribute data length: 6
        Attribute group list: 3 entries
        Handle range: 0x0001-0x0007
        UUID: Generic Access Profile (0x1800)
        Handle range: 0x0008-0x0011
        UUID: Generic Attribute Profile (0x1801)
        Handle range: 0x0012-0x0014
        UUID: Device Information (0x180a)
> HCI Event: Number of Completed Packets (0x13) plen 5    #22 [hci0] 14.162353
[... 260 lines cut ...]
        Latency: 5 msec (1-7 msec ~4 msec)
> HCI Event: Number of Completed Packets (0x13) plen 5    #68 [hci0] 14.290565
        Num handles: 1
        Handle: 3585 Address: DF:18:2E:C5:D8:AB (Static)
        Count: 1
        #66: len 9 (12 Kb/s)
        Latency: 6 msec (1-7 msec ~5 msec)
> ACL Data RX: Handle 3585 flags 0x02 dlen 5              #69 [hci0] 14.302173
      ATT: Write Response (0x13) len 0
> ACL Data RX: Handle 3585 flags 0x02 dlen 7              #70 [hci0] 14.302178
      ATT: Read Request (0x0a) len 2
        Handle: 0x0014 Type: PnP ID (0x2a50)
bluetoothd[555]: < ACL Data TX: Han.. flags 0x00 dlen 12  #71 [hci0] 14.302525
      ATT: Read Response (0x0b) len 7
        Value[7]: 026b1d46024d05
        Handle: 0x0014 Type: PnP ID (0x2a50)
        Value[7]: 026b1d46024d05
> HCI Event: Number of Completed Packets (0x13) plen 5    #72 [hci0] 14.305398
        Num handles: 1
        Handle: 3585 Address: DF:18:2E:C5:D8:AB (Static)
        Count: 1
        #71: len 12 (48 Kb/s)
        Latency: 2 msec (1-7 msec ~4 msec)
```
</details>

gustaphe, 2024-08-19:
`journalctl` has these, error colored:
```
Aug 17 10:02:56 user bluetoothd[555]: profiles/input/hog-lib.c:uhid_destroy() bt_uhid_destroy: Invalid argument
Aug 17 10:02:56 user bluetoothd[555]: profiles/input/hog-lib.c:uhid_destroy() bt_uhid_destroy: Invalid argument
Aug 17 10:02:56 user bluetoothd[555]: src/service.c:service_accept() input-hog profile accept failed for DF:18:2E:C5:D8:AB
```

That's in there multiple times, I'm guessing once per time I tried to reconnect (could check it but I don't have the mouse here right now).

Vudentz, 2024-08-19:
> `journalctl` has these, error colored:
>
> ```
> Aug 17 10:02:56 user bluetoothd[555]: profiles/input/hog-lib.c:uhid_destroy() bt_uhid_destroy: Invalid argument
> Aug 17 10:02:56 user bluetoothd[555]: [...]

Weird, I suspect there is something wrong with the bluetoothd permissions or something, since it is very likely it cannot open _/dev/uhid_, anyway the following should fix the bt_uhid_destroy since that is likely calling bt_uhid_destroy with a NULL pointer:

```
diff --git a/profiles/input/hog-lib.c b/profiles/input/hog-lib.c
index 1b4aca07486b..729603702e5f 100644
--- a/profiles/input/hog-lib.c
diff --git a/profiles/input/hog-lib.c b/profiles/input/hog-lib.c
index 1b4aca07486b..729603702e5f 100644
--- a/profiles/input/hog-lib.c
+++ b/profiles/input/hog-lib.c
@@ -830,14 +830,16 @@ static void uhid_destroy(struct bt_hog *hog, bool force)
 {
        int err;
 
+       if (!hog->uhid)
+               return;
+
+       bt_uhid_unregister_all(hog->uhid);
+
        err = bt_uhid_destroy(hog->uhid, force);
        if (err < 0) {
                error("bt_uhid_destroy: %s", strerror(-err));
                return;
        }
-
-       if (bt_uhid_created(hog->uhid))
-               bt_uhid_unregister_all(hog->uhid);
 }
 
 static void set_report(struct uhid_event *ev, void *user_data)
```

Now about the actual problem, can you try running and past its output: `sudo unit/test-uhid`

gustaphe, 2024-08-20:
```
Test Summary
------------
/uhid/command/create                                 Passed       0.000 seconds
/uhid/command/destroy                                Passed       0.000 seconds
/uhid/command/feature_answer                         Passed       0.000 seconds
/uhid/command/input                                  Passed       0.000 seconds
/uhid/event/output                                   Passed       0.000 seconds
/uhid/event/feature                                  Passed       0.000 seconds
/uhid/device/mx_anywhere_3                           Passed       0.083 seconds
Total: 7, Passed: 7 (100.0%), Failed: 0, Not Run: 0
```
Same result w/ and w/o `sudo`.

Caveat: I had not originally built from source but installed via Pacman, so in order to run the test I cloned the repo and built locally. The version running the test *may* not be exactly the version that was running before. 
(If this is unnecessary I don't know how to do it)

Vudentz, 2024-08-20:
> ```
> Test Summary
> ------------
> /uhid/command/create                                 Passed       0.000 seconds
> /uhid/command/destroy                                Passed       0.000 seconds [...]

Ok, any chance to test with latest git, I wanna know if that is now catching the error with bt_uid instance, etc.

gustaphe, 2024-08-20:
And now it cannot reproduce it. Sorry, it works perfectly.
Will try a couple of restarts and such. Really frustrated.

gustaphe, 2024-08-20:
Ooh, I tried reinstalling the Pacman package, and it said the permissions for `/etc/bluetooth/` changed from `755` to `555`. Reducing the permissions manually does not appear to have immediately changed anything, but there is either way a deal of hysteresis in this.

gustaphe, 2024-09-01:
It's back!
I tried `sudo btmon`. I get a bunch of  notifications when I move the mouse, I would not be surprised if they are at the correct rate, so it's on every move event.
I think this is one of the ones where the mouse moved:
```
> ACL Data RX: Handle 3585 flags 0x02 dlen 14             #24 [hci0] 11.368884
      ATT: Handle Value Notification (0x1b) len 9
        Handle: 0x0028
          Data[7]: 00000510000000
```

closed 2025-08-10 by github-actions[bot]