bluez/bluez #531: "input-hog profile accept failed" When connecting to a folio keyboard

AmirDahan, 2023-06-12:
Whenever i try to connect my lenovo duet 3igl's folio keyboard via bluetooth it fails with an error of
```
bluetoothd[87180]: src/service.c:service_accept() input-hog profile accept failed for C7:72:CF:BC:28:5C
bluetoothd[87180]: src/service.c:service_accept() input-hog profile accept failed for C7:72:CF:BC:28:5C
```

Battery info is visible and the device is still connected, Trackpad or Keyboard inputs do not work however.
The weird thing is that lenovo support and a few forms have said to try enabling caps lock before disconnecting,
and that actually seemed to work a while back (Very inconsistent though)

Any help would be appreciated.

Bluez: 5.66-1
OS: Arch linux

terrancewong, 2023-11-15:
This is very weird. I am using Debian testing / unstable. All 4 computers running  Linux 6.5, bluez 5.70, XFCE. And I have M585, MX Master 3S ,and MX Anywhere 3, and other cheap BT mice..

- one 12-gen Intel laptop has this issue only with logi mice. connected but not moving neither clicking. not appeared in xinput.
- one AMD R7-4800H laptop, all mice works fine.
- one Intel 8-gen laptop, all mice work fine
- another Intel 8-gen laptop, all mice work fine

I already messed with kernel options (removed all), udev rules, modprobe blacklists. Running out of ideas now. what else can I poke around?

terrancewong, 2024-01-27:
update: I swap SSDs between 2 computers, the issue moves with the SSD(OS). also the Cheap Ugreen mouse is dual mode. its BLE neither works, throws same error. Bluetooth 3 mode works.

attachment: log of `/usr/libexec/bluetooth/bluetoothd --debug`
[20240127.ble.issue.ligitech.log](https://github.com/bluez/bluez/files/14070800/20240127.ble.issue.ligitech.log)

terrancewong, 2024-01-27:
compared 2 machine's dmesg about bluetooth. diff:
```
 % diff <(cut -d ']' -f2 dmessage.fail.log) <(cut -d ']' -f2 dmessage.ok.log)
7,8c7
<  Bluetooth: hci0: Bootloader revision 0.3 build 0 week 24 2017
<  Bluetooth: hci0: Device revision is 1
---
>  Bluetooth: hci0: Device revision is 2
14,17c13,17
<  bluetooth hci0: firmware: direct-loading firmware intel/ibt-20-1-3.sfi
<  Bluetooth: hci0: Found device firmware: intel/ibt-20-1-3.sfi
<  Bluetooth: hci0: Boot Address: 0x24800
<  Bluetooth: hci0: Firmware Version: 255-255.255
---
>  Bluetooth: hci0: Bootloader timestamp 2019.40 buildtype 1 build 38
>  bluetooth hci0: firmware: direct-loading firmware intel/ibt-0040-4150.sfi
>  Bluetooth: hci0: Found device firmware: intel/ibt-0040-4150.sfi
>  Bluetooth: hci0: Boot Address: 0x100800
>  Bluetooth: hci0: Firmware Version: 98-13.23
22c22
<  Bluetooth: hci0: Firmware loaded in 2011017 usecs
---
>  Bluetooth: hci0: Firmware loaded in 1570624 usecs
24,26c24,26
<  Bluetooth: hci0: Device booted in 14629 usecs
<  bluetooth hci0: firmware: direct-loading firmware intel/ibt-20-1-3.ddc
<  Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-20-1-3.ddc
---
>  Bluetooth: hci0: Device booted in 15791 usecs
>  bluetooth hci0: firmware: direct-loading firmware intel/ibt-0040-4150.ddc
>  Bluetooth: hci0: Found Intel DDC parameters: intel/ibt-0040-4150.ddc
28,29c28
<  Bluetooth: hci0: Firmware revision 0.3 build 144 week 11 2023
<  Bluetooth: hci0: HCI LE Coded PHY feature bit is set, but its usage is not supported.
---
>  Bluetooth: hci0: Firmware timestamp 2023.13 buildtype 1 build 62562
31,32d29
<  Bluetooth: hci0: Bad flag given (0x1) vs supported (0x0)
<  Bluetooth: hci0: Bad flag given (0x1) vs supported (0x0)
36,44c33,46
<  Bluetooth: HIDP (Human Interface Emulation) ver 1.2
<  Bluetooth: HIDP socket layer initialized
<  Bluetooth: hci0: Ignoring error of Inquiry Cancel command
<  Bluetooth: hci0: Ignoring error of Inquiry Cancel command
<  input: UGREEN B3.0 Mouse as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-4/1-4.4/1-4.4:1.0/bluetooth/hci0/hci0:256/0005:2B89:3632.0002/input/input14
<  hid-generic 0005:2B89:3632.0002: input,hidraw1: BLUETOOTH HID v0.01 Mouse [UGREEN B3.0 Mouse
<  Bluetooth: hci0: Ignoring error of Inquiry Cancel command
<  input: UGREEN B3.0 Mouse as /devices/pci0000:00/0000:00:08.1/0000:04:00.3/usb1/1-4/1-4.4/1-4.4:1.0/bluetooth/hci0/hci0:256/0005:2B89:3632.0003/input/input15
<  hid-generic 0005:2B89:3632.0003: input,hidraw1: BLUETOOTH HID v0.01 Mouse [UGREEN B3.0 Mouse
---
>  hid-generic 0005:046D:B025.0006: input,hidraw5: BLUETOOTH HID v0.14 Mouse [MX Anywhere 3
>  Bluetooth: hci0: Bad flag given (0x1) vs supported (0x0)
>  hid-generic 0005:046D:B025.0007: input,hidraw5: BLUETOOTH HID v0.14 Mouse [MX Anywhere 3
>  hid-generic 0005:046D:B025.0008: input,hidraw5: BLUETOOTH HID v0.14 Mouse [MX Anywhere 3
>  hid-generic 0005:046D:B025.0009: input,hidraw5: BLUETOOTH HID v0.14 Mouse [MX Anywhere 3
>  hid-generic 0005:046D:B025.000A: input,hidraw5: BLUETOOTH HID v0.14 Mouse [MX Anywhere 3
>  hid-generic 0005:046D:B025.000B: input,hidraw5: BLUETOOTH HID v0.14 Mouse [MX Anywhere 3
>  hid-generic 0005:046D:B025.000C: input,hidraw5: BLUETOOTH HID v0.14 Mouse [MX Anywhere 3
>  hid-generic 0005:046D:B025.000D: input,hidraw5: BLUETOOTH HID v0.14 Mouse [MX Anywhere 3
>  hid-generic 0005:046D:B025.000E: input,hidraw5: BLUETOOTH HID v0.14 Mouse [MX Anywhere 3
>  hid-generic 0005:046D:B025.000F: input,hidraw5: BLUETOOTH HID v0.14 Mouse [MX Anywhere 3
>  hid-generic 0005:046D:B025.0010: input,hidraw5: BLUETOOTH HID v0.14 Mouse [MX Anywhere 3
>  hid-generic 0005:046D:B025.0011: input,hidraw5: BLUETOOTH HID v0.14 Mouse [MX Anywhere 3
>  hid-generic 0005:046D:B025.0012: input,hidraw5: BLUETOOTH HID v0.14 Mouse [MX Anywhere 3
```

Hardware and mouse connection difference should be fine. But the fail one has HIDP, what is HIDP??
[dmessage.fail.log](https://github.com/bluez/bluez/files/14071409/dmessage.fail.log)
[dmessage.ok.log](https://github.com/bluez/bluez/files/14071410/dmessage.ok.log)

terrancewong, 2024-01-27:
holy smoke! I fixed this, after months of research. 

hidp/uhid really IS the problem. I don't know what makes a Linux decide to load `uhid` or `hidp`, my "good" and "bad" Linux SSD are dd'ed from same source, and they even have same partition uuid.

long story short, blacklist hidp and force load uhid kernel module, all mice work happily now!

```
echo "blacklist hidp" | sudo tee /etc/modprobe.d/blacklist-hidp.conf
echo -e  "# Fix BLE mouse issue\nuhid" | sudo tee /etc/modules-load.d/uhid.conf
cp --update=none /etc/bluetooth/input.conf{,backup}
sudo sed -ie "s:.*UserspaceHID=.*:UserspaceHID=true:" /etc/bluetooth/input.conf
```

then reboot.

WhyNotHugo, 2024-04-01:
@terrancewong's workaround fixes the problem for me too. Should udev include a rule to enable this module in case this device is seen? Or would udev rule execution happen too late?

AmirDahan, 2024-04-07:
This seems to let the device connect, However i'm still not getting any input from the trackpad or keyboard.
Here's my log: [bluetooth.log](https://github.com/bluez/bluez/files/14895854/bluetooth.log)

Probable cause could be the errors i'm getting in dmesg?
```
[  586.346765] input: Duet 3 BT Keyboard as /devices/virtual/misc/uhid/0005:17EF:60FA.0009/input/input49
[  586.347167] input: Duet 3 BT Wireless Radio Control as /devices/virtual/misc/uhid/0005:17EF:60FA.0009/input/input50
[  586.347339] hid-generic 0005:17EF:60FA.0009: input,hidraw5: BLUETOOTH HID v0.08 Keyboard [Duet 3 BT] on 60:f2:62:87:
[  586.357638] hid-multitouch 0005:17EF:60FA.0008: failed to fetch feature 8
[  586.376475] hid-multitouch 0005:17EF:60FA.0008: failed to fetch feature 8
[  586.409652] hid-multitouch 0005:17EF:60FA.0008: failed to fetch feature 14
[  586.409835] input: Duet 3 BT Touchpad as /devices/virtual/misc/uhid/0005:17EF:60FA.0008/input/input51
[  586.410035] hid-multitouch 0005:17EF:60FA.0008: input,hidraw6: BLUETOOTH HID v0.08 Device [Duet 3 BT] on 60:f2:62:87:44:f6
```