bluez/bluez #532: bluez SEGV in current master

joakim-tjernlund, 2023-06-13:
I coworker see this SEGV when connecting a headset:
```
Jun 13 13:38:15 se-athatava2-lx.infinera.com systemd-coredump[3941]: Process 1897 (bluetoothd) of user 0 dumped core.
                                                                     Module bluetoothd without build-id.
                                                                     #0  0x0000556bea93277a a2dp_resume_complete (bluetoothd + 0x3e77a)
                                                                     #1  0x0000556bea9210a2 finalize_resume (bluetoothd + 0x2d0a2)
                                                                     #2  0x0000556bea92afaf avdtp_start_resp (bluetoothd + 0x36faf)
                                                                     #6  0x0000556bea9dcc85 mainloop_run (bluetoothd + 0xe8c85)
                                                                     #7  0x0000556bea9dd0dc mainloop_run_with_signal (bluetoothd + 0xe90dc)
                                                                     #8  0x0000556bea91a4ef main (bluetoothd + 0x264ef)
                                                                     #11 0x0000556bea91ace1 _start (bluetoothd + 0x26ce1)
Jun 13 13:38:15 se-athatava2-lx.infinera.com systemd[1]: bluetooth.service: Main process exited, code=dumped, status=11/SEGV
```

joakim-tjernlund, 2023-06-13:
Crash seems to be at:
```
342	static void a2dp_resume_complete(struct avdtp *session, int err,
343								void *user_data)
344	{
345		struct media_owner *owner = user_data;
346		struct media_request *req = owner->pending;
347		struct media_transport *transport = owner->transport;
348		struct a2dp_sep *sep = media_endpoint_get_sep(transport->endpoint);
```
Suggests that transport ptr is invalid.

joakim-tjernlund, 2023-06-13:
dunno if this helps but seems like a good first step:
```
--- profiles/audio/transport.c.org	2023-06-13 18:38:14.542137531 +0200
+++ profiles/audio/transport.c	2023-06-13 18:39:22.110965184 +0200
@@ -345,7 +345,7 @@
 	struct media_owner *owner = user_data;
 	struct media_request *req = owner->pending;
 	struct media_transport *transport = owner->transport;
-	struct a2dp_sep *sep = media_endpoint_get_sep(transport->endpoint);
+	struct a2dp_sep *sep;
 	struct avdtp_stream *stream;
 	int fd;
 	uint16_t imtu, omtu;
@@ -356,6 +356,7 @@
 	if (err)
 		goto fail;
 
+	sep = media_endpoint_get_sep(transport->endpoint);
 	stream = a2dp_sep_get_stream(sep);
 	if (stream == NULL)
 		goto fail;
```

joakim-tjernlund, 2023-06-13:
No, above hack did not help

jonathan-conder, 2023-06-30:
I'm seeing this too, with `bluez 5.67-1` on Arch. Can confirm owner->transport is NULL:
```
Program received signal SIGSEGV, Segmentation fault.
a2dp_resume_complete (session=0x5607705ae810, err=0, user_data=0x5607705af320) at profiles/audio/transport.c:348
348             struct a2dp_sep *sep = media_endpoint_get_sep(transport->endpoint);                                  
(gdb) p *owner
$1 = {transport = 0x0, pending = 0x0, name = 0x5607705973d0 ":1.68", watch = 0}
```

jonathan-conder, 2023-06-30:
looks like this was fixed: https://github.com/bluez/bluez/issues/542

joakim-tjernlund, 2023-07-01:
> looks like this was fixed: #542

It does, will update. Thanks

closed 2023-07-01 by joakim-tjernlund