bluez/bluez #540: [bluez 5.66] Segfault while sending mesh message

venngar, 2023-06-27:
I've had bluetooth-mesh service running on Raspberry Pi 4 running Ubuntu 23.04, and suddenly in one of them I've noticed that BlueZ seems to crash. Here are examples of logs

```
bluetooth-meshd[3161]: mesh/util.c:print_packet() 72245.429 Mesh Send Start: 2a444ea302ff01add7fa0c2610c0792ae0591a213092aa9e

bluetooth-meshd[3161]: mesh/util.c:print_packet() 72245.429 Mesh Send Complete: 5c

bluetooth-meshd[3161]: mesh/mesh-io-mgmt.c:send_queued() Mesh Send Failed: 3

bluetooth-meshd[3161]: free(): double free detected in tcache 2

systemd[1]: bluetooth-mesh.service: Main process exited, code=dumped, status=6/ABRT

systemd[1]: bluetooth-mesh.service: Failed with result 'core-dump'.
```

Any idea why it might happen? Could there be a bug in mesh code?

bgix, 2023-06-27:
Is this reproducable?  Have you ran it on valgrind and/or gdb to catch the moment of failure?

bgix, 2023-06-27:
also:  What is your kernal version and BlueZ version?

venngar, 2023-06-27:
On reproducing: that's the problem - it gets triggered only on one of the RPi devices, not on all of them, so I'm kinda blind about trying to reproduce.

Kernel version: 6.2.0-1004-raspi
BlueZ version: 5.66

My laptop isn't too powerful to run valgrind (it starts freezing significantly), I will upload any gdb results if I'm able to catch it (I have VSCode project setup, which includes debugging config).

venngar, 2023-06-28:
Few updates.

Sadly when I try to run valgrind on the Raspberry Pi that threw this error, it freezes significantly. And I can't reproduce it on other Raspberry Pi devices. So far I've only managed to confirm that error happens after io_send returns something less than 0 (which means it can't send message on a low level).

Could the problem be with the fact that compilation is happening on Ubuntu 20.x machines, and the installation (as a deb package) is on Ubuntu 23?

bgix, 2023-06-28:
I doubt if the cross-compilation is causing the problem. Especially if the gcc compilers are otherwise up-to-date.

I will examine the code in question, focusing on the "Mesh Send Complete" followed by the "Mesh Send Failed: 3" to see if I can find a vulnerable path.  What else can you tell me about the problem?  How long is the RPI4 mesh daemon running before SEG Fault? How frequently is it sending mesh packets?

bgix, 2023-06-29:
So there are two methods for passing Mesh packets through the kernel to the controller:  Kernel v6.0 and lower uses an HCI user socket (via mesh-io-generic.c), and Kernel v6.1 and higher uses the new MGMT interface (via mesh-io-mgmt.c). That explains the difference in behavior. Unfortunately, it is the mesh-io-mgmt that we want to be working for the future.

venngar, 2023-06-29:
> So there are two methods for passing Mesh packets through the kernel to the controller: Kernel v6.0 and lower uses an HCI user socket (via mesh-io-generic.c), and Kernel v6.1 and higher uses the new [...]

Meaning it's certainly a bug within mesh-io-mgmt which will be fixed in future?

bgix, 2023-06-29:
> Meaning it's certainly a bug within mesh-io-mgmt which will be fixed in future?

Working on it now

bgix, 2023-06-30:
So there was a v5.67 of BlueZ released today...  It has an identified (and fixed) bug in the bluetoothd daemon, but the bluetooth-meshd daemon is so far unchanged.  The version you are using (v5.66) was made last November and had a number of stability fixes added between then and today. Can you run the BlueZ 5.67 Mesh daemon and see if it has the same problems?  And if it does, you may need to back out patch c1f7aed635f39578cdb4a1870413278d29a081c5, which removes the logging you saw right before the double-free fault.

venngar, 2023-06-30:
> So there was a v5.67 of BlueZ released today... It has an identified (and fixed) bug in the bluetoothd daemon, but the bluetooth-meshd daemon is so far unchanged. The version you are using (v5.66) was made last November and had a number of stability fixes added between then and today. Can you run the BlueZ 5.67 Mesh daemon and see if it has the same problems? And if it does, you may need to back out patch [c1f7aed](https://github.com/bluez/bluez/commit/c1f7aed635f39578cdb4a1870413278d29a081c5), which removes the logging you saw right before the double-free fault.

Thanks! Will post here any updates after testing.

venngar, 2023-07-03:
So I've looked at Bluetooth Mesh changes added in 5.68, and it seems I'll have to do some major work on application level (since it's now done over mesh model). I don't know how soon I can test it, but can you please still keep me updated on when the fix will be added?

bgix, 2023-07-03:
Is v5.68 still Seg Faulting in the same way as v5.66?  If so, can you re-enable the debug messages that were disabled by https://github.com/bluez/bluez/commit/c1f7aed635f39578cdb4a1870413278d29a081c5 so that the same message flow can be confirmed?

I will add you to the Cc: line if I submit a patch that specifically fixes it.

venngar, 2023-07-03:
I'll attach the logs as soon as I'll be able to test it.

venngar, 2023-07-07:
So, in an attempt to test... I've encountered another issue: in 5.68, BlueZ doesn't see any mesh messages at all in the network, even unprovisioned beacons. 5.66 doesn't have this issue. I did some debugging (with provisioned previously devices) and it seems that in net.c, inside mesh_net_attach this call

`mesh_io_register_recv_cb(io, pkt, sizeof(pkt), net_msg_recv, NULL);`

doesn't work and the breakpoint in net_msg_recv function of net.c isn't triggered at all. It works fine in 5.66 though.

In addition to that, usually when mesh message arrives, the value of **_hdr->evt_** in process_event function of src/shared/hci.c is BT_HCI_EVT_LE_META_EVENT (0x3e) opcode. That's how it is in 5.66 and it's working well. But in 5.68 it's not.

Here are logs of 5.66:

`Loading node configuration from /var/lib/bluetooth/mesh
Loading configuration from /var/lib/bluetooth/mesh/cd6a8f3c13634364ab16b827eb93deb6/node.json
mesh/node.c:init_from_storage() relay 01, proxy 02, lpn 02, friend 00
mesh/model.c:add_sub() Added group subscription c000
mesh/model.c:add_sub() Added group subscription c000
mesh/model.c:add_sub() Added group subscription c000
mesh/model.c:add_sub() Added group subscription c000
mesh/model.c:add_sub() Added group subscription c000
mesh/model.c:add_sub() Added group subscription c000
mesh/model.c:add_sub() Added group subscription c000
mesh/model.c:add_sub() Added group subscription c000
mesh/model.c:add_sub() Added group subscription c000
mesh/model.c:add_sub() Added group subscription c000
mesh/model.c:add_sub() Added group subscription c000
mesh/model.c:add_sub() Added group subscription c000
mesh/model.c:add_sub() Added group subscription c000
mesh/model.c:add_sub() Added group subscription c000
mesh/model.c:add_sub() Added group subscription c000
mesh/net-keys.c:net_key_beacon_refresh() Setting SNB: IVI: 00000000, IVU: 0, KR: 0
mesh/util.c:print_packet() 49747.555 Set SNB Beacon to: 2b01002241a21365ee42ff00000000fcea045774b59951
mesh/net.c:trigger_heartbeat() HB: 0002 --> 0
mesh/net.c:mesh_net_set_friend_mode() mesh_net_set_friend_mode - 0
mesh/net.c:trigger_heartbeat() HB: 0001 --> 1
mesh/cfgmod-server.c:cfgmod_server_init() 00
mesh/mesh-mgmt.c:mesh_mgmt_list() send read index_list
mesh/mesh.c:mesh_init() io 0xaaaaaab897e0
mesh/mesh-mgmt.c:read_index_list_cb() Number of controllers: 1
mesh/mesh-mgmt.c:read_info_cb() hci 0 status 0x00
mesh/mesh-mgmt.c:read_info_cb() settings: supp 0001beff curr 00000080
mesh/mesh-mgmt.c:features_cb() Status: 1, Length: 0
mesh/mesh-mgmt.c:features_cb() Kernel mesh not supported for hci0
up:1 pwr: 0 mesh: 0
mesh/mesh-io-generic.c:hci_init() Started mesh on hci 0
mesh_ready_callback
Added Network Interface on /org/bluez/mesh
[DBUS] > 6c 04 01 01 38 00 00 00 04 00 00 00 68 00 00 00  l...8.......h...
[DBUS]   01 01 6f 00 01 00 00 00 2f 00 00 00 00 00 00 00  ..o...../.......
[DBUS]   03 01 73 00 0f 00 00 00 49 6e 74 65 72 66 61 63  ..s.....Interfac
[DBUS]   65 73 41 64 64 65 64 00 02 01 73 00 22 00 00 00  esAdded...s."...
[DBUS]   6f 72 67 2e 66 72 65 65 64 65 73 6b 74 6f 70 2e  org.freedesktop.
[DBUS]   44 42 75 73 2e 4f 62 6a 65 63 74 4d 61 6e 61 67  DBus.ObjectManag
[DBUS]   65 72 00 00 00 00 00 00 08 01 67 00 0a 6f 61 7b  er........g..oa{
[DBUS]   73 61 7b 73 76 7d 7d 00                          sa{sv}}.        
[DBUS]   0f 00 00 00 2f 6f 72 67 2f 62 6c 75 65 7a 2f 6d  ..../org/bluez/m
[DBUS]   65 73 68 00 20 00 00 00 17 00 00 00 6f 72 67 2e  esh. .......org.
[DBUS]   62 6c 75 65 7a 2e 6d 65 73 68 2e 4e 65 74 77 6f  bluez.mesh.Netwo
[DBUS]   72 6b 31 00 00 00 00 00                          rk1.....        
LE Scan disable failed (0x0c)
mesh/util.c:print_packet() 49750.041 RX: Network [enc] :: 5600a439e008c61bb99b48c6633a81be09105752e609322f51974c4d
mesh/util.c:print_packet() 49750.041 RX: Network [clr] :: 56070023ed00c1c00046c51523bc884a206808404b92aa81
mesh/net.c:msg_in_cache() Add 00c1 + 0023ed + 20680840
mesh/net.c:packet_received() RX: Network 00c1 -> c000 : TTL 0x07 : IV : 00000000 SEQ 0x0023ed
mesh/net.c:packet_received() RX: App 0x00c1 -> 0xc000 : TTL 0x07 : SEQ 0x0023ed
mesh/model.c:mesh_model_rx() iv_index 00000000 key_aid = 46
mesh/util.c:print_packet() 49750.041 Used App Key: fa45f4ea1f7d117f4ba48014787c6bba
mesh/util.c:print_packet() 49750.041 Clr Rx: c6aeaa00006d0c070801
mesh/util.c:print_packet() 49750.133 RX: Network [enc] :: 56e712861a3e3b1f6abecd66549a93b0885285c93a75c0269b69c1c1
mesh/util.c:print_packet() 49750.134 RX: Network [clr] :: 56060023ed00c1c00046c51523bc884a206808404b92aa81
mesh/net.c:msg_in_cache() Supressing duplicate 00c1 + 0023ed + 20680840
mesh/util.c:print_packet() 49755.925 RX: Network [enc] :: 56499d66318d2f9d8ad801c2f25ec86c351f2cd7758eb260d036a8d4
[... 62 lines cut ...]
mesh/mesh-mgmt.c:read_info_cb() settings: supp 0001beff curr 00000080
mesh/mesh-mgmt.c:features_cb() Status: 1, Length: 0
mesh/mesh-mgmt.c:features_cb() Kernel mesh not supported for hci0
index 0 up:1 pwr: 0 mesh: 0
mesh/mesh-io-generic.c:hci_init() Started mesh on hci 0
mesh_ready_callback
Added Network Interface on /org/bluez/mesh
[DBUS] > 6c 04 01 01 38 00 00 00 04 00 00 00 68 00 00 00  l...8.......h...
[DBUS]   01 01 6f 00 01 00 00 00 2f 00 00 00 00 00 00 00  ..o...../.......
[DBUS]   03 01 73 00 0f 00 00 00 49 6e 74 65 72 66 61 63  ..s.....Interfac
[DBUS]   65 73 41 64 64 65 64 00 02 01 73 00 22 00 00 00  esAdded...s."...
[DBUS]   6f 72 67 2e 66 72 65 65 64 65 73 6b 74 6f 70 2e  org.freedesktop.
[... cut ...]
[DBUS]   72 6b 31 00 00 00 00 00                          rk1.....        
mesh/util.c:print_packet() 50619.885 Set SNB to: 2b01002241a21365ee42ff00000000fcea045774b59951
mesh/net-keys.c:net_key_beacon_refresh() Set Beacon: IVI: 00000000, IVU: 0, KR: 0
mesh/net-keys.c:beacon_timeout() beacon 1 for 1 nodes, period 20, obs 2, exp 3
```

_Tested on Ubuntu 20.04 LTS though, with kernel version 5.4.0-1052-raspi - that one is my main and primary Pi, while the one with Ubuntu 23.04 is a secondary so I wanted to make it work on primary first._

venngar, 2023-07-11:
Found the reason why. It appears in mesh_io.c, function mesh_io_register_recv_cb, BlueZ doesn't start AD scanning, which is why it doesn't receive mesh messages.

closed 2025-08-09 by github-actions[bot]