bluez/bluez #543: Can't connect to bluetooth device

laloshifrin, 2023-06-30:
Hello, I can't connect to A2DP device either from Blueman or bluetootctl.

Here are some info:
lubuntu 18.04
kernel 5.0.0
bluez 5.48

this is Wireshark log on attempting connection:
1	0.000000	        host	controller	HCI_CMD	11	Sent LE Remove Device From White List
2	0.119013	        controller	host	HCI_EVT 	7	Rcvd Command Complete (LE Remove Device From White List)
3	0.119181	        host	controller	HCI_CMD	11	Sent LE Add Device To White List
4	0.120001	        controller	host	HCI_EVT	        7	Rcvd Command Complete (LE Add Device To White List)
5	0.120069	        host	controller	HCI_CMD	11	Sent LE Set Scan Parameters
6	0.121011	        controller	host	HCI_EVT 	7	Rcvd Command Complete (LE Set Scan Parameters)
7	0.121099	        host	controller	HCI_CMD	6	Sent LE Set Scan Enable
8	0.122051	        controller	host	HCI_EVT 	7	Rcvd Command Complete (LE Set Scan Enable)
9	40.981680	host	controller	HCI_CMD	6	Sent LE Set Scan Enable
10	41.098501	controller	host	HCI_EVT 	7	Rcvd Command Complete (LE Set Scan Enable)
after a while (timeout) bluetoothctl says: Failed to connect: org.bluez.Error.Failed

after issuing this command:
sdptool search --bdaddr XX:XX:XX:XX:XX:XX A2SNK
and retrying connection works flawlessly.
Can anybody help me?
Thank you

castillofrancodamian, 2023-07-01:
I also can't connect to my Bluetooth headphones or speaker after updating. I have Arch Linux.

castillofrancodamian, 2023-07-01:
Fixed in the new [version](https://github.com/bluez/bluez/commit/d764f78f27653bc1df71c462e9aca7a18bc75f9f).