bluez/bluez #548: 5.67: Headset disconnects right after connecting

UltraBlackLinux, 2023-07-01:
Hey there,
since I upgraded to 5.67, my bluetooth headset on my arch machine disconnects right after I can hear that audio profile initialization click. 5.66 works as expected.

I'm using Pipewire and Wireplumber.

Any idea what I could try?
Thanks!

bluetoothctl log:
```
[NEW] Endpoint /org/bluez/hci1/dev_00_1B_66_E8_CB_52/sep8
[NEW] Endpoint /org/bluez/hci1/dev_00_1B_66_E8_CB_52/sep6
[NEW] Endpoint /org/bluez/hci1/dev_00_1B_66_E8_CB_52/sep5
[NEW] Endpoint /org/bluez/hci1/dev_00_1B_66_E8_CB_52/sep3
[NEW] Endpoint /org/bluez/hci1/dev_00_1B_66_E8_CB_52/sep1
[NEW] Transport /org/bluez/hci1/dev_00_1B_66_E8_CB_52/sep8/fd0
Connection successful
[DEL] Media /org/bluez/hci1
	SupportedUUIDs: 0000110a-0000-1000-8000-00805f9b34fb
	SupportedUUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[DEL] Endpoint /org/bluez/hci1/dev_00_1B_66_E8_CB_52/sep8
[DEL] Endpoint /org/bluez/hci1/dev_00_1B_66_E8_CB_52/sep6
[DEL] Endpoint /org/bluez/hci1/dev_00_1B_66_E8_CB_52/sep5
[DEL] Endpoint /org/bluez/hci1/dev_00_1B_66_E8_CB_52/sep3
[DEL] Endpoint /org/bluez/hci1/dev_00_1B_66_E8_CB_52/sep1
[DEL] Transport /org/bluez/hci1/dev_00_1B_66_E8_CB_52/sep8/fd0
Agent unregistered
[DEL] Controller 5C:F3:70:8E:8F:4F lolcat [default]
Agent registered
[NEW] Media /org/bluez/hci1
	SupportedUUIDs: 0000110a-0000-1000-8000-00805f9b34fb
	SupportedUUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[NEW] Controller 5C:F3:70:8E:8F:4F maus [default]
[NEW] Device 00:1B:66:E8:CB:52 HD 350BT
[CHG] Controller 5C:F3:70:8E:8F:4F Name: lolcat
[CHG] Controller 5C:F3:70:8E:8F:4F Alias: lolcat
[CHG] Controller 5C:F3:70:8E:8F:4F Class: 0x00000104 (260)
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001133-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001105-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001132-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001104-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00005005-0000-1000-8000-0002ee000001
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000112f-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001106-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001133-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001105-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001132-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001104-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00005005-0000-1000-8000-0002ee000001
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000112f-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001106-0000-1000-8000-00805f9b34fb
[... 175 lines cut ...]
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001106-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F Class: 0x007c0104 (8126724)
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001133-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000110e-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001105-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001132-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001104-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00005005-0000-1000-8000-0002ee000001
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000110c-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001801-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000112f-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000180a-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000110b-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001800-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000111f-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000110a-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 00001106-0000-1000-8000-00805f9b34fb
[CHG] Controller 5C:F3:70:8E:8F:4F UUIDs: 0000111e-0000-1000-8000-00805f9b34fb
[CHG] Device 00:1B:66:E8:CB:52 Connected: no
```

jotomo, 2023-07-01:
Sound like https://github.com/bluez/bluez/issues/542 if you're seeing also seeing crashes of bluetoothd via journalctl. Release 5.68 fixes this. Arch bug about it: https://bugs.archlinux.org/task/78937

UltraBlackLinux, 2023-07-01:
oh yes, that's it, thank you!

closed 2023-07-01 by UltraBlackLinux