bluez/bluez #549: Headphones segfaults bluez after update

Draic, 2023-07-01:
After a recent update my headphones don't work with my laptop anymore. Connecting them will lead to bluez crashing the moment it tries to switch audio

```
[   46.006663] input: DROP Panda (AVRCP) as /devices/virtual/input/input59
[   49.405764] bluetoothd[823]: segfault at 20 ip 0000562b9446dfa5 sp 00007ffcd1cb9a20 error 4 in bluetoothd[562b94452000+d4000] likely on CPU 6 (core 12, socket 0)
[   49.405774] Code: 08 00 0f 1f 00 f3 0f 1e fa 41 55 41 54 55 53 48 83 ec 18 48 8b 2a 4c 8b 62 08 64 48 8b 04 25 28 00 00 00 48 89 44 24 08 31 c0 <48> 8b 45 20 48 8b 00 41 c7 44 24 08 00 00 00 00 85 f6 74 27 48 89
[   55.510504] bluetoothd[3162]: segfault at 20 ip 0000558796990fa5 sp 00007fff8350f770 error 4 in bluetoothd[558796975000+d4000] likely on CPU 0 (core 0, socket 0)
[   55.510514] Code: 08 00 0f 1f 00 f3 0f 1e fa 41 55 41 54 55 53 48 83 ec 18 48 8b 2a 4c 8b 62 08 64 48 8b 04 25 28 00 00 00 48 89 44 24 08 31 c0 <48> 8b 45 20 48 8b 00 41 c7 44 24 08 00 00 00 00 85 f6 74 27 48 89
[   61.729048] bluetoothd[3182]: segfault at 20 ip 000055e333f4bfa5 sp 00007fff348c0e70 error 4 in bluetoothd[55e333f30000+d4000] likely on CPU 8 (core 16, socket 0)
[   61.729058] Code: 08 00 0f 1f 00 f3 0f 1e fa 41 55 41 54 55 53 48 83 ec 18 48 8b 2a 4c 8b 62 08 64 48 8b 04 25 28 00 00 00 48 89 44 24 08 31 c0 <48> 8b 45 20 48 8b 00 41 c7 44 24 08 00 00 00 00 85 f6 74 27 48 89
[   65.695241] bluetoothd[3198]: segfault at 20 ip 0000558f3f230fa5 sp 00007ffcdaf7f9c0 error 4 in bluetoothd[558f3f215000+d4000] likely on CPU 2 (core 4, socket 0)
[   65.695251] Code: 08 00 0f 1f 00 f3 0f 1e fa 41 55 41 54 55 53 48 83 ec 18 48 8b 2a 4c 8b 62 08 64 48 8b 04 25 28 00 00 00 48 89 44 24 08 31 c0 <48> 8b 45 20 48 8b 00 41 c7 44 24 08 00 00 00 00 85 f6 74 27 48 89
[   72.534533] bluetoothd[3209]: segfault at 20 ip 0000555f1cfc1fa5 sp 00007ffc05014b40 error 4 in bluetoothd[555f1cfa6000+d4000] likely on CPU 18 (core 30, socket 0)
[   72.534551] Code: 08 00 0f 1f 00 f3 0f 1e fa 41 55 41 54 55 53 48 83 ec 18 48 8b 2a 4c 8b 62 08 64 48 8b 04 25 28 00 00 00 48 89 44 24 08 31 c0 <48> 8b 45 20 48 8b 00 41 c7 44 24 08 00 00 00 00 85 f6 74 27 48 89
[   76.890977] bluetoothd[3672]: segfault at 20 ip 000055639f8a6fa5 sp 00007ffe91286240 error 4 in bluetoothd[55639f88b000+d4000] likely on CPU 6 (core 12, socket 0)
[   76.890988] Code: 08 00 0f 1f 00 f3 0f 1e fa 41 55 41 54 55 53 48 83 ec 18 48 8b 2a 4c 8b 62 08 64 48 8b 04 25 28 00 00 00 48 89 44 24 08 31 c0 <48> 8b 45 20 48 8b 00 41 c7 44 24 08 00 00 00 00 85 f6 74 27 48 89
[  554.927462] input: DROP Panda (AVRCP) as /devices/virtual/input/input60
[  603.610097] input: DROP Panda (AVRCP) as /devices/virtual/input/input61
[  622.466806] input: DROP Panda (AVRCP) as /devices/virtual/input/input62
[  622.486742] bluetoothd[3687]: segfault at 20 ip 000055871b982fa5 sp 00007ffe607f8ab0 error 4 in bluetoothd[55871b967000+d4000] likely on CPU 4 (core 8, socket 0)
[  622.486753] Code: 08 00 0f 1f 00 f3 0f 1e fa 41 55 41 54 55 53 48 83 ec 18 48 8b 2a 4c 8b 62 08 64 48 8b 04 25 28 00 00 00 48 89 44 24 08 31 c0 <48> 8b 45 20 48 8b 00 41 c7 44 24 08 00 00 00 00 85 f6 74 27 48 89
[  625.736991] input: DROP Panda (AVRCP) as /devices/virtual/input/input63
[  627.853300] bluetoothd[5578]: segfault at 20 ip 0000561492f04fa5 sp 00007ffd408f5030 error 4 in bluetoothd[561492ee9000+d4000] likely on CPU 6 (core 12, socket 0)
[  627.853310] Code: 08 00 0f 1f 00 f3 0f 1e fa 41 55 41 54 55 53 48 83 ec 18 48 8b 2a 4c 8b 62 08 64 48 8b 04 25 28 00 00 00 48 89 44 24 08 31 c0 <48> 8b 45 20 48 8b 00 41 c7 44 24 08 00 00 00 00 85 f6 74 27 48 89
[  629.209910] bluetoothd[5661]: segfault at 20 ip 000055d466ccdfa5 sp 00007ffcd4016c20 error 4 in bluetoothd[55d466cb2000+d4000] likely on CPU 10 (core 20, socket 0)
[  629.209921] Code: 08 00 0f 1f 00 f3 0f 1e fa 41 55 41 54 55 53 48 83 ec 18 48 8b 2a 4c 8b 62 08 64 48 8b 04 25 28 00 00 00 48 89 44 24 08 31 c0 <48> 8b 45 20 48 8b 00 41 c7 44 24 08 00 00 00 00 85 f6 74 27 48 89
[  632.513513] input: DROP Panda (AVRCP) as /devices/virtual/input/input64
[  634.729840] bluetoothd[5672]: segfault at 20 ip 000055a89d971fa5 sp 00007ffd45634760 error 4 in bluetoothd[55a89d956000+d4000] likely on CPU 6 (core 12, socket 0)
[  634.729850] Code: 08 00 0f 1f 00 f3 0f 1e fa 41 55 41 54 55 53 48 83 ec 18 48 8b 2a 4c 8b 62 08 64 48 8b 04 25 28 00 00 00 48 89 44 24 08 31 c0 <48> 8b 45 20 48 8b 00 41 c7 44 24 08 00 00 00 00 85 f6 74 27 48 89
[  637.816925] input: DROP Panda (AVRCP) as /devices/virtual/input/input65
[  639.640007] bluetoothd[5692]: segfault at 20 ip 0000559cb328dfa5 sp 00007fffab831b10 error 4 in bluetoothd[559cb3272000+d4000] likely on CPU 6 (core 12, socket 0)
[  639.640024] Code: 08 00 0f 1f 00 f3 0f 1e fa 41 55 41 54 55 53 48 83 ec 18 48 8b 2a 4c 8b 62 08 64 48 8b 04 25 28 00 00 00 48 89 44 24 08 31 c0 <48> 8b 45 20 48 8b 00 41 c7 44 24 08 00 00 00 00 85 f6 74 27 48 89
[  642.766979] input: DROP Panda (AVRCP) as /devices/virtual/input/input66
[  645.036492] bluetoothd[5717]: segfault at 20 ip 0000560ff4d9cfa5 sp 00007fff72d51af0 error 4 in bluetoothd[560ff4d81000+d4000] likely on CPU 4 (core 8, socket 0)
[  645.036502] Code: 08 00 0f 1f 00 f3 0f 1e fa 41 55 41 54 55 53 48 83 ec 18 48 8b 2a 4c 8b 62 08 64 48 8b 04 25 28 00 00 00 48 89 44 24 08 31 c0 <48> 8b 45 20 48 8b 00 41 c7 44 24 08 00 00 00 00 85 f6 74 27 48 89
[  646.233171] bluetoothd[5744]: segfault at 20 ip 00005571de5cefa5 sp 00007fff00f66d10 error 4 in bluetoothd[5571de5b3000+d4000] likely on CPU 10 (core 20, socket 0)
[  646.233182] Code: 08 00 0f 1f 00 f3 0f 1e fa 41 55 41 54 55 53 48 83 ec 18 48 8b 2a 4c 8b 62 08 64 48 8b 04 25 28 00 00 00 48 89 44 24 08 31 c0 <48> 8b 45 20 48 8b 00 41 c7 44 24 08 00 00 00 00 85 f6 74 27 48 89
[  647.646588] input: DROP Panda (AVRCP) as /devices/virtual/input/input67
```

I am using EndevourOS with KDE

Duck375, 2023-07-01:
Same. Using Arch Linux, KDE. Headphones: HECATE G33BT. Upon connection everything seems fine, but as soon as i try to play any audio it segfaults (according to dmesg). Downgrading by 1 version fixed the issue.
Broken: bluez-5.67-1
Works: bluez-5.66-1

acepukas, 2023-07-01:
Pretty much the same situation as @Duck375. Arch Linux with GNOME. Headphones: Sony WH-1000XM3

jstelzer, 2023-07-01:
Similar situation but using Broadcom Corp. BCM20702A0 Bluetooth 4.0 usb dongle to stream to usb speakers. Arch linux, kde, v5.66-1 works, v5.67-1 segfaults when anything tries to play sound

chromer030, 2023-07-01:
Aparently fixed in https://github.com/bluez/bluez/commit/19319bd3e22ed7945f118bc5faf62a6380070e4a

tuananh, 2023-07-01:
can also confirm this issue. also EndevourOS

Andrei-Pozolotin, 2023-07-13:
this crash is resolved on arch by :
https://archlinux.org/packages/extra/x86_64/bluez/
```
bluez 5.68-1
Build Date: | 2023-07-01 13:59 UTC
```

Vudentz, 2023-07-17:
Close as duplicated to https://github.com/bluez/bluez/issues/542

closed 2023-07-17 by Vudentz