bluez/bluez #550: Bluetooth works but full of junk in journalctl

Freyon77, 2023-07-03:
I checked myself these new versions of bluez.
A lot of spam appeared.

src/adapter.c:reset_adv_monitors_complete() 
Failed to reset Adv monitors: Failed (0x03)

Failed to add UUID: Failed (0x03)

Failed to add UUID: Failed (0x03)

Failed to add UUID: Failed (0x03)

Freyon77, 2023-07-03:
It dissappeared after reboot

Freyon77, 2023-07-03:
Steps to reproduce

Boot Arch Linux installation medium

Mount your drive

Arch-chroot /mnt

Su (your name)

Logout

(You will get an error message saying you should type "exit")

Exit

Exit

Reboot

Now you will see mentioned errors in journalctl

Freyon77, 2023-07-04:
Closing - kernel issue

closed 2023-07-04 by Freyon77

Crandel, 2023-08-12:
Can you share this kernel issue, please?

Freyon77, 2023-08-12:
> @Freyon77 Can you share this kernel issue, please?

No, i have made research again, and on driver site it says my device will no longer be supported by latest driver